realsenseai/librealsense #8851: Incomplete video frame detected - D455 on ubuntu 16.04 via VBox

DukeCheung, 2021-04-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Operating System & Version |   Linux (Ubuntu 16) via VBox | 
| Kernel Version (Linux Only)    |  4.15.0-140-generic                                | 
| Platform                                 | PC |
| SDK Version                            | SDK 2.0 }                          | 

I bought a D455 camera, and it works well on the Windows 10 platform with SDK 2.0. When I try to use it on the Ubuntu via VBox, it gets "Incomplete video frame detected" errors but can still render some frames with RGB Camera or Stereo Module, and **the Motion module cannot be used totally**.
And I have turned on the USB 3.0 port on the VBox and used the official short USB cable.

MartyG-RealSense, 2021-04-17:
Hi @DukeCheung   If you need to use a VM, Intel recommend **VMWare Workstation Player** instead of Oracle VirtualBox due to USB 3 compatability.  

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#linux-ubuntu-installation

DukeCheung, 2021-04-18:
Thanks for your advice!
I've read the installation document and tried to deploy it on VMWare Workstation 16 Player while turned on the USB 3.1. The Motion module works well but the it still shows the error message such as "Incomplete video frame detected" or "Asic Temperature value is not valid". I guess it may be caused by the different USB version between the camera and VM.

MartyG-RealSense, 2021-04-18:
Is the incomplete frames error still present if you reduce stream resolution / FPS to lower the demand on the USB system, please?

DukeCheung, 2021-04-18:
Thank you!
I reduced stream resolution / FPS to lower and it do help that the frequency of the errors has decreased but still not disappear.

MartyG-RealSense, 2021-04-18:
Are you enabling Depth, RGB and Motion Module all at the same time in the RealSense Viewer (all three streams) please?

DukeCheung, 2021-04-18:
Yes, there are much less errors if I only enable RGB & Motion Module.

MartyG-RealSense, 2021-04-18:
This is expected behavior.  It is a known issue that when Depth, RGB and Motion Module are enabled simulateously then errors can occur, most commonly **Frames Not Received** on the RGB stream.  

Whilst there are workarounds in C++ and Python code, there is no fix for it in the Viewer other than to avoid enabling all three streams (for example, use Depth / Motion Module, Color / Motion Module, or just use Depth + RGB and use the Motion Module on its own in a different Viewer session.

DukeCheung, 2021-04-18:
OK, thank you for your patience!

closed 2021-04-25 by DukeCheung

Christie-zhz, 2021-06-13:
Thank you for your answer, I started two streams at the same time, RGB and Depth, using a resolution of 640*480, do I need to upgrade the firmware?