realsenseai/librealsense #8852: Unix Makefiles does not support platform specification, but platform x64

jhynes94, 2021-04-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | N/A| 
| Firmware Version                   | N/A | 
| Operating System & Version |   Linux (Ubuntu 20) | 
| Kernel Version (Linux Only)    |  5.8.0-50-generic         | 
| Platform                                 | PC Laptop  |
| SDK Version                            |   Current 4/17/2021     2.44                     | 
| Language                            |   unity                    | 
| Segment			|  N/A       | 

### Issue Description

Folloing the tutorial: https://github.com/IntelRealSense/librealsense/tree/master/wrappers/unity

At the part:
cmake .. -DBUILD_CSHARP_BINDINGS=ON -DBUILD_UNITY_BINDINGS=ON -DBUILD_SHARED_LIBS=ON -DDOTNET_VERSION_LIBRARY=3.5 -DCMAKE_GENERATOR_PLATFORM=x64 -DUNITY_PATH=~/Unity/Hub/Editor/2020.3.3f1/Editor/Unity

I execute the script as follows and get the following error. I've spent hours trying to figure out what this means and how to solve it.

CMake Error at CMakeLists.txt:3 (project):
  Generator
    Unix Makefiles
  does not support platform specification, but platform
    x64
  was specified.
CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
CMake Error: CMAKE_C_COMPILER not set, after EnableLanguage
-- Configuring incomplete, errors occurred!

MartyG-RealSense, 2021-04-18:
Hi @jhynes94   Unity for Linux is relatively recent.  The RealSense Unity wrapper was designed for use with Windows.  I am not aware of a case where the wrapper has successfully been used with Linux.

jhynes94, 2021-04-19:
Hi Matry,

Thank you for the quick response.

You're right, booted my windows machine and it ran like a charm.

Thanks,

Justin

MartyG-RealSense, 2021-04-19:
You are very welcome  :)

closed 2021-04-27 by MartyG-RealSense