realsenseai/librealsense #8853: rs2_deproject_pixel_to_point vs pointcloud filter in obtaining an aligned point cloud

ssumoo, 2021-04-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR305| 
| Firmware Version                   | 3.27.3.0 | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.4.0-42-generic  | 
| Platform                                 | PC |
| SDK Version                            | 2.44  | 
| Language                            |  C++ | 
| Segment			|  Robot | 

### Issue Description

#### Introduction
Hi, I am relatively new to the community, but I have made sure that I followed the guidelines and checked all of the former issues, I found none of them exactly matching my question.

#### TL;DR
The question is: is it possible to obtain an aligned point cloud via the `rs2::pointcloud `, in contrast to the original aligning the depth to the color frame, then doing the `rs2_deproject_pixel_to_point` for each of the depth frame? If yes, will it be faster? and if yes, how?

(an align point cloud in this issue is defined as a 6-channel image where the channels are r, g, b, x, y, z, which is basically a point cloud where the original pixel locations in the original image are known as well)

#### Contexts & details

After following the examples provided around Github, I worked out how I can obtain my aligned point cloud by basically first passing the frameset into the `rs2::align` filter, then for each pixel, I obtain the relevant information and call `rs2_deproject_pixel_to_point`  to get my remaining x, y, z channel to augment my color frame into my aligned point cloud.

The entire loop takes quite long, which ends up in quite a substantial hit to my framerate as a whole.

I also found out that there is also another potential solution for me to instead use `rs2::pointcloud` [such as in here](https://github.com/IntelRealSense/librealsense/tree/master/examples/pointcloud). Where the generated point cloud actually has a texture map (u, v), which from my understanding, should be able to serve as a pair of indices to the original image, and I can use it to construct my aligned point cloud as well. The apparent advantage of this is that `rs2::pointcloud` runs a lot faster than `rs2_deproject_pixel_to_point`

This leads me to a set of questions that I think at this point is a lot more effective to ask than to blindly experiment:
- Can an aligned point cloud be extracted from `rs2::pointcloud` as I mentioned?
- Will it actually leads to a faster framerate?
- If yes, is it by querying u, v location on the point cloud to get the color information of the aligned point cloud as I mentioned?
- Do I still need to call `rs2::align` to explicitly align the depth to color frame? or is it already handled in the `rs2::pointcloud` filter?

#### Thank you!

MartyG-RealSense, 2021-04-18:
Hi @ssumoo   The function **rs2_deproject_pixel_to_point** is useful for converting 2D pixel information to 3D points.  **rs2::pointcloud** meanwhile is useful for generating a 3D point cloud directly.

I could not see a benefit in using align with rs2::pointcloud, as **pc.map_to(color);** is already mapping the point cloud to the color frame to generate a textured point cloud.

If you use rs2::pointcloud then could adapt this technique further to add **GLSL** acceleration, where some of the processing work is offloaded from the CPU to the graphics GPU,  There may not be a noticable improvement from this method when used on low-end devices though.  The SDK has an example program called **rs-gl-sample** that demonstrates use of GL with an rs2::pointcloud (which becones **rs2::gl::pointcloud** when GL support is added).

https://github.com/IntelRealSense/librealsense/tree/master/examples/gl

If you are primarily concerned with obtaining the details of a single specific RGB pixel coordinate instead of generating an entire aligned cloud, then an alternative that will have faster processing is to use **rs2_project_color_pixel_to_depth_pixel**

https://github.com/IntelRealSense/librealsense/issues/6239#issuecomment-614261704

ssumoo, 2021-04-18:
Hi, thank you for the reply. I think this confirms my guess that an aligned point cloud can be obtained from the point cloud filter instead of manually deprojecting each depth pixel. Thank you!

ssumoo, 2021-04-18:
Hi, I have been trying out the pointcloud filter, the point cloud creation and the uploading to the gpu went fine, and I am seeing a great performance boost on my module, but I have found out a strange issue

A huge portion of the texture coordinates are not between 0 and 1, which is strange to me

Here is roughly what I did:
```cpp
rs2::points points;
rs2::gl::pointcloud pc;
rs2::gl::uploader uploader;

// starting pipes, frameset acquisition into the variable frames

auto color = frames.get_color_frame();
auto depth = frames.get_depth_frame();
color = uploader.process(color);
depth = uploader.process(depth);
pc.map_to(color);
points = pc.calculate(depth);

auto xyz_vertices = points.get_vertices();
auto uv_vertices = points.get_texture_coordinates();

for(std::size_t i=0; i<points.size(); ++i) {
    // from my understanding and https://dev.intelrealsense.com/docs/projection-texture-mapping-and-occlusion-with-intel-realsense-depth-cameras, this should be the normalized coordinates of the color frame that corresponds to the current point of the point cloud
    float nx = uv_vertices[i].u;
    float ny = uv_vertices[i].v;
    if (is_zero(nx) && is_zero(ny)) {
        continue;
    }
    if (nx < 0.0 || nx > 1.0 || ny < 0.0 || ny > 1.0) {
        // This is to prevent the confusing thing that u and v can live outside the 0, 1 range
        continue;
    }
    int px = static_cast<int>(nx * img_cols);
    int py = static_cast<int>(ny * img_rows);
    assign_2x2_patch(px, py, xyz_vertices[i]);
}
```

where `assign_2x2_patch` sets x, y, z channel of the aligned point cloud (to augment the original rgb image with xyz information).

This does actually lead to realistic x, y, z values appended to my aligned point cloud, but it looks like I am losing a lot of x, y, z points due from those points where the u, v values are not between 0 and 1.

Please let me know if I misunderstood something, and where might I be able to find my solutions, thank you

ssumoo, 2021-04-18:
As an update, I think I have managed to worked out the u, v values of the point cloud, the values out of the 0, 1 range indeed needs to be discarded, I think it is due to the fact that the undistorted depth image might contains points that are actually outside of the color image.

However, after doing the mapping and dilating each depth pixel into a 2x2 patch, I am met with this issue that there are these checkerboard-like artifacts that doesn't occur when I do deprojection?

May I ask if there is anything that can be done to it so that clearer x, y, and z images are obtained? Thank you

[image]

MartyG-RealSense, 2021-04-19:
The Align block of the SDK resizes the field of view (FOV) of the stream type that is being mapped to the FOV size of the target stream type.  So if depth to color alignment is being performed then the depth FOV is resized to the color FOV size.  If the color sensor's FOV is smaller than the depth sensor's FOV (as shown on the chart below of the SR300 sensors) then the depth data on the outer regions of the image can get cut off.

[image]

Given your mention of 2x2 patches, can you confirm whether you are using a **decimation filter** with your SR305 please?

https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md#decimation-filter

ssumoo, 2021-04-19:
Hi, I can confirm that I am indeed using the decimation is being used, it is used as the first filter after the pipe's `wait_for_frame` is called. The decimation filter is initialized like so:

```cpp
rs2::decimation_filter decimation_filter(2.0);
```

ssumoo, 2021-04-19:
I have just checked the results without applying the decimation filter, the grid artifacts are now gone. Although I understand that this has something to do with the fact that the decimation filter subsamples the depth image, ending up in 0.25x the original number of points, I still do not yet see why did this grid artifacts not appear when I was doing `rs2_deproject_pixel_to_point`

MartyG-RealSense, 2021-04-20:
Thank you for the confirmation.   Are you applying the decimation filter **before** the align, please?  Doing so helps to avoid a distortion effect called aliasing (jagged lines).

ssumoo, 2021-04-21:
I am using the decimation filter before the align, which strangely worsen the lines problem :/ (this is when I do deprojection with the point cloud filter). The decimation does help with the `rs2_deproject_pixel_to_point` function as you said

MartyG-RealSense, 2021-04-21:
Whilst the recommendation is to perform post-processing before align, there are cases where RealSense users find that the opposite (align after post-process) works best for their particular project.

ssumoo, 2021-04-21:
Hi, thank you for the reply, I have tried switching the orders of the point generation and the alignment, I don't see significant differences between the two variants

MartyG-RealSense, 2021-04-22:
The SR305 has a depth and IR resolution of **640x480** as the SR300 camera technology inside it dates back to early 2016.  So I wonder whether the poor image when decimating is simply a consequence of reducing an already low default resolution by a factor of 0.25.

[image]

ssumoo, 2021-04-24:
I think it's fine for me to stick with the deprojection for now, as it is still able to give me an aligned point cloud as I want, I will look into this problem of the jagged lines when using the point cloud filter with the decimation filter later, thank you!

closed 2021-04-24 by ssumoo