realsenseai/librealsense #8857: How to get all resolutions supported? (C#)

anhTuan0712, 2021-04-19:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | all |
| Firmware Version                   | all | 
| Operating System & Version |   Win (8.1/10)  | 
| Kernel Version (Linux Only)    |                                      | 
| Platform                                 | PC |
| SDK Version                            | 2.43                       | 
| Language                            |  c#                          | 
| Segment			|                       | 

### Issue Description
According to the information here https://github.com/IntelRealSense/librealsense/issues/8710, the support resolution in different FWs is different.
### Question 
I want to use SDK to get all resolutions at current FW using sharp c language.
Please show me how to do it.
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2021-04-19:
Hi @anhTuan0712   It sounds as though you wish to check all supported stream profiles with C#.  I researched your question carefully but could not find a complete script solution for doing so.  However the link below does feature a RealSense user who wrote C# code for a Unity application to list sensor options.  This may be a useful reference for you in regards to developing your own C# sensor query code.

https://support.intelrealsense.com/hc/en-us/community/posts/360033065534-How-to-retrieve-advanced-mode-options-in-C-

If your application is always going to use the same resolution and FPS then such code may be unnecessary though, as the camera data sheet documents list the supported resolutions for each device.  So if you choose one of those standard-supported resolution modes in the data sheet then it is not going to change when new firmwares are introduced.

The hub page for all data sheet documents is here:

https://dev.intelrealsense.com/docs/datasheets

Bear in mind that if you anticipate needing to support use of a 400 Series camera on a USB 2 connection then the list of supported resolution / FPS modes will be more limited, and so you would need to pick a configuration that is supported on both USB 2 and USB 3.  The USB 2 document in the link below provides guidance about this subject.

https://www.intelrealsense.com/usb2-support-for-intel-realsense-technology/

[image]

anhTuan0712, 2021-04-19:
Hi @MartyG-RealSense .
Thanks for your promptness, this problem I have solved already.

MartyG-RealSense, 2021-04-19:
Great news @anhTuan0712 that you found a solution!

anhTuan0712, 2021-04-22:
Yes, thank you once! @ MartyG-RealSense!

closed 2021-04-22 by anhTuan0712