realsenseai/librealsense #8859: Cannot get Imu and Image topic - D455 Ubuntu 16.04

DukeCheung, 2021-04-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Operating System & Version |   Linux (Ubuntu 16) via VMWare | 
| Kernel Version (Linux Only)    |  4.15.0-140-generic                                | 
| Platform                                 | PC |
| SDK Version                            | SDK 2.0                           | 
| Firmware Version                   | 05.12.12.100  | 

I have modified the `rs_camera.launch` file as follow:
`<arg name="unite_imu_method"      default="copy"/>`
`<arg name="enable_sync"           default="true"/>`

And when I run the following command, there is no topic named `/camera/imu`.
`roslaunch realsense2_camera rs_camera.launch` 
`rostopic list`

I used the camera to record a .bag file on windows 10 platform and ran the `rosbag info xx.bag | grep Imu` command on Ubuntu 16.04 via VMWare, I found there are two topics named `/device_0/sensor_2/Accel_0/imu/data` and `/device_0/sensor_2/Gyro_0/imu/data` instead of  `/camera/imu`. How could I rename these topics and merge them into one topic to subscribe?

And there are also two Image topics: `/device_0/sensor_0/Depth_0/image/data` and `/device_0/sensor_1/Color_0/image/data`. Which one should I subscribe?

MartyG-RealSense, 2021-04-19:
Hi @DukeCheung  You can enable the IMU topics in the RealSense ROS wrapper by adding the instructions below to the end of your roslaunch instruction:

**enable_gyro:=true enable_accel:=true**

Using **unite_imu_method**, as you have already done, should cause a single **imu** topic to be published instead of two individual gyro and accel topics when IMU topic publishing is enabled by the roslaunch instructions above.  

The wrapper documentation states: "Setting unite_imu_method creates a new topic, imu, that replaces the default gyro and accel topics.  The imu topic is published at the rate of the gyro.  All the fields of the Imu message under the imu topic are filled out".

Which of thse two topics you subscribe to will depend on whether you want to access depth data or color data from the bag file.  

**Depth**
/device_0/sensor_0/Depth_0/image/data

**Color**
/device_0/sensor_1/Color_0/image/data

DukeCheung, 2021-04-19:
Hi @MartyG-RealSense , we have revised rs_camera.launch file as suggested. However, it turns out that we cannot create the topic named `/camera/imu`. Please find attached our configuration details.
`<arg name="camera"              default="camera"/>`
`<arg name="enable_accel"        default="true"/>`
`<arg name="enable_accel"        default="true"/>`
`<arg name="enable_sync"               default="true"/>`
`<arg name="enable_sync"               default="true"/>`

MartyG-RealSense, 2021-04-19:
Could you try this please:

`<arg name="unite_imu_method" default="linear_interpolation"/>`

DukeCheung, 2021-04-19:
I am so sorry @MartyG-RealSense , I have set "unite_imu_method" with "true", and I changed it to "copy", it solved.
Thank you!

MartyG-RealSense, 2021-04-19:
No problem at all @DukeCheung - it is great to hear that you were successful.  Thanks for the update!

closed 2021-04-25 by DukeCheung