realsenseai/librealsense #8863: Passing config_file argument for ROS2 Wrapper v3.1.5

daziFR, 2021-04-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D435i | 
| Firmware Version                   | 5.12.7.100 | 
| Operating System & Version |   Linux (Ubuntu 20.04)   | 
| Kernel Version (Linux Only)    |  5.11.10                                        | 
| Platform                                 | PC  |
| SDK Version                            |  2.44                        | 
| Language                            |  {C/C++/Python}                          | 
| Segment			|  ROS2                        | 

### Issue Description
Hi!

Intel releases new ROS2 wrapper supporting foxy, eloquent and dashing. 
I've modified `rs_launch.py` by moving all the configurable parameters into `d435i.yaml` file and updating `config_file` default value in `rs_launch.py`.
```
configurable_parameters = [{'name': 'camera_name',                  'default': 'camera', 'description': 'camera unique name'},
                           {'name': 'serial_no',                    'default': "''", 'description': 'choose device by serial number'},
                           {'name': 'usb_port_id',                  'default': "''", 'description': 'choose device by usb port id'},
                           {'name': 'device_type',                  'default': "''", 'description': 'choose device by type'},
                           {'name': 'config_file',                  'default': "'config/d435i.yaml'", 'description': 'yaml config file'}
                          ]
```
With run : 
`ros2 launch realsense2_camera rs_launch.py`
warning is printed:
`[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: 'config/d435i.yaml'`
and parameters are not loaded correctly. 

Parameters written in rs_launch.py or given as cmdl argument are loaded correctly.

MartyG-RealSense, 2021-04-19:
Hi @daziFR   As a starting point in investigating your case, a ROS2 case from February 2021 in the link below provides a demonstration of configuration with a yaml file that may provide useful insights.

https://github.com/IntelRealSense/realsense-ros/issues/1640

daziFR, 2021-04-20:
Hi @MartyG-RealSense, this works fine with the wrapper intended for foxy only (v3.1.3 and v3.1.4). The problem is introduced with added support for dashing and eloquent. 

From commits, change from `' '` to `" ' ' "` in configurable_parameters default values and in `IfCondition` is done for dashing support. But this introduced problem with foxy. Or I set the value wrong for the new wrapper.

daziFR, 2021-04-20:
Additionally, parsing config_file fails in dashing too:
```
[WARNING] [launch_ros.actions.node]: Parameter file path is not a file: '/home/test/intel_realsense_ros2_dashing/src/realsense2_camera/config/d435i.yaml'
[INFO] [realsense2_camera_node-1]: process started with pid [25304]
[realsense2_camera_node-1] terminate called after throwing an instance of 'std::runtime_error'
[realsense2_camera_node-1]   what():  Failed to parse parameters from file ''/home/test/intel_realsense_ros2_dashing/src/realsense2_camera/config/d435i.yaml'': Error opening YAML file, at /tmp/binarydeb/ros-dashing-rcl-yaml-param-parser-0.7.9/src/parser.c:1387
[ERROR] [realsense2_camera_node-1]: process has died [pid 25304, exit code -6, cmd 'stdbuf -o L /home/test/intel_realsense_ros2_dashing/install/realsense2_camera/lib/realsense2_camera/realsense2_camera_node --ros-args --log-level info __node:=camera __ns:=/camera __params:=/tmp/launch_params_r6nc9itn __params:='/home/test/intel_realsense_ros2_dashing/src/realsense2_camera/config/d435i.yaml''].
```

daziFR, 2021-04-20:
Additional info, If I make change in rs_launch.py:
```
parameters = [set_configurable_parameters(configurable_parameters)
                            ,{LaunchConfiguration("config_file")}
                            ],
```
to:
```
parameters = [set_configurable_parameters(configurable_parameters)
                            , PythonExpression([LaunchConfiguration('config_file')])
                            ],
```

in ros2 foxy, parameters from config_file do get parsed and there is no warning any more "parameter file is not a file".

MartyG-RealSense, 2021-04-20:
Given the solution that you described above for Foxy, do you still have a problem that needs to be solved please?  Thanks!

daziFR, 2021-04-22:
With the given proposal problem is solved. 

There is another problem with wrapper v3.1.5 when using Dashing.

If node is run before rviz2, Fixed frame cannot be modified. Check image:

[image]

If I run rviz2 first, and then node, this works fine.

Should i open new case for this?

daziFR, 2021-04-22:
There is an issue with dashing: [tf2_ros::StaticTransformBroadcaster does not keep last msg for later subscription](https://github.com/ros2/geometry2/issues/183) and a solution for intel wrapper: [static tf broadcaster: workaround transient local durability (master branch)](https://github.com/intel/ros2_intel_realsense/pull/87).

With this fix issue mentioned above is gone. 

Does Intel plan to support dashing, and include this missing patch? 

Thank you.

MartyG-RealSense, 2021-04-22:
RealSense ROS does now support ROS2 Dashing, Eloquent and Foxy with a single ROS2 wrapper.

https://github.com/IntelRealSense/realsense-ros/tree/ros2

At the time of writing this though, the Debian package versions of the ROS wrapper are temporarily inaccessible.

https://github.com/IntelRealSense/realsense-ros/issues/1814#issuecomment-823020133

The solution that you identified for the static tf broadcaster is for a ROS2 wrapper that is now obsolete and no longer supported.  The current actively developed ROS2 wrapper is unrelated to that older wrapper and so a fix made for that old wrapper could likely not be applied directly to the current wrapper.

daziFR, 2021-04-22:
"The solution that you identified for the static tf broadcaster is for a ROS2 wrapper that is now obsolete and no longer supported. The current actively developed ROS2 wrapper is unrelated to that older wrapper and so a fix made for that old wrapper could likely not be applied directly to the current wrapper."

I agree. For testing purposes I tried to apply patch and it does work, but as you mentioned, there might be another solution. 

I hope this issue will be reviewed by the Intel ROS2 wrapper developers.

MartyG-RealSense, 2021-04-22:
Thanks very much @daziFR for your test.  I suggest creating an issue about it on the RealSense ROS forum.

https://github.com/IntelRealSense/realsense-ros/issues

daziFR, 2021-04-26:
Thanks @MartyG-RealSense, I've created new issue on the RealseSense ROS forum.

MartyG-RealSense, 2021-04-26:
Thanks very much for the RealSense ROS issue creation.

MartyG-RealSense, 2021-05-03:
Hi @daziFR  Do you require further assistance with this case, please?  Thanks!

daziFR, 2021-05-03:
No thanks.

Topic can be closed.

closed 2021-05-03 by daziFR