realsenseai/librealsense #8865: librealsense ros kinetic is not installable in docker

tfoote, 2021-04-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  N/A | 
| Firmware Version                   | N/A | 
| Operating System & Version |   Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  (5.4 and 5.8 in docker)                                         | 
| Platform                                 | amd64, i386, arm32, arm64 |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  ROS                      | 
| Segment			|  Robot                     | 

### Issue Description
`ros-kinetic-librealsense` fails to install inside docker and consequently on the ROS buildfarm due to an error in the postinstall. 

This is urgent because we're [preparing for the final sync](https://discourse.ros.org/t/preparing-for-final-kinetic-sync/19805) of kinetic and if this package is not installable all dependencies of this package will no longer be installable in the final snapshot of Kinetic. 

This is failing [downstream builds of realsense_camera](https://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__realsense_camera__ubuntu_xenial_amd64__binary/) 
Example failure (https://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__realsense_camera__ubuntu_xenial_amd64__binary/14/consoleText) 

```
  4559 'Setting up ros-kinetic-librealsense (1.12.1-0xenial-20191214-013909+0000) ...'
  4560 '/var/lib/dpkg/info/ros-kinetic-librealsense.postinst: 40: /var/lib/dpkg/info/ros-kinetic-librealsense.postinst: udevadm: not found'
  4561 'libkmod: ERROR ../libkmod/libkmod.c:586 kmod_search_moddep: could not open moddep file '/lib/modules/5.4.0-1029-aws/modules.dep.bin''
  4562 'modinfo: ERROR: Module alias uvcvideo not found.'
  4563 'dpkg: error processing package ros-kinetic-librealsense (--configure):'
```

The [previous build](https://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__realsense_camera__ubuntu_xenial_amd64__binary/13/consoleText) was successful and appears to avoid trying to do kernel mods: 
```
Setting up ros-kinetic-librealsense (1.12.1-0xenial-20191214-013909+0000) ...
INFO: Docker environment not supported for udevadm; skipping.
INFO: Docker environment not supported for DKMS; skipping.
```

To reproduce:

`docker run -ti --rm ros:kinetic bash -c 'apt-get update && apt-get install -y ros-kinetic-librealsense'`

<details>
<summary>I was able to reproduce this locally in docker. Click for full console output.</summary>

```
$ docker run -ti --rm ros:kinetic bash -c 'apt-get update && apt-get install -y ros-kinetic-librealsense'
Get:1 http://security.ubuntu.com/ubuntu xenial-security InRelease [109 kB]
Get:2 http://packages.ros.org/ros/ubuntu xenial InRelease [4692 B]             
Get:3 http://archive.ubuntu.com/ubuntu xenial InRelease [247 kB]                          
Get:4 http://packages.ros.org/ros/ubuntu xenial/main amd64 Packages [855 kB]                 
Get:5 http://security.ubuntu.com/ubuntu xenial-security/main amd64 Packages [2040 kB]             
Get:6 http://security.ubuntu.com/ubuntu xenial-security/restricted amd64 Packages [15.9 kB]                       
Get:7 http://security.ubuntu.com/ubuntu xenial-security/universe amd64 Packages [985 kB]
Get:8 http://security.ubuntu.com/ubuntu xenial-security/multiverse amd64 Packages [8820 B]
Get:9 http://archive.ubuntu.com/ubuntu xenial-updates InRelease [109 kB]       
Get:10 http://archive.ubuntu.com/ubuntu xenial-backports InRelease [107 kB]
Get:11 http://archive.ubuntu.com/ubuntu xenial/main amd64 Packages [1558 kB]
Get:12 http://archive.ubuntu.com/ubuntu xenial/restricted amd64 Packages [14.1 kB]
Get:13 http://archive.ubuntu.com/ubuntu xenial/universe amd64 Packages [9827 kB]
Get:14 http://archive.ubuntu.com/ubuntu xenial/multiverse amd64 Packages [176 kB]
Get:15 http://archive.ubuntu.com/ubuntu xenial-updates/main amd64 Packages [2549 kB]
Get:16 http://archive.ubuntu.com/ubuntu xenial-updates/restricted amd64 Packages [16.4 kB]
Get:17 http://archive.ubuntu.com/ubuntu xenial-updates/universe amd64 Packages [1545 kB]
Get:18 http://archive.ubuntu.com/ubuntu xenial-updates/multiverse amd64 Packages [26.2 kB]
Get:19 http://archive.ubuntu.com/ubuntu xenial-backports/main amd64 Packages [10.9 kB]
Get:20 http://archive.ubuntu.com/ubuntu xenial-backports/universe amd64 Packages [12.6 kB]
Fetched 20.2 MB in 3s (6231 kB/s)                             
Reading package lists... Done
Reading package lists... Done
Building dependency tree       
Reading state information... Done
[... 82 lines cut ...]
Setting up libusb-1.0-0-dev:amd64 (2:1.0.20-1) ...
Setting up libusb-1.0-doc (2:1.0.20-1) ...
Setting up linux-headers-4.4.0-209 (4.4.0-209.241) ...
Setting up linux-headers-4.4.0-209-generic (4.4.0-209.241) ...
/etc/kernel/header_postinst.d/dkms:
 * dkms: running auto installation service for kernel 4.4.0-209-generic
   ...done.
Setting up linux-headers-generic (4.4.0.209.215) ...
Setting up ros-kinetic-librealsense (1.12.1-0xenial-20191214-013909+0000) ...
debconf: unable to initialize frontend: Dialog
debconf: (No usable dialog-like program is installed, so the dialog based frontend cannot be used. at /usr/share/perl5/Debconf/FrontEnd/Dialog.pm line 76.)
debconf: falling back to frontend: Readline
/var/lib/dpkg/info/ros-kinetic-librealsense.postinst: 40: /var/lib/dpkg/info/ros-kinetic-librealsense.postinst: udevadm: not found
libkmod: ERROR ../libkmod/libkmod.c:586 kmod_search_moddep: could not open moddep file '/lib/modules/5.8.0-49-generic/modules.dep.bin'
modinfo: ERROR: Module alias uvcvideo not found.
dpkg: error processing package ros-kinetic-librealsense (--configure):
 subprocess installed post-installation script returned error exit status 1
Processing triggers for systemd (229-4ubuntu21.29) ...
Processing triggers for libc-bin (2.23-0ubuntu11.2) ...
Errors were encountered while processing:
 ros-kinetic-librealsense
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

</details>

MartyG-RealSense, 2021-04-20:
Hi @tfoote  Thanks very much for the urgent information.  I will direct it to @doronhi the RealSense ROS wrapper developer.

doronhi, 2021-04-20:
Thanks for the tip. I was aiming for the next librealsense2 release but this can be changed. @tfoote , can you estimate how much time do I have?

tfoote, 2021-04-20:
Today is the start of the freeze, we're officially bugfix/regression only until the 26th.

doronhi, 2021-04-22:
I mistakenly thought you referred to realsense2_camera and librealsense2. I took the opportunity, made a final release, and submitted PRs.
As to realsense_camera and librealsense, they were not changed for at least 3 years. I also didn't understand what the issue was. I see they both available in my last `apt update` and also successfully run `sudo apt install ros-kinetic-realsense-camera`
I tried the reproduction method using the docker you suggested and indeed it fails. Did it succeed before?

tfoote, 2021-04-23:
@doronhi I think that the issue is that the logic to detect the kernel version/that you're inside docker has broken with some docker update in the last 4-6 months. librealsense is failing during the postinstall stage which used to skip the kernel module work inside docker. The reason they're still available via apt-get is that our public repositories are latched at each sync. You'll find that they're missing from the testing repository and when I run the next (and final sync they will disappear) 

Yes the reproduction used to pass. The skipping logic that I showed from the last successful build would have been expected.

doronhi, 2021-04-29:
Using the solution given [here](https://github.com/IntelRealSense/librealsense/issues/4781#issue-488076227) by Andy Edmonds (@dizz) I created a debian that can be installed on the docker.
[ros-kinetic-librealsense_1.12.1-0xenial-20191214-docker.zip](https://github.com/IntelRealSense/librealsense/files/6398574/ros-kinetic-librealsense_1.12.1-0xenial-20191214-docker.zip)
I still need to figure out how to upload it correctly but since we are so short on time, I wondered if it will be possible to use it for the release.

doronhi, 2021-04-29:
The change is in the file DEBIAN/postinst. 
I guess the DEBIAN directory is created while building the debian but since that process happens in the ros build farm I could use some help as to how can I change the results of that process.

tfoote, 2021-05-04:
The recommended way to inject this is to use a release patch in the bloom process. http://wiki.ros.org/bloom/Tutorials/ReleaseThirdParty

But it appears that the the "upstream" branch for releasing is inserting the debian content: https://github.com/IntelRealSense/librealsense/tree/rosdebian is listed as the upstream source for the [last kinetic release](https://github.com/intel-ros/librealsense-release/blob/a6b99c9ed8559fd56c84990a45c6b9c16a858f60/tracks.yaml#L40)

I believe that the patched postinst file from @dizz should be applied to https://github.com/IntelRealSense/librealsense/tree/rosdebian

After applying that the version should be incremented following the usual release process to create a new version and then the bloom release run from the newly tagged version.

doronhi, 2021-05-04:
Hi @tfoote . Thanks for all your help in understanding the issue.
Unfortunately, Intel no longer supports this version of librealsense. Therefore will not be updating or maintaining it. Users are to upgrade to librealsense2 or download and build the code from the still available legacy branch.

tfoote, 2021-05-05:
@doronhi That's very unfortunate, there appears to be a communication issue here. It's currently [marked as maintained](https://github.com/ros/rosdistro/blob/44a1203ef2b6dfe89259cde0bdc3aec96ab69be1/kinetic/distribution.yaml#L6885) and I don't know of any announcement of the deprecation, migration path, or other communication to downstream maintainers that they must migrate. I don't find any mention of the deprecation or migration path in the [package documentation](http://wiki.ros.org/librealsense) either. Dropping support within the distro is very uncommon and unexpected by downstream maintainers, to the level that I would have recommended a highly visible public announcement as well as ticketing any downstream packages so that they knew that they're using something that is no longer supported.

There are 35 other packages relying on this package in kinetic [which are all now regressed and no longer available in the distro because of this.](http://repositories.ros.org/status_page/ros_kinetic_default.html?q=REGRESSION) And the maintainers will now have no opportunity to fix them because there was no deprecation notice. This includes a large fraction of the turtlebot ecosystem and the realsense_camera package, also marked as maintained,which were relying on the packages released into the distro to be stable for the lifetime of the distribution.

Is there anything you can do to get permission to apply this existing patch make a release which will help us avoid announcing the regression of librealsense and 35 downstream packages in the final release announcement?

doronhi, 2021-05-05:
Noted. Will update...

MartyG-RealSense, 2021-05-16:
Hi @tfoote   Do you have further comment that you would like to add to this discussion please?  Thanks!

tfoote, 2021-05-16:
This has been resolved with the release of 2.45.0 in the PR linked above by doronhi.

PR #8976 (2021-05-06): fix to docker detection
aim to fix #8865
Applied solution by @dizz at #4781
commit 3a1af4d6 (2021-05-06): fix to docker detection (@dizz at #4781) fix #8865
closed 2021-05-16 by tfoote