realsenseai/librealsense #8866: Receiver light saturation

Lostosaurus, 2021-04-20:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (01.05.04.01) | 
| Operating System & Version |   {Win (10) /raspbian buster/jetpack4.5  | 
| Platform                                 | PC/Raspberry Pi/ Jetson Xavier NX   |
| SDK Version                            |  { legacy / 2.0 }                          | 
| Language                            |  {python }                          | 

### Issue Description
Error that may be overcome in a few sec. x41
Timestamp. etc.
Severity: Error
Description: Receiver light saturation, ,stream stopped for 1 second.

The L515 worked properly for a few weeks, but recently whenever it is plugged in it continually has the above error. It will turn off the laser repeatedly.

It does this inside with no lights on, only ambient light, as well as in brightly lit large rooms.

I have tested it on a windows pc, jetson xavier nx and a raspi with the same results.

Is this due to some setting inside the l515? I think there is very little chance that there is enough light to overload the sensor.

woscho, 2021-04-21:
Hi @Lostosaurus 
imho, it's the same error, we have over here  IntelRealSense/realsense-ros "Hardware Notification L515 LiDAR Camera" #1750. 

Could you please try to run the Realsense Viewer and watch the system temperatures of the L515 camera you have. With mine, the same error occures, if temperatures reach values around LDD=48, Mc=42, Ma=41,Apd=38 (room temperature is around 23°C), then the Apd goes wild and the error reappears every 2 seconds or so. May be you can confirm the same behavior at the same temperature levels? Something inside gets too hot, I guess. :-/

Lostosaurus, 2021-04-22:
Hello @woscho 

I'll take a look. I didn't know that the viewer reports the temp.

RealSenseSupport, 2021-05-06:
Hi @Lostosaurus 

Are you still having the issue reported?  Also just verifying that nothing, no object, is really close to the camera face cover glass?

If you're still having the issue please submit a ticket here:  https://support.intelrealsense.com/hc/en-us/requests/new you may need to RMA the unit that you have.

closed 2021-06-18 by RealSenseSupport