realsenseai/librealsense #8868: load config.json in python

julienguegan, 2021-04-20:
I would like to **load a json config file** to read .bag file previoulsy saved with the good config but it is not clear how to do it ... I have found this [tutorial](https://github.com/IntelRealSense/librealsense/blob/development/wrappers/python/examples/python-rs400-advanced-mode-example.py).

From my understanding, we have to use an advanced_mode to do so but there is no information on how to integrate it on a whole pipeline to read a .bag file (as [here](https://github.com/IntelRealSense/librealsense/blob/development/wrappers/python/examples/read_bag_example.py)). I would need to understand to points : 

1. The tutorial use a connected device but I would like to use the one used when creating the .bag file so I do :

     `dev = ctx.load_device("path/to/my/file.bag")`

     but then I have the exception  **RuntimeError: null pointer passed for argument "dev"** when doing (after loading device) 
     `advnc_mode.load_json(json_string)` or `advnc_mode.toggle_advanced_mode(True)` ... 

2. Another thing that is not clear and that would deserve more explanations. Is what to do when the json is load to the advnc_mode object, normally you would create a pipeline object and call pipeline.start(config). But I have no idea on how to integrate the advanced mode on this pipeline...

If someone have some more insight on this matter, I would be extremly grateful !

Thank you

MartyG-RealSense, 2021-04-20:
Hi @julienguegan   There was a past case (in the link below) that was experiencing **RuntimeError: null pointer passed for argument "dev"** when loading a json file.  As that case did not involve reading a bag, I recommend focusing on the json-loading aspect of the project first.

https://github.com/IntelRealSense/librealsense/issues/7122

The Python script in the link below will hopefully provide a successful template for setting up json loading correctly.  This particular script has the simplicity advantage of demonstrating the loading of a specifically named json file (e.g **custom.json**).

 https://github.com/IntelRealSense/librealsense/issues/6075

julienguegan, 2021-04-20:
Yes, ok I had no problem do run this template. I was doing quite the same but I am not quite sure how do you use **advnc_mode**  object in your following pipeline to read frame ...

Also tell me if my understanding from [7122](https://github.com/IntelRealSense/librealsense/issues/7122) is correct but when reading a .bag file, it is not needed to load the config file as the data are already correctly read ? Then, do I have a way to check that the config parameters used are actually those that I want ? (because when looking to distance in one saved file, I have inaccurate values that I would like to understand)

MartyG-RealSense, 2021-04-20:
There is an Advanced Mode example script for Python.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-rs400-advanced-mode-example.py

The bag file captures the streams and their settings that were active at the time of recording and recreates them during playback.  This setting from a recording of values such as exposure and gain is similar to the function that loading a json file would perform with real-time camera data).

[image]

If your goal is to read current configuration settings from a bag, I believe that you would do so by accessing the bag's **metadata** values.

https://github.com/IntelRealSense/librealsense/issues/3717#issuecomment-482124573

closed 2021-04-21 by julienguegan