realsenseai/librealsense #8872: Issue of multiple devices record through realsense-viewer

victorll998, 2021-04-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 T265 | 
| Firmware Version                   | v2.42.0 | 
| Operating System & Version |   Linux Ubuntu 18.04   | 
| Kernel Version (Linux Only)    |  4.15.0-142-generic                                   | 
| Platform                                 | PC  |
| SDK Version                            |  2.42.0                     | 
| Language                            |  C                   | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

---

### Issue Description
Hi there,

I haven't encountered this issue earlier when dealing with the record of d455 and t265. 

For now I am trying to record 3 devices (2 d455 and 1 t265) that are connected through a 3.0 USB docker. I use all three 3.0 cables and realsense-viewer to obtain the bag format files. I noticed that warning  ```(messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61``` keeps occurring. From what I have learned from other posts, it suggests communication problems with the device. I have tried ```initial_reset``` and ```set all stream rate (color and depth fps) to 5``` and these warnings are somehow relieved. But when I move the camera these warnings pop out again.

```
 21/04 14:35:12,548 WARNING [140476483663616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 14:35:31,617 WARNING [140476483663616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 14:35:37,674 WARNING [140476483663616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 14:35:38,726 WARNING [140476483663616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 14:35:42,780 WARNING [140476483663616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 14:35:42,831 WARNING [140476483663616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 14:35:42,881 WARNING [140476483663616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 14:35:42,932 WARNING [140476483663616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 14:35:42,982 WARNING [140476483663616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 14:35:43,034 WARNING [140476483663616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 14:35:43,085 WARNING [140476483663616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 14:35:45,138 WARNING [140476483663616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 14:35:46,190 WARNING [140476483663616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 21/04 14:35:48,243 WARNING [140476483663616] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

```

Though I can visualize related topics in Rviz, I am worried about this issue could affect the quality of the recording (i.e periodic frame missing)

- Besides, the log file shows other warnings when the viewer is recording. I checked the bag file with ```rs-rosbag-inspector``` and there is no missing message from what I have seen. I wonder what are these warnings about and if there is a way to fix them.

```
21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Motion
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Video
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Roi
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Depth Frame
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Advanced Mode
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Video Profile
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Disparity Frame
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Motion Profile
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Pose Frame
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Pose Profile
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Tm2
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Software Device
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Software Sensor
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Decimation Filter
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Threshold Filter
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Disparity Filter
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Spatial Filter
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Temporal Filter
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Hole Filling Filter
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Zero Order Filter
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Pose
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Pose Sensor
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Wheel Odometer
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Global Timer
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Updatable
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Update Device
 21/04 14:43:25,032 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Tm2 Sensor
[... 26 lines cut ...]
 21/04 14:43:25,033 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Disparity Filter
 21/04 14:43:25,033 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Spatial Filter
 21/04 14:43:25,033 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Temporal Filter
 21/04 14:43:25,033 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Hole Filling Filter
 21/04 14:43:25,033 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Zero Order Filter
 21/04 14:43:25,033 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Pose
 21/04 14:43:25,033 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Pose Sensor
 21/04 14:43:25,033 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Wheel Odometer
 21/04 14:43:25,033 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Global Timer
 21/04 14:43:25,033 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Updatable
 21/04 14:43:25,033 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Update Device
 21/04 14:43:25,033 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Tm2 Sensor
[... cut ...]
 21/04 14:43:25,033 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Calibrated Sensor
 21/04 14:43:25,033 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Hdr Merge
 21/04 14:43:25,033 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Sequence Id Filter
 21/04 14:43:25,033 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Max Usable Range Sensor
 21/04 14:43:25,033 WARNING [139693896828672] (record_device.cpp:260) Extensions type is unhandled: Debug Stream Sensor

```

RealSenseSupport, 2021-04-23:
@victorll998 Just want to check with you that the 3 devices (2 d455 and 1 t265) were connected to one USB port through USB hub or three USB ports separately？Did you have issue if you only start streaming(no recording started) for these 3 devices?  Which streaming did you enable in your testing and how about the resolution and fps? Looking forward to your reply. Thanks!

victorll998, 2021-04-27:
@RealSenseSupport I am sorry for my late responses. 

For your questions:

- I used a powered USB3.0 hub (external 5V power supply) to connect 3 devices.  
- Everything was fine until I started to record.
- I only enabled depth and color streams (640 x 480 with a rate of 5 fps) for two d455 and pose, gyro and accel streams (848 x 800 with a rate of 200 fps) for t265

victorll998, 2021-04-27:
I updated both firmware and software to 2.44.0 and I used the same camera configurations at ```realsense-viewer```.

camera 1[image]
camera 2[image]
camera 3 [image]
 

You can see both of the previous warnings are still existing. Here is what I get from when I start recording:

```
 27/04 15:04:16,955 WARNING [140469052897024] (backend-v4l2.cpp:1057) Incomplete frame received: Incomplete video frame detected!
Size 279552 out of 614655 bytes (45%)
 27/04 15:04:22,106 ERROR [140469027718912] (record_device.cpp:212) Failed to downcast snapshot of type librealsense::debug_interface
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Motion
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Video
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Roi
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Depth Frame
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Advanced Mode
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Video Profile
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Disparity Frame
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Motion Profile
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Pose Frame
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Pose Profile
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Tm2
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Software Device
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Software Sensor
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Decimation Filter
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Threshold Filter
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Disparity Filter
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Spatial Filter
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Temporal Filter
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Hole Filling Filter
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Zero Order Filter
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Pose
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Pose Sensor
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Wheel Odometer
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Global Timer
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Updatable
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Update Device
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Tm2 Sensor
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Auto Calibrated Device
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Depth Huffman Decoder
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Serializable
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Fw Logger
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Auto Calibration Filter
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Device Calibration
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Calibrated Sensor
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Hdr Merge
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Sequence Id Filter
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Max Usable Range Sensor
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Debug Stream Sensor
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Calibration Change Device
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Motion
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Video
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Roi
 27/04 15:04:22,106 WARNING [140469027718912] (record_device.cpp:260) Extensions type is unhandled: Depth Frame
[... cut ...]
 27/04 15:15:33,123 WARNING [140469036111616] (backend-v4l2.cpp:1057) Incomplete frame received: Incomplete video frame detected!
Size 214016 out of 614655 bytes (34%)
 27/04 15:15:41,140 WARNING [140469052897024] (backend-v4l2.cpp:1057) Incomplete frame received: Incomplete video frame detected!
Size 410624 out of 614655 bytes (66%)
 27/04 15:17:41,137 WARNING [140469052897024] (backend-v4l2.cpp:1057) Incomplete frame received: Incomplete video frame detected!
Size 508928 out of 614655 bytes (82%)
```

RealSenseSupport, 2021-04-30:
@victorll998 According to https://github.com/IntelRealSense/librealsense/issues/5006#issuecomment-540415227 , the warning logs regarding the recordings are expected and not indicative of misbehavior.

victorll998, 2021-05-03:
@RealSenseSupport thank you, I am glad to know there's nothing wrong with the extension type warning. 

What about the issue of the incomplete frame received when recording? It seems to be of quite high frequency to me.

RealSenseSupport, 2021-05-10:
@victorll998 According to https://github.com/IntelRealSense/librealsense/issues/927#issuecomment-375963696 ,  the partial frames issue usually indicate that the uvcvideo driver does not get enough USB buffers from the USB subsystem. This can be either because the system is overloaded (possibly thermal-throttled), CPU utilisation is near 100%, or the USB controller is not handling the requests in time.  Could you please try to reduce the camera number or connect them to separate USB port to see if you still get the warning?

victorll998, 2021-05-12:
thank you @RealSenseSupport

I did consider the case when the system is overloaded. So I have switched from Nano Tx2 to Intel Nuc-i7. The number of incomplete frame warnings is largely decreased when I try to record three cameras. 

From what I learned, the fps performance is also somehow related to latency. Is there a way (i.e camera configure) to reduce latency when using realsense-viewer?

RealSenseSupport, 2021-05-17:
@victorll998 If you're asking for depth streaming, please try with better performance CPU and high FPS configuration. Thanks!

victorll998, 2021-05-19:
@RealSenseSupport big thanks. I think the issue of the incomplete frame while recording is solved. I have mentioned "control transfer returned error" in the other thread https://github.com/IntelRealSense/realsense-ros/issues/1843, therefore I will close this issue.

closed 2021-05-19 by victorll998