realsenseai/librealsense #8874: D435 "No device detected". Raspberry Pi 3B+

williamx97, 2021-04-21:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.12.12.100 | 
| Operating System & Version |  Raspbian GNU / Linux 9 | 
| Kernel Version (Linux Only)    |  4.14.79                                        | 
| Platform                                 | Raspberry Pi 3B+ |
| SDK Version                            |  2.43.0                        | 
| Language                            |  ROS (Using Realsense ROS Wrapper 2.2.23)                    | 
| Segment			|  Robot                       | 

### Issue Description
Hi there,

Our team is having issues getting our D435 to show on ```rs-enumerate-devices``` . We are getting the ```No device found. Is it plugged in?``` error message. 
We are using a custom version of raspbian provided by our robot manufacturer called TurtleOS: https://github.com/TurtleRover/turtleos  
It is based on Linux Kernel ```4.14.79```

Here are the troubleshooting steps applied:
1. Installing librealsense via these instructions https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md
2. Installing librealsense via direct download of v2.43.0: https://github.com/IntelRealSense/librealsense/releases/tag/v2.43.0 and into ```~/catkin_ws/src``` and using ROS command ```catkin_build``` . The UDEV rules were also updated. 
2. Upgrading the linux kernel using ```sudo apt-get update && sudo apt-get upgrade ```. Kernel patched to 3.19
3. Upgrading the linux kernel using ```sudo rpi-update```. Kernel patched to 5.10 ,still did not work
4. Upgrading the linux kernel to a supported version using ```sudo rpi-update 453e49bdd87325369b462b40e809d5f3187df21d``` the specific commit this command used was here https://github.com/Hexxeh/rpi-firmware/commit/453e49bdd87325369b462b40e809d5f3187df21d we could not connect to our raspberry pi however

Note both methods 1 and 2 seem to detect our other T265 camera that we have with us: 
```
pi@raspberrypi:~ $ rs-enumerate-devices
Device info:
    Name                          :     Intel RealSense T265
    Serial Number                 :     948422110797
    Firmware Version              :     0.2.0.951
    Physical Port                 :     1-1.1.2-9
    Product Id                    :     0B37
    Usb Type Descriptor           :     2.1
    Product Line                  :     T200

Stream Profiles supported by Tracking Module
 Supported modes:
    stream       resolution      fps       format
    Fisheye 1     848x800       @ 30Hz     Y8
    Fisheye 2     848x800       @ 30Hz     Y8
    Gyro         N/A            @ 200Hz    MOTION_XYZ32F
    Accel        N/A            @ 62Hz     MOTION_XYZ32F
    Pose         N/A            @ 200Hz    6DOF
pi@raspberrypi:~ $

```

The outputs of ```lsusb``` and `'`dmesg``` seem to also detect the D435 camera:
```
pi@raspberrypi:~/turtle_ws $ lsusb
Bus 001 Device 004: ID 148f:3070 Ralink Technology, Corp. RT2870/RT3070 Wireless Adapter
Bus 001 Device 006: ID 8086:0b07 Intel Corp.
Bus 001 Device 005: ID 0424:7800 Standard Microsystems Corp.
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
```
pi@raspberrypi:~/turtle_ws $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.79-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1159 SMP Sun Nov 4 17:50:20 GMT 2018
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] Memory policy: Data cache writealloc
[... 273 lines cut ...]
[   10.462541] brcmfmac: power management disabled
[   10.868640] IPv6: ADDRCONF(NETDEV_UP): wlan_int: link is not ready
[   13.388616] uart-pl011 3f201000.serial: no DMA platform data
[   14.585410] IPv6: ADDRCONF(NETDEV_UP): wlan_int: link is not ready
[   14.585465] brcmfmac: power management disabled
[   21.971220] IPv6: ADDRCONF(NETDEV_CHANGE): wlan_int: link becomes ready
[   29.110708] Voltage normalised (0x00000000)
[   37.429479] Under-voltage detected! (0x00050005)
[   66.546543] Voltage normalised (0x00000000)
[   74.865926] Under-voltage detected! (0x00050005)
[   83.185400] Voltage normalised (0x00000000)
[  220.451859] usb 1-1.1.2: new high-speed USB device number 6 using dwc_otg
[  220.583208] usb 1-1.1.2: New USB device found, idVendor=8086, idProduct=0b07
[  220.583221] usb 1-1.1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  220.583230] usb 1-1.1.2: Product: Intel(R) RealSense(TM) Depth Camera 435
[  220.583239] usb 1-1.1.2: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435
pi@raspberrypi:~/turtle_ws $ rs-enumerate-devices
No device detected. Is it plugged in?
pi@raspberrypi:~/turtle_ws $

```

Was it strictly necessary that the Linux kernel be upgraded to a supported version mentioned in the release notes?

Furthermore was there an alternate way to upgrade the Linux Kernel, considering method 5 caused a disconnect from the raspberry pi?

RealSenseSupport, 2021-04-25:
@williamx97 So I understood you followed https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md#install-realsense-sdklibrealsense to build librealsense, right?  LibUVC and WinUSB are deprecated from the build in SDK 2.30+, replaced by RSUSB. So could you please try to re-build with the following? Thanks! 
cmake .. .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release -DFORCE_RSUSB_BACKEND

williamx97, 2021-04-27:
Thanks! I've recompiled it with the cmake flags listed above and now ```rs-enumerate-devices```  is detecting the D435, however I am getting the ```RS2_USB_STATUS_BUSY``` error:
```
pi@raspberrypi:~ $ rs-enumerate-devices
Could not create device - failed to set power state . Check SDK logs for details
 27/04 10:00:11,542 ERROR [1939432528] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
 27/04 10:00:11,543 ERROR [1995515488] (sensor.cpp:526) acquire_power failed: failed to set power state
```
I have rebooted the raspberry pi and changed the usb ports. The T265 camera still detects correctly though.

Would there be a way to detect whether a linux process is using the D435 or were there some other realsense debugging tools I could employ?

williamx97, 2021-04-27:
Update to the above: Reversing the USBC connection to the D435 seemed to fix the issue temporarily. I unplugged and replugged the USBC sides to confirm that only one side of the USBC connection was causing the ```RS2_USB_STATUS``` error.

However after running some tests within the Realsense ROS wrapper seemed to cause the ```RS2_USB_STATUS``` message to come back after a few roslaunch files:

<details>
<summary> Output of Roslaunch commands (click to expand)</summary>

```
pi@raspberrypi:~/turtle_ws $ roslaunch realsense2_camera rs_camera.launch filters:=pointcloud
... logging to /home/pi/.ros/log/f217f562-a759-11eb-856b-b827eb203c41/roslaunch-raspberrypi-26953.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://raspberrypi:38661/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: -1
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file:
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: -1
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: -1
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: -1
 * /camera/realsense2_camera/confidence_fps: -1
 * /camera/realsense2_camera/confidence_height: -1
 * /camera/realsense2_camera/confidence_width: -1
 * /camera/realsense2_camera/depth_fps: -1
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: -1
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: -1
 * /camera/realsense2_camera/device_type:
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_confidence: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: -1
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: -1
[... 301 lines cut ...]
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [27988]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 4f8ad624-a75a-11eb-bc05-b827eb203c41
process[rosout-1]: started with pid [28001]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [28018]
process[camera/realsense2_camera-3]: started with pid [28019]
[ INFO] [1619529202.909940366]: Initializing nodelet with 4 worker threads.
[ INFO] [1619529203.289425338]: RealSense ROS v2.2.24
[ INFO] [1619529203.289801534]: Built with LibRealSense v2.44.0
[ INFO] [1619529203.290116794]: Running with LibRealSense v2.44.0
[ INFO] [1619529203.445774134]:
 27/04 14:13:23,492 ERROR [1765798944] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
[ WARN] [1619529203.503196492]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1619529203.504015344]: The requested device with  is NOT found. Will Try again.

```
</details>

```rs-enumerate-devices``` now shows the ```RS2_USB_STATUS_BUSY``` for both sides of the USBC connection.

RealSenseSupport, 2021-04-29:
@williamx97 Did you use the original USB cable when get the issue?

williamx97, 2021-04-29:
@RealSenseSupport Yes I am using the original USB cable that came with D435. There are occasional moments where reversing the USB C side seems to work, however most times the ```RS2_USB_STATUS``` error message appears on both sides.

RealSenseSupport, 2021-05-10:
@williamx97 Did you get such issue if you connect this D435 to other PC like x86 PC with the original cable? Thanks!

williamx97, 2021-05-12:
@RealSenseCustomerSupport I'm not getting the ```RS2_USB_STATUS``` error when connected to my x64 laptop. I dont have an x84 laptop handy at the moment. My x64 laptop is running Ubuntu 20.04 and librealsense was not built from source but simply installed using the default ```sudo apt-get``` method. It is still using the original cable

RealSenseSupport, 2021-05-14:
@williamx97 Did you install ROS wrapper by  _sudo apt-get install ros-$ROS_DISTRO-realsense2-camera_  ? According to https://github.com/IntelRealSense/realsense-ros/issues/1187#issuecomment-759367599 , please try the following:
1. Remove the ros-librealsense2 and realsense2_camera installation: _sudo apt purge ros-melodic-realsense2_camera_
2. Rebuild realsense2_camera from code according to https://github.com/IntelRealSense/realsense-ros#step-2-install-intel-realsense-ros-from-sources

RealSenseSupport, 2021-05-25:
@williamx97 Any update from your side? Looking forward to your reply. Please note that the ticket will be closed if we don't hear from your for another 7 days. Thanks!

RealSenseSupport, 2021-06-02:
@williamx97 Any update from your side? Thanks!

RealSenseSupport, 2021-07-13:
@williamx97 Sorry that we don't hear from you for weeks. This issue will be closed. Please create new issue if you still have other questions or issues. Thanks!

closed 2021-07-13 by RealSenseSupport