realsenseai/librealsense #8877: Grouping depth values by change with respect to pixel count over frame length.

PlasticBlaze, 2021-04-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  05.12.12.100  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Raspberry Pi  |
| SDK Version                            |  v2.43.0        | 
| Language                            |  opencv/ python }                          | 
| Segment			| Robot                         | 

### Issue Description
Hello,
Not sure if I really titled this correctly, figured I might be able to explain it a little better in the body of the question.
I had a question about a topic I've been searching around for a solution to but wasn't able to find anything yet. I'm running a D435 with a Raspberry Pi. My camera will be stationary at a *key word here* "somewhat" consistent distance from a surface, starting the stream and capturing 1 frame from the stream. My snippet of code performing this right now is below:

[image]

Currently what I have set the visual preset to 'Near' and then set min and max distance with the colorizer. The output from that is shown below: 

[image]

I'm going to be using OpenCV to identify that shape in the middle. My Realsense Viewer output looks pretty similar to the Python one other than the particular colors are different.

My question is, is there a compatible setting or another way to group depth values by change over pixels by intensity? For example, in the output my surface may only change the 10-15mm range of my min and max over the entire image (see blow) where as the object I want to highlight might change that range in a much smaller "time-frame" or pixel count/ area.

[image]

My hope is to be able to sperate the long-span depth changes with the sharp/ short-span depth changes.

MartyG-RealSense, 2021-04-22:
Hi @PlasticBlaze  An OpenCV discussion in the link below suggests measuring pixel brightness changes of time using  mean and standard deviation with an OpenCV function such as **meanStdDev** and specifying a particular area of the image to focus on with **frame.operator()(rect)**

https://answers.opencv.org/question/68988/how-to-find-brightness-of-given-area/

PlasticBlaze, 2021-04-23:
Thanks Marty, I will look into that! Is there a way to range the depth colorization dynamically? Instead of setting a min and max depth prior, the camera can change it based on the nearest and farthest depth value it finds?

MartyG-RealSense, 2021-04-23:
There is a colorization option called **Histogram Equalization** that has the following information in the SDK documentation.

**************

Histogram equalization makes the color changes proportionally within the available valid pixel value range.  

Disabling the feature will cause the color representation changes based on the min / max distance configuration.  For example, if the depth image contains valid depth pixels value ranging from 45cm ~ 3m, the color will spread proportionally within the range. 
 When disabled, the color will spread base on min / max distance settings.

***********

Python code for controlling the status of Histogram Equalization with **rs.option.histogram_equalization_enabled** can be found in the link below.

https://github.com/IntelRealSense/librealsense/issues/7089#issuecomment-672105974

MartyG-RealSense, 2021-04-29:
Hi @PlasticBlaze  Do you require further assistance with this case, please?  Thanks!

PlasticBlaze, 2021-04-29:
No I'm fine. Thanks.

closed 2021-04-29 by PlasticBlaze