realsenseai/librealsense #8878: UE4 Wrapper Crashing When Opening Blueprints with T265

LukeMTheatrical, 2021-04-21:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 } | 
| Firmware Version                   | (0.2.0.951 | 
| Operating System & Version |   {Win (10)   |   
| Platform                                 | PC  |
| SDK Version                           |  { 2.44.0 / 2.43.0 / 2.42.0 }    | 
| Language                               |  {Unreal }  | 
| Segment			         |  {VR/AR }  | 

### Issue Description
Hello there, I've been running into an issue with the Unreal Engine Wrapper. The issue being anytime I click on or try to work with the RealSenseInspector Blue Print, UE4 Crashes but only with the T265. I have a D435 and when that is plugged in the wrapper works correctly. 
: Things I have tried
- Double check that I am following the documentation exactly (I am, the UE sample project opens just fine and can confirm that the plugins are correctly showing up)
- Attempted different versions of the SDK (Same crash with the three versions listed above)
- Recompile the Unreal Engine from the official github (no change)
- Change the the location of the realsense2.dll from the UE engine binaries to directly in the wrappers plugin binaries (no change, still crashes)
- Have D435 and T265 Plugged in at same time. (Crashed, even when I start with the D435 and have RealSenseInspector running, hot plug in the T265, refresh devices... and crash)
- Any help or advice would be appreciated, thank you.

factify, 2021-07-26:
Hi @LukeMTheatrical 

I'm having a similar problem with the Unreal Wrapper. I looked into this a little closer and it appears that when the T256 is connected (either on its own or with another camera) there is effectively a nullptr returned on the device. To replicate this:

1, Connect the camera before loading the example TestLevel of the Unreal Wrapper
2, Now in the World Outliner simply select the BP_RealSenseInspector

This will cause Unreal to crash. However, if you connect the camera after you selected the RealSenseInspector then Unreal would not crash and you are able to "refresh" the Hardware Inspector and the camera would show there. This would not be much help to you as you would still need to modify some of the codebase to get the appropriate data out of the T256 and use it for your tracking needs.

To start with, you'll have to extend the RealSenseInspector.cpp - ARealSenseInspector::GetDeviceDefaultProfileConfig() to include the T256 and handle it's profile data type (STREAM_POSE and FORMAT_6DOF) which also means you'll need to extend the RealSenseTypes.h which currently only covers a selected number of profiles.  

You'll also have to think about modifying the ARealSenseInspector::Start() to cater for multiple input streams if you have more than 1 cameras connected and make sure that when you enable_stream on the T256 you enable the appropriate data i.e.: enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);

Having said all of this I'm puzzled by why am still getting a nullptr on the T256 device. So if there is anyone who solved this problem before could reply, it would be much appreciated.

factify, 2021-07-27:
Managed to dig a little deeper again today and I found that when URealSenseContext::QueryDevices() gets called the rs2_create_device(List, i, &e); returns a none valid rs2_device* RsDevice (only when the T256 is connected). Which means it's not getting added to the Devices array. 

Is there anyone out there who knows about this? Appears as the call to rs2_create_device() requires an additional step that is missing for the tracker camera but I'm finding it hard to spot.

closed 2026-10-23 by MartyG-RealSense