realsenseai/librealsense #8879: Accuracy of 3D Points

snopytas, 2021-04-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i  | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.4                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.43                         | 
| Language                            |  {C++ }                          | 

Hi, 

we are throwing a ball (diameter ~5cm) from a distance of 4-5m in the direction of the camera and try to predict its trajectory.

To do this we align the depth to the color frame. Then we measure the center of the object in the color frame, extract the depth at this point (if available, otherwise we take the closest available depth) and calculate its 3D coordinates with rs2_deproject_pixel_to_point(...) using the provided camera intrinsics. The resolution is 848x480@60 for both depth and color frame.

It works, but not as good as desired. Could it be possible that the measured depth and hence the result of rs2_deproject_pixel_to_point() is inaccurate when the distance is > 2.5 meters? I have the feeling that all points with a depth < 2.5m are measured accurate but points with a depth of >2.5m are inaccurate. Could it help if we use a larger object or doesn't this affect the accuracy? 

If my assumptions should be wrong than I have to revise our detection algorithm. But if my assumptions are correct, what could help to improve? 

Thank you very much

MartyG-RealSense, 2021-04-22:
Hi @snopytas  On 400 Series stereo depth cameras, error increases linearly as distance from the camera increases. This is a phenomenon called **RMS error**.  On the D415 and D435 / D435i models, 3 meters and beyond is the approximate distance at which drift in accuracy of the measured values starts to becone noticable.  So this is consistent with your own observations.

The D455 model has 2x the accuracy over distance of the earlier 400 Series models, meaning that it has the same accuracy at 6 meters that the D415 and D435(i) models do at 3 meters.

When tracking a small moving object, the advice in the link below is that the object should take up at least 9 pixels on the image.  So the ball that you are using is likely large enough already.

https://github.com/IntelRealSense/librealsense/issues/4175#issuecomment-507448389

As mentioned in the link above, using a high FPS speed is desirable for an application such as this.  You could try 90 FPS to see how it affects your results.  If the ball travels along a predictable path (e.g thrown straight ahead) then you could also contemplate using the 300 FPS **High Speed Capture** mode supported by the D435(i) models.  This restricts the field of view to a central strip of the screen 848x100 in size (full-screen width and a central letterbox in height).

https://dev.intelrealsense.com/docs/high-speed-capture-mode-of-intel-realsense-depth-camera-d435

[image]

snopytas, 2021-04-23:
@MartyG-RealSense  Thank you very much! Indeed we've tried 90FPS for the depth stream but there is no really improvement. From the improvement section you mentioned I get that we could try some post processing and play a bit with the settings. Or we need a D455 maybe. Do you have any other considerations to increase the accuracy of the depth stream for distances > 3m?

MartyG-RealSense, 2021-04-23:
If the scene that the camera is observing is well lit then you could try turning off the laser to improve RMS error.  This can be achieved in C++ using an instruction similar to the one below:

**depth_sensor.set_option(RS2_OPTION_EMITTER_ENABLED, 0.f); // Disable emitter**

https://github.com/IntelRealSense/librealsense/wiki/API-How-To#controlling-the-laser

[image]

You can also test the observed scene with the laser off in the RealSense Viewer program by using the drop-down menu beside the **Emitter Enabled** option to set it to 'Off'.

[image]

MartyG-RealSense, 2021-04-29:
Hi @snopytas  Do you require further assistance with this case, please?  Thanks!

snopytas, 2021-04-29:
I think I have all information that I need. Thank you very much!

MartyG-RealSense, 2021-04-29:
Thanks very much @snopytas for the update   :)

snopytas, 2021-05-03:
Hi again @MartyG-RealSense , 

unfortunately turning off the emitter didn't help. We noticed that often the inaccuracy occurs even in small distances. So, the measurments of the ball positions are quite noisy. 

Could it be possible that a cluttered environments lead to bad results for the depth measurement, respectively is the environment crucial? In the 3D view of the realsense-viewer the ball often isn't visible at all or one can see just a little piece of it. I don't know how to continue investigating. Is the D435 suited for tracking the position of a thrown tennis ball / object? 

Thank you very much

MartyG-RealSense, 2021-05-04:
Hi @snopytas   I would say that the trackability of the ball by the D435 would depend on how fast the ball is thrown.  The D435 / D435i camera models are suited for tracking a thrown tennis ball / object using depth, as they have a fast 'global' shutter on the depth sensor that enables objects as fast as a moving car on a road to be captured.  However, a baseball can be thrown by a pitcher as fast as 100 miles per hour.

There is also the complication that RGB color is being used too, and the RGB sensor on the D435 / D435i has a slower 'rolling' shutter that is susceptible to blurring when capturing fast motion.  The D455 camera model has a fast global shutter on both the depth and RGB sensors.  

If you are using auto-exposure then you could try disabling an RGB sensor setting called **Auto-Exposure Priority**.  Doing so enforces a constant FPS speed instead of allowing the FPS to vary.  If you are using C++ language, the code in the link below shows how to disable Auto-Exposure Priority, taking care to ensure that the command is directed at the RGB sensor (index '1') instead of the depth sensor (index '0').

https://github.com/IntelRealSense/librealsense/issues/5290

You can also test it in the RealSense Viewer program by going to the RGB section of the options side-panel and disabling **Auto-Exposure Priority** by left-clicking on the blue box (On) beside it to toggle it to black (Off).

A cluttered background would not necessarily result in poorer results, especially if you are using depth to color alignment which makes it easier to distinguish foreground pixels from background pixels.  There is the potential for the depth sensing to become confused though if it observes repeating patterns in the scene though, such as a row of fence posts or window frames (horizontal repeating pattern) or the slats of a window blind (vertical repeating pattern).  Do you have anything resembling such a repeating pattern in the scene that your camera is observing, please?

If the tracked object is a distinct color, such as tennis-ball yellow, then an alternate approach may be to track it by its **color**.  A project in the link below uses RealSense D435, ROS and OpenCV to detect an object of a particular color (a cube) and track their pose.

https://by-the-w3i.github.io/2019/10/06/ColorBlockTracking/ 

Regarding poor visibility of the ball in the RealSense Viewer, could you check please in the **Post-Processing** section of the options side-panel whether a filter called **Threshold Filter** is enabled?  When this flter is enabled, it limits the range of depth that is rendered to 4 meters by default.  If it is enabled, left-clicking on the blue icon beside it to turn it to red (Off) should enable the Viewer to render the full distance of depth data that is visible to the camera.

[image]

agrunnet, 2021-05-04:
Please also read the white paper on high speed capture with frame rates up to 300fps if you have not already. It summarizes a bunch of observations.

snopytas, 2021-05-04:
Thank you @MartyG-RealSense . The threshold filter is disabled.  I will play with the exposure settings when I am in the lab next time. In the picture below there is a repeating pattern at the ceiling. I will try to find a better environment

We already detect the ball by its color. Then we compute its center of gravity and extract the depth at (or near) this point . 

I have made some screenshots (848x480@90 for the depth and 848x480@60 for the color frame)

[image]

As you can see in the left picture the color detection suffers from motion blur but I think that is ok. 
But in the middle (depth stream) and on the right (3D View of realsense-viewer) one can see that the ball is smeared or distorted. This is the case in most of the frames. Therefore the depth extracted at the center of gravity (obtained from the color frame) is probably very inaccurate. 

In the next picture the ball (blue) has no holes but it is deformed. The smallest depth of the ball here is 1.17m and the biggest is 1.27m. But the ball itself has a diameter of 0.06m.  
[image]

All in all the depth is inaccurate which probably leads to inaccurate results when computing the deprojection. And therefore the measured trajectory is unclean.

snopytas, 2021-05-04:
@agrunnet Ok, I will read the whitepaper, thanks.

MartyG-RealSense, 2021-05-04:
@snopytas  Given that the ball seems to be leaving a trail, you could check whether the **Temporal Filter** in the list of post-processing filters is enabled, and disable it if it is.  

An example of a trail being left by walking-pace motion on the depth stream is shown in the link below.

https://github.com/IntelRealSense/librealsense/issues/7445

snopytas, 2021-05-05:
You're right. But also without trails the accuracy isn't that good. The Ball often isn't fully visible in the depth frame and the 3D view.

But I think I come to the conclusion that I could improve the situation but there will always be some inaccuracy or noise. Probably our expectation of a nearly perfect position measurement is too high and when we calculate our predicted trajectory we just need to consider this kind of noise (e.g. drift of 5cm in the depth or same depth in two or three consecutive frames) . Is this perspective maybe the better approach instead of looking for the perfect position measurment?

snopytas, 2021-05-05:
@MartyG-RealSense You're right. But also without trails the accuracy isn't that good. The Ball often isn't fully visible in the depth frame and in the 3D view.

But I think I come to the conclusion that I could improve the situation but there will always be some inaccuracy or noise. Probably our expectation of a nearly perfect position measurement is too high and when we calculate our predicted trajectory we just need to consider this kind of noise (e.g. drift of 5cm in the depth or same depth in two or three consecutive frames) . Is this perspective maybe the better approach instead of looking for the perfect position measurment?

MartyG-RealSense, 2021-05-05:
If accuracy is low without the Temporal Filter enabled to generate a trail, you could try enabling the filter but increasing the **Filter Smooth Alpha** slider value to reduce update delay whilst still having a bit of trail present.

[image]

You could also try applying the **Medium Density** preset, which provides a balance between accuracy and a good amount of detail on the image.

[image]

A perspective of realism in evaluations of results, rather than aiming for perfection, is a good approach to take given the number of real-time variables that can affect a scene, such as lighting and environment.  Consistent accuracy and stability of results is desirable and important though, of course.

snopytas, 2021-06-03:
Hi @MartyG-RealSense 

I found out why the trajectory predictions with the least square method were so bad. The fundamental problem was that I didn't considered the inclination of the camera. For example the depth-axis of the camera does not correspond to the "real world" depth-axis. After a coordinate transformation the predictions are much more better. So it was my mistake. I don't know why I missed such a basic thing. 

Anyway, I will consider your suggestions to improve the depth quality because the are of course still variations with larger distances. But the fundamental problem is solved. Thank you very much.

MartyG-RealSense, 2021-06-03:
Thanks very much @snopytas for the update - it is great to hear that you found the solution!