realsenseai/librealsense #8880: IR frames occasionally corrupt until hardware reset

SirDifferential, 2021-04-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415, D435, D455 } | 
| Firmware Version                   | 5.12.9.0 and 5.12.12.100 | 
| Operating System & Version |  Windows 10, Raspbian 10  | 
| Kernel Version (Linux Only)    |  5.4.83-v7l+                                         | 
| Platform                                 | x86 laptop, Raspberry Pi 4  |
| SDK Version                            |   2.36 and 2.43  }                          | 
| Language                            |  C/C++ }                          | 
| Segment			|  Others                          | 

### Issue Description

In #5209, we were told to open a new issue if the problem happens again.

On D415, D435, and D455, we occasionally see the IR frames getting corrupted, which produces a depth frame with mostly invalid depth pixels. Please see the reference images for the normal, expected operation of the sensor:

[image]
[image]
[image]

and the following images showing the problem case:

[image]
[image]
[image]

Unfortunately, due to software error by me, only one IR image was saved on disk. Note that in the error case the RGB image turns black as well.

This problem has happened with Realsense SDK versions 2.36 and 2.43 on Raspberry Pi 4 and Windows 10 (x86) using Newnex USB cables (part number: U3SA01C12-050 recommended by Intel) with firmware versions 5.12.9.0 and 5.12.12.100. Official Realsense cables are yet to be tested at length, but we do see at least some type of invalid depth frames with the devices installed at our customer locations using the official Realsense USB cable. As of yet we don't know what the IR frames look like in those scenarios, but I'll try to include some of those in this issue when I get a chance to debug one.

When the problem happens, there are two solutions for fixing it:

* Unplugging the cable
* issuing `rs2_hardware_reset`. Does not always work with Raspberry 4 using the RSUSB backend.

Restarting the program and re-initializing the librealsense2 context is not enough to fix the issue. We also plan to test using [uhubctl](https://github.com/mvp/uhubctl) on RPI4 to disable power delivery to the USB ports to see if that's enough to reset the Realsense.

The problem has happened both at night when our office is empty and devoid of light, and at day when we have people working with the lights on. No strong IR sources exist in the office, execpting perhaps the metal door that reflects some IR light back towards the sensors (it appears black in the depth frame, as seen in the normal pictures above).

We can reproduce this issue by restarting our test program approximately once every minute, and then wait until we receive 100 frames in row with average depth frame "blackness" percentage larger than 85%. In that case we save depth, IR, and RGB images on disk and consider the bug that have been reproduced and make a note in our debugging folder. Our next step is to see if this issue also happens with the sensor running for several days with no program rebooting.

In the previous ticket where we discussed black depth frames, the leading theories involved Realsense temperature going too high as well as external strong IR pulses hitting the IR sensor. Neither of those events happens in our office which is air conditioned and has no sunlight.

MartyG-RealSense, 2021-04-22:
Hi @SirDifferential   As you are considering methods for resetting your Pi, a good starting point for investigating this new case may be to look at the case in the link below.  In that case, a RealSense user who had hundreds of 400 Series cameras deployed at remote locations with Pi as the computing-board used dealt with freezes by using a custom Power Over Ethernet (PoE) HAT solution to reset the Pi when the Pi was unresponsive after a certain period of time.

https://github.com/IntelRealSense/librealsense/issues/8088

SirDifferential, 2021-04-22:
We do use the RPI4 POE hats, but don't have the ability to perform that type of power cycling. However, the RPI4 bootloader has a flag called [USB_MSD_PWR_OFF_TIME](https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md) which can toggle USB power on boot and we have started using that on our devices.

Yet, we would like to understand what causes these problems and how to stop them from happening in the first place. Power cycling the device every time this happens seems like a bit extreme solution.

MartyG-RealSense, 2021-04-22:
Comparing the good and bad infrared images, it looks as though the IR emitter dot pattern projection may be absent in the bad image.  If the emitter pattern disappears, or the entire projector has turned off, this could account for the bad depth image that has little detail.

Since the emitter is able to function, you could try setting **RS2_OPTION_EMITTER_ALWAYS_ON** to true if you are not doing so already.

https://github.com/IntelRealSense/librealsense/issues/6247

SirDifferential, 2021-04-22:
Good to know about `RS2_OPTION_EMITTER_ALWAYS_ON` - I'll try using that. We load custom settings on the Realsense, and they do include the laser as enabled. Here are the full settings for D415/D435 (D455 uses different settings):

```
{
    "aux-param-autoexposure-setpoint": "1536",
    "aux-param-colorcorrection1": "0.298828",
    "aux-param-colorcorrection10": "0",
    "aux-param-colorcorrection11": "0",
    "aux-param-colorcorrection12": "0",
    "aux-param-colorcorrection2": "0.293945",
    "aux-param-colorcorrection3": "0.293945",
    "aux-param-colorcorrection4": "0.114258",
    "aux-param-colorcorrection5": "0",
    "aux-param-colorcorrection6": "0",
    "aux-param-colorcorrection7": "0",
    "aux-param-colorcorrection8": "0",
    "aux-param-colorcorrection9": "0",
    "aux-param-depthclampmax": "65536",
    "aux-param-depthclampmin": "0",
    "aux-param-disparityshift": "0",
    "controls-autoexposure-auto": "True",
    "controls-autoexposure-manual": "8500",
    "controls-color-autoexposure-auto": "True",
    "controls-color-autoexposure-manual": "156",
    "controls-color-backlight-compensation": "0",
    "controls-color-brightness": "0",
    "controls-color-contrast": "50",
    "controls-color-gain": "64",
    "controls-color-gamma": "300",
    "controls-color-hue": "0",
    "controls-color-power-line-frequency": "3",
    "controls-color-saturation": "64",
    "controls-color-sharpness": "50",
    "controls-color-white-balance-auto": "True",
    "controls-color-white-balance-manual": "4600",
    "controls-depth-gain": "16",
    "controls-laserpower": "360",
    "controls-laserstate": "on",
    "ignoreSAD": "0",
    "param-amplitude-factor": "0",
    "param-autoexposure-setpoint": "1536",
    "param-censusenablereg-udiameter": "9",
    "param-censusenablereg-vdiameter": "9",
    "param-censususize": "9",
    "param-censusvsize": "9",
    "param-depthclampmax": "65536",
    "param-depthclampmin": "0",
    "param-depthunits": "1000",
    "param-disableraucolor": "0",
    "param-disablesadcolor": "0",
    "param-disablesadnormalize": "0",
    "param-disablesloleftcolor": "0",
    "param-disableslorightcolor": "1",
    "param-disparitymode": "0",
    "param-disparityshift": "0",
    "param-lambdaad": "751",
    "param-lambdacensus": "6",
    "param-leftrightthreshold": "10",
    "param-maxscorethreshb": "2893",
    "param-medianthreshold": "796",
    "param-minscorethresha": "4",
    "param-neighborthresh": "701",
    "param-raumine": "6",
    "param-rauminn": "3",
    "param-rauminnssum": "7",
    "param-raumins": "2",
    "param-rauminw": "2",
    "param-rauminwesum": "12",
    "param-regioncolorthresholdb": "0.785714",
    "param-regioncolorthresholdg": "0.565558",
    "param-regioncolorthresholdr": "0.985323",
    "param-regionshrinku": "3",
    "param-regionshrinkv": "0",
    "param-robbinsmonrodecrement": "25",
    "param-robbinsmonroincrement": "2",
    "param-rsmdiffthreshold": "1.65625",
    "param-rsmrauslodiffthreshold": "0.71875",
    "param-rsmremovethreshold": "0.809524",
    "param-scanlineedgetaub": "13",
    "param-scanlineedgetaug": "15",
    "param-scanlineedgetaur": "30",
    "param-scanlinep1": "155",
    "param-scanlinep1onediscon": "160",
    "param-scanlinep1twodiscon": "59",
    "param-scanlinep2": "190",
    "param-scanlinep2onediscon": "507",
    "param-scanlinep2twodiscon": "493",
    "param-secondpeakdelta": "575",
    "param-texturecountthresh": "0",
    "param-texturedifferencethresh": "1722",
    "param-usersm": "1",
    "param-zunits": "1000",
    "stream-depth-format": "Z16",
    "stream-fps": "30",
    "stream-height": "720",
    "stream-width": "1280"
}
```

Are you sure about the emitter? I did a test with the Realsense viewer and enabled/disabled the emitter. The following images show the emitter patterns appearing at my colleague's head and at the metal door frame:

[image]

[image]

And those same patterns can be seen in the corrupted frame:

[image]

Just as an idea, to me the corruption on the IR frame looks like invalid stride used when parsing the frame, as though the format changed mid-stream (ie, parsing the image using Y16 instead of Y8). I've accidentally used invalid strides when programming image processing things before and the results often looked similar to the corrupted frame here.

MartyG-RealSense, 2021-04-22:
There is a phenomenon where the visibility of the dot pattern can change depending on the lighting levels and the type of material that the pattern is projected onto.  For example, in medium lighting conditions the pattern may be visible on some objects but not others.  Then as the light level dims as the day progresses, the situation can reverse and the objects that had an invisible pattern now have visible dots and the objects that had a visible pattern now have no dots.  

That may not be the case here though, as your good infrared image shows that when the camera is operating normally the room is uniformly covered in dots.

MartyG-RealSense, 2021-04-28:
Hi @SirDifferential  Do you have an update that you can provide about this case, please?  Thanks!

SirDifferential, 2021-04-28:
I haven't yet had the time to repeat the test with the official Realsense cables, but the issue stated above remains open with the Newnex cables, which Intel recommends as an option.

MartyG-RealSense, 2021-04-28:
I have not heard of past cases where the IR format changed from Y8 to Y16 mid-stream.

I recall your previous case https://github.com/IntelRealSense/librealsense/issues/6952 in which options were sometimes failing when being set, and it was thought that the timing on an RSUSB build of librealsense compared to a kernel-patched build might be causing the problem.

https://github.com/IntelRealSense/librealsense/issues/6952

Are you setting any options in this particular case that might result in a timing error if the sensor was not ready when making the activation call?

SirDifferential, 2021-04-29:
Yes, we set plenty of options at startup. The process is about this

* Enable advanced mode
* Load the above preset
* Enable streams
* Start pipeline
* We set `RS2_OPTION_AUTO_EXPOSURE_PRIORITY`, `RS2_OPTION_GAIN`, and, `RS2_OPTION_ENABLE_AUTO_EXPOSURE`
* Later, if needed, we manually control color camera gain and exposure

No errors / exceptions are thrown. The issue in #6952 remains open, but doesn't have with these particular options.

I wondered about if this was an issue with RSUSB myself, but it also happens on Windows, which does not use RSUSB. A supported kernel with patched UVC is not yet tested.

MartyG-RealSense, 2021-04-29:
Can you confirm whether there is any use of hardware sync cabling with the camera please?

SirDifferential, 2021-04-30:
No, no sync cables are used.

MartyG-RealSense, 2021-04-30:
I think that the RGB going black is an important indicator in this case, as it suggests a problem that goes beyond just the IR imagers alone being affected (since depth and infrared come from the same sensor).  And if the laser was being affected then the RGB image should not be affected by that.

During further tests, I found that I could replicate your broken images if both depth exposure and RGB exposure dropped to a minimum of '1' (causing a broken depth image and entirely black RGB image).  Resetting the camera or performing an unplug-replug returned the camera configuration to the default exposure values to restore the images.

SirDifferential, 2021-04-30:
Nice findings!

We're using USB3 ports on both RPI4 and Windows 10.

MartyG-RealSense, 2021-04-30:
I note from the opening paragraph of the case linked to above that the RealSense user in that case also experienced the problem in an indoor room without sunlight.  "If placed in reduced light, such as a closet away from any direct sun exposure, it is unable to update the exposure and gain to proper settings" (after exposure has dropped to '1').

This raises the question of why the image is not auto-correcting.  This would require auto-exposure to be enabled.  So perhaps dropping the exposure to '1' also turns off auto-exposure in the process and leaves the camera fixed on the manual epxosure value of '1' until the camera is reset.

Perhaps a check could be put into your application that if RGB or depth exposure value = 1 then reset the camera, and keep trying until exposure is no longer '1' (to take account of the fact that the reset instruction might have to be requested more than once before it is heard due to RSUSB's timing).

MartyG-RealSense, 2021-05-06:
Hi @SirDifferential  Do you have an update about this case that you can provide, please?  Thanks!

SirDifferential, 2021-05-10:
I haven't yet had the time available for investigating this further. Please keep this issue open until I can provide the following results:

* What both IR feeds looks like when the error happens
* If official realsense cables reproduce the issue
* What the exposure values are when the problem happens, if they work over RSUSB
* If `RS2_OPTION_EMITTER_ALWAYS_ON` changes something

MartyG-RealSense, 2021-05-10:
No problem to keep the case open for a further time period whilst you investigate the above issues  Thanks for the update!

SirDifferential, 2021-05-14:
I started a new test on 4 Raspberrys: 1x D415, 3x D435. I did not yet enable `RS2_OPTION_EMITTER_ALWAYS_ON`. All sensors are now using official cables, and the missing IR feed is now saved when there's a corrupted frame. I noticed the RGB frame being black can be eliminated as a problem - we were reading the color frames at different FPS than depth frames, and the problems often happened when we didn't have color frames handy.

I'll report back some time next week on the findings.

puzzlepaint, 2021-05-14:
I would like to note that the same kind of IR image corruption has been reported previously (I also observed it a few times in the past, but haven't worked with the cameras much lately). Two older threads describing it which are known to me are:
* https://github.com/IntelRealSense/librealsense/issues/7195
* https://support.intelrealsense.com/hc/en-us/community/posts/360033503254-Infrared-camera-image-is-damaged

In the first one of these two, the issue was not observed by the user who reported it anymore using SDK version 2.44 and camera firmware version 05.12.12.100. Thus, perhaps this combination of versions might be worth a try.

In general, to me this issue looks like a bug in either librealsense or the camera firmware that has already occurred to various people. Thus it seems very unlikely to me that changing emitter or exposure settings will help anything.

MartyG-RealSense, 2021-05-20:
Thanks very much @puzzlepaint for your advice to @SirDifferential 

@SirDifferential  Do you have test findings available to report yet, please?  Thanks!

SirDifferential, 2021-05-20:
I've been running tests since last friday with official cables without constant program rebooting, and haven't seen the issue yet. I'm going to now switch the tests to use reboots every 2 minutes to see if that reproduces the issue. However, the above issues appear to be the same problem.

MartyG-RealSense, 2021-05-20:
Okay @SirDifferential  Thanks very much.  I look forward to your next test results from the reboot tests.  Good luck!

MartyG-RealSense, 2021-05-31:
Hi @SirDifferential  Have you experienced the problem involved in this case recently, please?  Thanks!

SirDifferential, 2021-06-02:
None of my current 4 Realsenses with official cables have reproduced the issue, with a program reboot every 2 mins or with several weeks of stable running. From this test it looks like the issue, in this case, is with the Newnex cables, althouh I again stress @puzzlepaint 's comment above with the links to similar issues having been noticed by other people.

I doubt debugging Newnex cables is the purpose of this github repo, so I suppose we can close this issue and continue the discussion in the existing, open issues.