realsenseai/librealsense #8881: SR300  DCM

L-xn, 2021-04-22:
The DCM of SR300 can't be installed, please ask for advice from seniors!
[image]

MartyG-RealSense, 2021-04-22:
Hi @L-xn  In 2017Intel introduced the ability for the SR300 drivers to be automatically installed by Windows over an internet connection simply by plugging the camera into the USB port of the computer.  This was intended as a replacement for the DCM installer program.

Could you access the Windows **Device Manager** please and check in the **Cameras** and **Imaging Devices** categories for three drivers (Depth, RGB and Virtual Driver).  Are these three drivers present in the Device Manager when your SR300 is plugged in, please?

[image]

L-xn, 2021-04-22:
> Hi @L-xn In 2017 Intel introduced the ability for the SR300 drivers to be automatically installed by Windows over an internet connection simply by plugging the camera into the USB port of the [...]

NO
[image]

MartyG-RealSense, 2021-04-22:
Have you got the camera plugged into a USB 3 port?  SR300 is USB 3 only and does not support USB 2.

L-xn, 2021-04-22:
> Have you got the camera plugged into a USB 3 port when you run the DCM program? SR300 is USB 3 only and does not support USB 2.

I use USB3.0. This device requires the computer processor to be i6, but mine is i5. I don't know if this is the reason!

MartyG-RealSense, 2021-04-22:
It is not the 'i' number that matters but the first number of the processor's name, which indicates which **architecture generation** the processor is.  For example, if your processor was an ' i5-7360U' then the '7' would indicate that the processor is 7th Generation.

For the majority of computers, a 6th generation Skylake processor or newer was required to use an SR300 with the old Windows SDK that the DCM program was part of.  There was a 50% chance that the camera might work with a slightly older generation 'enterprise' processor such as a 5th generation Intel Xeon CPU.  In most cases though, using an SR300 with that old SDK on a computer with a processor older than 6th generation means that it will not work.

L-xn, 2021-04-23:
> It is not the 'i' number that matters but the first number of the processor's name, which indicates which **architecture generation** the processor is. For example, if your processor was an ' [...]

Thanks for your quick reply! Thank you very much!I know.

closed 2021-05-05 by MartyG-RealSense