realsenseai/librealsense #8885: configure pipeline with callback stops receiving frames...

drhalftone, 2021-04-22:
I don't think this is a bug by any means, but without any documentation or examples, I don't know where else to turn but post it as an issue.  I'm still trying to grab video from multiple sensors using the C API.  I thought I had was that using a synchronous get frame may block the RS2 library from reading the other sensors.  So I tried using the callback pipeline instead.  I am getting frames as I see debug messages printing out; however, after about 10 or so frames, the callbacks stop coming. From the outside, this looks like the driver has a fixed buffer for receiving frames and it is filling up.  However again, without documented examples, I don't know how to clear the frames that are supplied to the callback.

drhalftone, 2021-04-22:
Ok, I see that I have to release the frame myself in order to free up the buffer for future frames.  Here is my callback:

```
void LAURealSenseFrameCallback(rs2_frame *frame, void *packet)
{
    qDebug() << "LAURealSenseFrameCallback(rs2_frame *frame, void *packet)" << ((LAURealSenseCamera::CameraPacket*)packet)->serialString;
    rs2_release_frame(frame);
}

```
Of course getting back to multiframe, I see that once the second camera starts streaming, the first camera stops.  So I'm still missing something in the C API as it relates to multiple cameras.

RealSenseSupport, 2021-04-27:
@drhalftone Is this issue the same as you mentioned in https://github.com/IntelRealSense/librealsense/issues/8864? Did you resolve this? Looking forward to your update. Thanks!

drhalftone, 2021-04-27:
Hold tight.  I have that code on another computer.  When I get to my office, I'll test it.

RealSenseSupport, 2021-05-12:
@drhalftone Any update from your side? Thanks!

RealSenseSupport, 2021-05-19:
@drhalftone Could you please update? Please note that the issue will be closed if we don't hear from you for another 7 days. Thanks!

RealSenseSupport, 2021-05-25:
@drhalftone Sorry that we don't hear from you for weeks. Close the ticket at this point. Please feel free to create another new one if you still have questions or issues. Thanks!

closed 2021-05-25 by RealSenseSupport