realsenseai/librealsense #8886: point cloud and RGB alignment off in Labview

johnras, 2021-04-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.09.00 | 
| Operating System & Version |   Win10 | 
| Kernel Version (Linux Only)    |  NA                            | 
| Platform                                 | PC  |
| SDK Version                            |  Viewer v2.40.0; Labview 2.33 (with 2.40.0 dll)                         | 
| Language                            |  labview                          | 
| Segment			|  {others }                          | 

### Issue Description
When I am acquiring data with LabView (example) my RGB image (texture) does not line up with the point cloud. (see 
[image]

However I do not have this issue with the viewer. (see 
[image]

How would I go about fixing the alignment in Labview?  

Thanks for any help you can provide!

MartyG-RealSense, 2021-04-23:
Hi @johnras   The LabVIEW textured point cloud does not look too bad to me.  It is worth bearing in mind that the RealSense Viewer applies a range of post-processing filters by default, so an image in the Viewer may have differences to an image generated in another RealSense program.

Could you try the **Hello World: Point Cloud Plus** LabVIEW example that provides post-processing options please to see whether the resulting image is closer to that of the Viewer's? 

https://dev.intelrealsense.com/docs/labview-wrapper#section-hello-world-point-cloud-plus

johnras, 2021-04-23:
Thanks for the response MartyG.  The point cloud itself looks fine, particularly since the only region I'm really interested in for my application will be in the area with the red box.  The Hello World: Point Cloud Plus example looks better on the point cloud with the post processing and the texture overlay from the depth camera is perfect (since it is the same camera), but it doesn't allow me to overlay the RGB image.  I can eventually combine the post processing to the ext RGB example, but I'm not sure that will actually fix the alignment issue with the ext RGB image.  Essentially, what I am trying to figure out is how to adjust the overlay of the RGB texture in labview so that it is actually on the box, not split between the box and the wall in the point cloud.  

In the near future, I will be integrating the D455 into a separate imaging system (which runs on labview) so that I can overlay an image acquired from that device onto the point cloud generated from the D455.  I've been operating under the assumption that the texture is overlaid onto the point cloud based on the intrinsic and extrinsic parameters of the cameras and so it would be possible to use my external image as the texture if I can properly calibrate the two systems together.

agrunnet, 2021-04-23:
Correct. The current labview example does not map the third color imager to the depth map. It does allow you to use the left color image though as this is taken from the exact same viewpoint as the depth (so no additional occlusions or calculations needed to map that color to depth). 
To improve the color for that left imager you could consider using the laser on/off toggle feature and sort images received according to metadata of laser on or off. 

But if you want the really good RGB image you will need to use the third RGB which is going through the ISP. However, as you note this has different intrinsics and extrinsics so you need to now undistort, rectify, and map the color to the depth (aka UV mapping). This can be done. There are currently c-code examples of this (also GPU accelerated) but not yet any Labview example for that.

johnras, 2021-04-23:
Thanks agrunnet.  Unfortunately for me, I'm am much more familiar with Labview than C. :)  Which c-code example would be my best bet to start off with?  Is there an equivalent example in python?

closed 2021-05-05 by MartyG-RealSense