realsenseai/librealsense #8887: finger tracking/Win 10

L-xn, 2021-04-23:
My undergraduate graduation project is to use an RGB-D camera as a finger motion recognition system. Measure the frequency and angle of finger movement to evaluate the patient's recovery. I would like to ask what kind of depth camera is suitable for the predecessors! I have tried realsense D415, but it doesn't work. I don't know which series of kinect is suitable?

MartyG-RealSense, 2021-04-23:
A D415 - although it has lesser motion tracking capability than a D435(i) or D455 - should be suitable for this application as although it involves observation of movement, the movement would not be very fast.  What was the particular problem that you experienced with D415, please?

In regard to a programming approach to the project, a Python conversion of the **Openpose** body point detection system in the link below may be a helpful foundation for you.  As well as recognizing 21 finger joints, its documentation states that it can calculate the angles of the fingers by drawing upon hand data stored in a numpy array.

https://github.com/StrongRay/Openpose-Hand-Detection 

If you can read the angle values from the numpy array then it should be straightforward to calculate the frequency of changes of angle over time.

MartyG-RealSense, 2021-04-29:
Hi @L-xn  Do you require further assistance with this case, please?  Thanks!

L-xn, 2021-04-29:
Thanks! I am using the first generation of kinect for windows to do it!

MartyG-RealSense, 2021-04-29:
Okay, thanks for the update.  We cannot provide advice for Kinect cameras.  If your project is working for you with a Kinect though then I am glad that you achieved a solution.

closed 2021-05-05 by MartyG-RealSense