realsenseai/librealsense #8892: D435I doesn't work

Kylemaycry, 2021-04-23:
| | |
|---|---|
|**librealsense**|2.43.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435I|
|**Serial Number**|838412070598|
|**Firmware Version**|05.12.12.100|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|903223052509|
|**Firmware Update Id**|903223052509|

Please provide a description of the problem

[image]
Camera cannot output RGB image and Depth image, when connected to the realsense viewer

MartyG-RealSense, 2021-04-23:
Hi @Kylemaycry   Can you check the camera privacy settings in Windows please to ensure that pernission for applications to access the camera is set to **On**.  Instructions for doing so are in the link below.

https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

Kylemaycry, 2021-04-23:
I connected it to another PC. It's working now.

closed 2021-04-23 by Kylemaycry