realsenseai/librealsense #8893: depth images convert to point cloud

xunyi-yin, 2021-04-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I have rgb images and depth images, how can i get the point cloud files with a python code?

MartyG-RealSense, 2021-04-25:
Hi @xunyi-yin   Are you aiming to create a point cloud from live depth and RGB data streamed from the camera, please?  Or are you using pre-made depth and color image files such as PNG format images?

Can you also confirm please whether you want to save the point cloud data as a **ply file**.  Thank you very much.

xunyi-yin, 2021-04-25:
Hi @MartyG-RealSense  I want to use pre-made PNG depth images only or PNG depth images and JPG color images to create point cloud. I think ply file is OK

MartyG-RealSense, 2021-04-25:
I looked carefully at what you want to achieve.  It may be difficult to implement though, with a number of problems involved.  The best reference available for doing so in Python is the recent discussion in the link below.

https://github.com/IntelRealSense/librealsense/issues/8756

xunyi-yin, 2021-04-27:
Hi @MartyG-RealSense  Do you know how to get the depth image and color image together? My device is RealSense D455. My system is Ubuntu16.04

MartyG-RealSense, 2021-04-27:
Could you confirm your meaning when you ask about "getting the depth image and color image together", please?  If you are aligning pre-made depth and color image files in Python, information about doing this was provided in https://github.com/IntelRealSense/librealsense/issues/8756

xunyi-yin, 2021-04-27:
I mean I have D455 camera and my system is ubuntu16.04. how can i get depth images. i ever used 'rosbag record' but i can not extract the depth images from the .bag files. I want to know if i can use python code to do this or other methods. I do not want to use c++ code

MartyG-RealSense, 2021-04-27:
If you are asking about extracting depth frames from a bag file and saving them as PNG images with Python then the script in the link below may suit your needs.

https://github.com/IntelRealSense/librealsense/issues/1887#issuecomment-397529590

If you wanted to play back depth frames from a bag but not save them as image files then the SDK's **read_bag_example** Python example program may be helpful.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/read_bag_example.py

Intel's Python tutorial **distance_to_object.py** demonstrates how to align depth and color data from a bag.

https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb

xunyi-yin, 2021-04-28:
Thank you! I have solved it

MartyG-RealSense, 2021-04-28:
Excellent news that you found a solution, @xunyi-yin   :)

closed 2021-05-03 by MartyG-RealSense