realsenseai/librealsense #8895: Create OpenGL resource in onResume() of Activity

yixun, 2021-04-25:
Hi,
For Java Capture example
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/capture/src/main/java/com/intel/realsense/capture/MainActivity.java

I saw the following code:

    Runnable mStreaming = new Runnable() {
        @Override
        public void run() {
            try {
                try(FrameSet frames = mPipeline.waitForFrames()) {
                    try(FrameSet processed = frames.applyFilter(mColorizer)) {
                        mGLSurfaceView.upload(processed);
                    }
                }
                mHandler.post(mStreaming);
            }
            catch (Exception e) {
                Log.e(TAG, "streaming, error: " + e.getMessage());
            }
        }
    };

I am wondering is the mHandler an OpenGL thread? Because  some OpenGL resource are created in mGLSurfaceView.upload(processed);

Thanks

Yixun

MartyG-RealSense, 2021-04-25:
Hi @yixun  I would say that whilst **mHandler** is not specifically an OpenGL function (it is a 'message hander'), in the case of the MainActivity file of the Android **capture** example it is primarily used in a block of code related to OpenGL processing.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/examples/capture/src/main/java/com/intel/realsense/capture/MainActivity.java#L152

yixun, 2021-04-26:
Hi @MartyG-RealSense Thanks for your quick reply. I know mHandler is a message handler and it should be associated with a thread. In this capture example, it should be associated with the main UI thread I think.  As I know the OpenGL resource can be only created in a thread associated with graphic context. In the capture example, since the OpenGL resource is created in the UI thread, does this mean the UI thread has been associated with a graphic context?

MartyG-RealSense, 2021-04-26:
This handler section of the **capture** example does certainly include an OpenGL graphics reference in the form of **mGLSurfaceView**.  

https://developer.android.com/reference/android/opengl/GLSurfaceView

I would suspect that the thread that it is being associated with in that code block is coming from the **frames** definition.

[image]

MartyG-RealSense, 2021-05-03:
Hi @yixun  Do you require further assistance with this case, please?  Thanks!

yixun, 2021-05-08:
As yous said the handler section of the capture example does certainly include an OpenGL graphics reference, but the streaming in which the OpenGL resource is created is running in the UI thread. I am not sure if we can create OpenGL resource in UI thread.

MartyG-RealSense, 2021-05-08:
I researched your question carefully but could not find any information.  As the question is being asked about the **capture** SDK Android example, which is known to work, I am guesing that you are asking because you wish to better understand how the program works and the program itself runs okay.  Is that correct please?  Thanks!

yixun, 2021-05-10:
Yes. You are right. The example works and I just want to know more details about how the OpenGL resources are created in UI thread.

MartyG-RealSense, 2021-05-14:
Hi @yixun  I wanted to update to let you know that I have not forgotten about your wish for more detals about how OpenGL resources are created in the UI thread.  Thanks very much for your patience and I will get back to you as soon as I am able to.

MartyG-RealSense, 2021-05-15:
Hi again @yixun  I conducted further research into your question.  As there is not much available information on the subject, I took the approach of examining the **mHandler** sections of the various different example programs provided in the Android wrapper to look for insights about the code structure.  

I identified the following common factors between the programs.  I apologize in advance for any obvious errors in my interpretation of the code.

******************

-  The mHandler for the **mGLSurfaceView** is contained within a public void called **run()**

-  Within the run() void, the frames are obtained from **mPipeline.waitForFrames()**

-  Some kind of action may be taken with the frames using a **try** instruction, such as applying a colorizer with ** try(FrameSet processed = frames.applyFilter(mColorizer))**.  

-  It looks as though the OpenGL image is updated using ** mGLSurfaceView.upload(processed);**

-  The graphics routine is concluded with the **mHandler.post(mStreaming);** instruction to tell the handler to execute the code.

yixun, 2021-05-17:
Hi @MartyG-RealSense  Many thanks for your efforts in answering my questions. I very appreciate you examining different examples and finding the same pattern.

closed 2021-05-17 by yixun

MartyG-RealSense, 2021-05-17:
You are very welcome @yixun - thanks for the update!