realsenseai/librealsense #8897: SR305/SR300 depth image incorrect with BUILD_WITH_CUDA=true

andrusza2, 2021-04-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR305 (SR300 also exposed to this bug) | 
| Firmware Version                   | 3.27.3.0  | 
| Operating System & Version |   Linux (Ubuntu 18) | 
| Kernel Version (Linux Only)    |  4.9.201                                       | 
| Platform                                 | NVIDIA Jetson (tested also on x86 PC)  |
| SDK Version                            |  2.44                          | 
| Language                            |  C++                      | 
| Segment			| Depth Camera                    | 

### Issue Description
As mentioned in /issues/6373 - depth camera is not working properly when librealsense is build with CUDA. There is a bug in `src/proc/depth-formats-converter.cpp` file, which causes the wrong data to be copied for build with CUDA.

The functions `unpack_z16_y8_from_sr300_inzi` and `unpack_z16_y16_from_sr300_inzi` use the source data pointer (`in`) - the first half of the data is bit-shifted and stored in `dest[1]`, the second half is directly copied to `dest[0]`. In the CPU version, the pointer is moved in a loop and after its execution it points to `in + count` - so the copying works as it should. In the CUDA version, the pointer is never moved, so bad data is copied to `dest[0]`, resulting in an incorrect depth image. Simply moving the pointer to the right place fixes the problem.

MartyG-RealSense, 2021-04-25:
Thanks very much @andrusza2 for your report of your own experience of this SR300 issue and your kind contribution of a PR solution to address the problem.  This case will be kept open for the duration that the PR is active.  Thanks again!

MartyG-RealSense, 2021-06-20:
Hi @andrusza2   As your kindly contributed PR fix for incorrect depth with CUDA was released in SDK **2.47.0**, I will close this case.  Thanks so much for your work in resolving this SR300 issue!

[image]

closed 2021-06-20 by MartyG-RealSense