realsenseai/librealsense #8898: Require json configure document for L515.

cdb0y511, 2021-04-26:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L500 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> 1.05.05) | 
| Operating System & Version |   {Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  (5.4)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.<4>.<4> }                          | 
| Language                            |  {C++}                          | 
| Segment			|  {Robot}                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hi, could you give me a full JSON configure document for l515?
I can not match it for rs_option.h, so some config confused me like the "Visual Preset".
Does the JSON "Visual Preset" equal to rs2_l500_visual_preset in the  rs_option.h? 
Could you give me a full document for JSON configure? @MartyG-RealSense 
Thanks

I only find a simple example from others like 
  {
    "Ambient Light": 1,
    "Apd Temperature": -9999,
    "Confidence Threshold": 1,
    "Depth Offset": 4.5,
    "Depth Units": 0.000250000011874363,
    "Error Polling Enabled": 1,
    "Frames Queue Size": 16,
    "Freefall Detection Enabled": 1,
    "Global Time Enabled": 0.0,
    "Inter Cam Sync Mode": 0.0,
    "Invalidation Bypass": 0.0,
    "Laser Power": 100,
    "Lld Temperature": 25.1394882202148,
    "Ma Temperature": 25.067985534668,
    "Mc Temperature": 24.8353538513184,
    "Min Distance": 0.1,
    "Noise Filtering": 2,
    "Post Processing Sharpening": 1,
    "Pre Processing Sharpening": 4,
    "Receiver Gain": 9,
    "Sensor Mode": 0.0,
    "Visual Preset": 0.0,
    "Zero Order Enabled": 0.0,
    "stream-depth-format": "Z16",
    "stream-fps": "30",
    "stream-height": "768",
    "stream-ir-format": "Y8",
    "stream-width": "1024"
  }

cdb0y511, 2021-04-26:
Related to 
https://github.com/IntelRealSense/realsense-ros/issues/1540

cdb0y511, 2021-04-26:
For now I use realsense-view to export JSON configure.

RealSenseSupport, 2021-05-07:
Thanks @cdb0y511 for the request for JSON config documentation.  We will have a look at it for potential future.  Exporting from RealSense Viewer is the best option at this time, so glad you were able to move forward that way.

closed 2021-06-10 by RealSenseSupport