realsenseai/librealsense #8899: How to get the IMU Noise Value

seungsunkim, 2021-04-26:
When I run ORB SLAM3 with RealSense D455, I input Monocular + IMU data to extract the feature points. I am curious how to get the value of the IMU noise item in the yaml file.
Also, please tell me what the Tbc mean value.
I upload my yaml file.

%YAML:1.0

#--------------------------------------------------------------------------------------------
# Camera Parameters. Adjust them!
#--------------------------------------------------------------------------------------------
Camera.type: "PinHole"

# Camera calibration and distortion parameters (OpenCV) 
Camera.fx: 635.759338
Camera.fy: 635.412842
Camera.cx: 630.737305
Camera.cy: 363.161835

Camera.k1: 0.0
Camera.k2: 0.0
Camera.p1: 0.0
Camera.p2: 0.0
Camera.k3: 0.0

Camera.width: 1280
Camera.height: 720

# Camera frames per second 
Camera.fps: 30.0

# IR projector baseline times fx (aprox.)
Camera.bf: 40.0

# Color order of the images (0: BGR, 1: RGB. It is ignored if images are grayscale)
Camera.RGB: 1

# Close/Far threshold. Baseline times.
ThDepth: 40.0

# Deptmap values factor 
DepthMapFactor: 1000.0

# Transformation from camera 0 to body-frame (imu)
Tbc: !!opencv-matrix
   rows: 4
   cols: 4
   dt: f
   data: [0.99996271, -0.00181009, 0.00844446, 0.02651246,
         0.00171691, 0.9999377, 0.01102943, 0.00830124,
        -0.0084639, -0.01101452, 0.99990352, -0.01286511,
         0.0, 0.0, 0.0, 1.0]

# IMU noise
IMU.NoiseGyro: 0.01
IMU.NoiseAcc: 0.1
IMU.GyroWalk: 0.0001
IMU.AccWalk: 0.001
IMU.Frequency: 9.805 #200

#--------------------------------------------------------------------------------------------
# ORB Parameters
#--------------------------------------------------------------------------------------------

# ORB Extractor: Number of features per image
ORBextractor.nFeatures: 1000

# ORB Extractor: Scale factor between levels in the scale pyramid 	
ORBextractor.scaleFactor: 1.2

# ORB Extractor: Number of levels in the scale pyramid	
ORBextractor.nLevels: 8

# ORB Extractor: Fast threshold
# Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
# Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
# You can lower these values if your images have low contrast			
ORBextractor.iniThFAST: 20
ORBextractor.minThFAST: 7

#--------------------------------------------------------------------------------------------
# Viewer Parameters
#--------------------------------------------------------------------------------------------
Viewer.KeyFrameSize: 0.05
Viewer.KeyFrameLineWidth: 1
Viewer.GraphLineWidth: 0.9
Viewer.PointSize:2
Viewer.CameraSize: 0.08
Viewer.CameraLineWidth: 3
Viewer.ViewpointX: 0
Viewer.ViewpointY: -0.7
Viewer.ViewpointZ: -1.8
Viewer.ViewpointF: 500

PointCloudMapping.Resolution: 0.01
meank: 50
thresh: 2.0

Map.mapfile: map.bin

MartyG-RealSense, 2021-04-26:
Hi @seungsunkim   Some people take the approach of accessing the IMU noise variances directly from the camera using the command **rs-enumerate-devices -c** to list the camera's calibration information.  This command to launch the 'rs-enumerate-devices' SDK tool in calibration information mode can be input into a Linux terminal or the Command Prompt terminal of Windows.

After the command is input, a long list of calibration information is generated.  Scroll upward from the botto of the listing until the **Motion Intrinsic Parameters** section is found.  The noise and bias variances for the IMU gyro and accel of the particular camera attached to the computer are listed here.

The information below was generated by a D435i.

[image]

This next chart is from a D455.

[image]

As can be observed in the above charts, the bias and noise variances are typically always represented as all zeroes when read directly from the camera no matter how many times the information is checked.

In regard to the meaning of **tbc**, the best explanation that I could find was "the extrinsic transition matrix from body of IMU to camera from IMU-camera calibration".

seungsunkim, 2021-04-27:
@MartyG-RealSense 

Thank you for answer.
Do you know why this is happening?

[image]

MartyG-RealSense, 2021-04-27:
In my experience it is normal for the noise and biases to all be zero when reading them with rs-enumerate-devices.

Having said that, RealSense users do tend to obtain different results when reading the raw, noisy IMU data with non-official calibration tools such as **Kalibr**.  An example of noise readings from Kalibr is provided at the bottom of the comment lonked to below.

https://github.com/IntelRealSense/librealsense/issues/7103#issuecomment-675284211

[image]

Note though that the above values have **e-** on the end of them, indicating that they are actually very tiny numbers that may as well be stated as being zero.

seungsunkim, 2021-04-29:
@MartyG-RealSense 
To find the IMU Noise value, which of the following is it possible?
https://www.programmersought.com/article/92054092003/            No. 1
https://github.com/IntelRealSense/realsense-ros                              No.2

MartyG-RealSense, 2021-04-29:
You could obtain the IMU noise values with Kalibr, like a RealSense user in case https://github.com/IntelRealSense/librealsense/issues/7103 did.  Like the chart above and in link **No. 1** though, the results are tiny **e-** values.  In my opinion it would therefore be simpler just to hard-code all the noise values to have a fixed value of '0' instead of retrieving them in real-time from the IMU.

MartyG-RealSense, 2021-05-05:
Hi @seungsunkim  Do you require further assistance with this case, please?  Thanks!

seungsunkim, 2021-05-06:
@MartyG-RealSense 
What should I do to run Kalibr with Real Sense D455?

MartyG-RealSense, 2021-05-06:
I have not used Kalibr myself.

https://github.com/ethz-asl/kalibr/

Someone did post a Kalibr setup guide for RealSense D435i that should be adaptable for D455 though, as they are both IMU-equipped 400 Series cameras.

https://www.programmersought.com/article/29581858144/

seungsunkim, 2021-05-06:
@MartyG-RealSense 
Thank you for answer.
If I have any questions, I will ask again.

closed 2021-05-18 by MartyG-RealSense