realsenseai/librealsense #8907: L515 Depth Sensor Stopped Working

talhakhan13, 2021-04-26:
| | |
|---|---|
|**librealsense**|2.44.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense L515|
|**Serial Number**|f0350097|
|**Firmware Version**|01.05.05.00|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|0003b538dc53|
|**Firmware Update Id**|0003b538dc53|

Please provide a description of the problem

When I try to run the depth sensor on the device I get the following error:

**Invalid Value in rs2_open_multiple(sensor:0000017FD67CEFB0,
profiles:0000017FD65CFA88, count:2):**

**Failed to resolve the request:
	Format: Z16, width: 640, height: 480 	Format: Y8, width: 640, height: 480**

**Into:
	Formats:
	 Y8 	 Z16**

When I try to change the preset I get the following error:

**Backend in rs2_set_option(options:0000017FD67CEFB0, option:Visual Preset,
value:2):
hr returned: HResult 0x8007001f: "A device attached to the system is not
functioning."**

I have tried changing the version of the realsense viewer and have also tried using a different computer (mac) but the depth sensor does not work, although the RGB camera and motion module work fine. The only thing that changed since my last use of the device is that a windows 10 update was installed. Any help is appreciate! Thanks

RealSenseSupport, 2021-05-08:
Hi @talhakhan13 

Sorry you're running into this particular issue.  Which versions of RealSense Viewer have you tried?  Does the behavior change if you're plugged into a USB2 port or USB3 port?  Does behavior change if you change the orientation of the USBC plug that goes into the L515 camera?  Are you using a USB hub or connecting directly into the port on computer system?

If you can open a ticket here:  https://support.intelrealsense.com/hc/en-us/requests/new

talhakhan13, 2021-05-09:
I have tried the stuff that you have mentioned but It did not resolve the issue. For now, I have returned the device back to the folks that purchased it so we can close this thread. Thanks for the help!

closed 2021-05-09 by talhakhan13