realsenseai/librealsense #8911: Ubuntu doesn't recognise the D435i camera

Makekihe, 2021-04-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | RealSense Viewer v2.44.0 | 
| Operating System & Version |  Linux (Ubuntu 18 LTS) | 
| Kernel Version (Linux Only)    |  5.4.0-72-generic                                        | 
| Platform                                 | PC (Dual booted with Windows)  |
| SDK Version                            |  2.0                          | 
| Language                            |                            | 
| Segment			|  Robot                          | 

### Issue Description
We are struggling to connect the Intel Realsense D435i to our Ubuntu 18 operating system, on a dual booted computer. We've followed the [installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) for installing the realsense SDK, DKMS package. We've connected the camera using a USB 3.0 port, like mentioned in the [troubleshooting page](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A#q-i-connected-the-camera-to-the-usb-port-but-it-is-not-recognized) and ensured that the camera is recognised using the following command:

```
$ lsusb | grep 8086 
```

where we get the output:

```
Bus 002 Device 003: ID 8086:0b3a Intel Corp.
```

Inside the realsense viewer we receive the following error message, but no other errors seem to appear:

```
backend-v4l2.cpp:502 - Cannot access /sys/class/video4linux
```

When checking the kernel version as part of the [installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) we receive the following information:

```
$ modinfo uvcvideo | grep "version:"
version:        1.1.2.realsense-1.3.16
srcversion:     59E337C6EFCFABEFC2444BA
```

Additional information:
- We've tested the camera on the windows part of the computer, with the same USB port, and here it worked fine. 
- We've tested if the camera is registered with other webcam apps like Cheese, but without any luck
- The built in camera of the computer also isn't recognised by these webcam apps (like Cheese)

What can we do to resolve this issue?

Makekihe, 2021-04-27:
Problem seems to be resolved - the realsense viewer recognises the camera now. 

During the installation of the repository with command: **sudo add-apt-repository "deb https://librealsense.intel.com/Debian/apt-repo bionic main" -u**, we noticed the following in the terminal:

```
E: The repository 'http://ppa.launchpad.net/xqms/opencv-nonfree/ubuntu bionic Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

Here we managed to resolve the issue by following the suggestion provided in [this issue](https://stackoverflow.com/questions/41244349/getting-following-error-after-the-command-sudo-apt-get-update-on-ubuntu-16-04), where apparently incorrect options were set in the Software Updates application in Ubuntu. 

Note that it is unknown if this will resolve similar issues for others, or if it will cause other problems in the future.

closed 2021-04-27 by Makekihe

MartyG-RealSense, 2021-04-27:
Thanks very much @Makekihe for the update and the highlighting of your solution!