realsenseai/librealsense #8914: D435i coordinates

jingshaojing, 2021-04-27:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i| 
 I want to know the coordinnates for D435i for the color ,the left infra ,the IMU and the right infra. As explained in 
[https://github.com/IntelRealSense/librealsense/blob/master/doc/d435i.md]
the origin is depth ,and x point to right ,y points to down, z  points forward . And said :"The depth<->IMU sensor extrinsic (rigid body transformation) is precalculated based on mechanical drawings and cannot be modified.", what is this sensor extrinsic? or is there some drawing to  express the coordinate for the color ,the IMU ,the infra like the following :
[image]

MartyG-RealSense, 2021-04-27:
Hi @jingshaojing   Reading the three links below in the order that they are listed may be helpful in answering your questions about the coordinate system.

https://github.com/IntelRealSense/realsense-ros/issues/1449#issuecomment-709889034
https://github.com/IntelRealSense/librealsense/issues/7279#issuecomment-689031488
https://github.com/IntelRealSense/librealsense/issues/7279#issuecomment-690188950

Please feel free to post further questions below after having read the above links,  Thanks!

jingshaojing, 2021-04-27:
After I have read  above link, I get something I want to know .thanks very much . @MartyG-RealSense

MartyG-RealSense, 2021-04-27:
You are very welcome @jingshaojing    :)

MartyG-RealSense, 2021-05-04:
Hi @jingshaojing  Do you require further assistance with this case, please?  Thanks!

jingshaojing, 2021-05-04:
I shouldn’t need it in the near future. You can close this issue.thanks!

closed 2021-05-04 by jingshaojing