realsenseai/librealsense #8915: I could not launch rs_rgbd.launch to open realsense D435i camera node

sivaeinfo, 2021-04-27:
---------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D435i } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {(Ubuntu 18  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| ### Issue Description

**I have installed all necessary files for the mentioned packages. Then I ran the below commands bit@bit-OptiPlex-990:~$ roslaunch realsense2_camera rs_rgbd.launch**

**It shows an error and did not open rviz** 

WARNING: Package name "object_detection_with_intel_D415" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes. ... logging to /home/bit/.ros/log/a3d4b3c0-a724-11eb-a4aa-d4bed9944d8b/roslaunch-bit-OptiPlex-990-5534.log Checking log directory for disk usage. This may take a while. Press Ctrl-C to interrupt Done checking log file disk usage. Usage is <1GB.

WARNING: Package name "object_detection_with_intel_D415" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes. started roslaunch server http://bit-OptiPlex-990:41865/

SUMMARY
PARAMETERS * /camera/realsense2_camera/accel_fps: -1 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti... * /camera/realsense2_camera/align_depth: True * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de... * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de... * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de... * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de... * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de... * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de... * /camera/realsense2_camera/allow_no_texture_points: False * /camera/realsense2_camera/base_frame_id: camera_link * /camera/realsense2_camera/calib_odom_file: * /camera/realsense2_camera/clip_distance: -1.0 * /camera/realsense2_camera/color_fps: -1 * /camera/realsense2_camera/color_frame_id: camera_color_frame * /camera/realsense2_camera/color_height: -1 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti... * /camera/realsense2_camera/color_width: -1 * /camera/realsense2_camera/confidence_fps: 30 * /camera/realsense2_camera/confidence_height: 480 * /camera/realsense2_camera/confidence_width: 640 * /camera/realsense2_camera/depth_fps: -1 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame * /camera/realsense2_camera/depth_height: -1 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti... * /camera/realsense2_camera/depth_width: -1 * /camera/realsense2_camera/device_type: * /camera/realsense2_camera/enable_accel: False * /camera/realsense2_camera/enable_color: True * /camera/realsense2_camera/enable_confidence: True * /camera/realsense2_camera/enable_depth: True * /camera/realsense2_camera/enable_fisheye1: False * /camera/realsense2_camera/enable_fisheye2: False * /camera/realsense2_camera/enable_fisheye: False * /camera/realsense2_camera/enable_gyro: False * /camera/realsense2_camera/enable_infra1: False * /camera/realsense2_camera/enable_infra2: False * /camera/realsense2_camera/enable_infra: False * /camera/realsense2_camera/enable_pointcloud: False * /camera/realsense2_camera/enable_pose: False * /camera/realsense2_camera/enable_sync: True * /camera/realsense2_camera/filters: * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f... * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o... * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f... * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o... * /camera/realsense2_camera/fisheye_fps: -1 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame * /camera/realsense2_camera/fisheye_height: -1 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op... * /camera/realsense2_camera/fisheye_width: -1 * /camera/realsense2_camera/gyro_fps: -1 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic... * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica... * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt... * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt... * /camera/realsense2_camera/infra_fps: -1 * /camera/realsense2_camera/infra_height: -1 * /camera/realsense2_camera/infra_rgb: False * /camera/realsense2_camera/infra_width: -1 * /camera/realsense2_camera/initial_reset: False * /camera/realsense2_camera/json_file_path: * /camera/realsense2_camera/linear_accel_cov: 0.01 * /camera/realsense2_camera/odom_frame_id: camera_odom_frame * /camera/realsense2_camera/ordered_pc: False * /camera/realsense2_camera/pointcloud_texture_index: 0 * /camera/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR * /camera/realsense2_camera/pose_frame_id: camera_pose_frame * /camera/realsense2_camera/pose_optical_frame_id: camera_pose_optic... * /camera/realsense2_camera/publish_odom_tf: True * /camera/realsense2_camera/publish_tf: True * /camera/realsense2_camera/rosbag_filename: * /camera/realsense2_camera/serial_no: * /camera/realsense2_camera/stereo_module/exposure/1: 7500 * /camera/realsense2_camera/stereo_module/exposure/2: 1 * /camera/realsense2_camera/stereo_module/gain/1: 16 * /camera/realsense2_camera/stereo_module/gain/2: 16 * /camera/realsense2_camera/tf_publish_rate: 0.0 * /camera/realsense2_camera/topic_odom_in: camera/odom_in * /camera/realsense2_camera/unite_imu_method: none * /camera/realsense2_camera/usb_port_id: * /rosdistro: melodic * /rosversion: 1.14.10

NODES /camera/ color_rectify_color (nodelet/nodelet) points_xyzrgb_hw_registered (nodelet/nodelet) realsense2_camera (nodelet/nodelet) realsense2_camera_manager (nodelet/nodelet)

auto-starting new master process[master]: started with pid [5547] ROS_MASTER_URI=http://localhost:11311

setting /run_id to a3d4b3c0-a724-11eb-a4aa-d4bed9944d8b WARNING: Package name "object_detection_with_intel_D415" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes. process[rosout-1]: started with pid [5558] started core service [/rosout] process[camera/realsense2_camera_manager-2]: started with pid [5569] process[camera/realsense2_camera-3]: started with pid [5570] process[camera/color_rectify_color-4]: started with pid [5571] process[camera/points_xyzrgb_hw_registered-5]: started with pid [5579] 27/04 02:49:00,893 ERROR [140341701707520] (ds5-options.cpp:88) Asic Temperature value is not valid!

MartyG-RealSense, 2021-04-27:
Hi @sivaeinfo   You have confirmed in https://github.com/IntelRealSense/librealsense/issues/7560#issuecomment-827437922 that you have installed rs_rgbd using **sudo apt-get install ros-melodic-rgbd-launch**.  I also note that you are using the project https://github.com/dovanhuong/object_detection_2d for ROS, tested with the D435 camera model.

The **object_detection_2d** project was created in 2018, which makes me think that it may be have been designed for ROS **Kinetic** instead of the more recent ROS Melodic that you are using.

That project uses C++, so I researched similar but much more recent 2D-to-3D object detection projects for ROS that are based on C++ and have been tested with D435.  The project in the link below may be worth trying.

https://github.com/Mazhichaoruya/ROS-Object-Detection-2Dto3D-RealsenseD435

sivaeinfo, 2021-04-27:
@MartyG-RealSense Thank you. I have solved previous issues. Now I will look in to your suggestion.

closed 2021-04-28 by sivaeinfo