realsenseai/librealsense #8919: Flexible connector between d450 and d4v3 board

TheBricktop, 2021-04-28:
Hello, 
we've just received a batch of d450 modules and d4 vision processor boards and yet didn't get any way of connecting those two.
From spreadsheet and some teardown photos we've learned that the connecting part in the d455 is made with flexible tape allowing to place the board behind the camera. But the only part that weve managed to find on resellers sites (digkey and mouser) is a rigid board with IPEX connector.
Is there any similar part that we could use instead of the rigid connector? Preferably we would like to hide the board behind the camera to reduce the dimensions of the assembled modules.
[image]

MartyG-RealSense, 2021-04-28:
Hi @TheBricktop   Pre-made flexible interposer cables for connecting the D4 board to the Depth Module board are available from **FRAMOS** in lengths ranging from 0.005 meters to 0.2 meters.

https://www.framos.com/en/hsf100-22274

Left-click on the drop-down menu selector beside **Cable length (m)** to select the preferred length.

You could also create your own custom interposer flexi-cable using details in the data sheet document of the 400 Series cameras.

MartyG-RealSense, 2021-05-05:
Hi @TheBricktop  Do you require further assistance with this case, please?  Thanks!

TheBricktop, 2021-05-05:
Well I would never think of naming cable as descriptive as HSF100. But no I'm good.

MartyG-RealSense, 2021-05-05:
Yes, in that particular case you need to know what to search for,  :)   I will close the case now as you are satisifed with the outcome.  Thanks again!

closed 2021-05-05 by MartyG-RealSense