realsenseai/librealsense #8923: Realsenses freeze after running for a while on Windows 10

t-kole-yipp, 2021-04-28:
In our setup, we have multiple realsense D455's hooked up to a single computer.
After leaving the application on for a while - sometimes days, sometimes only an hour - one or multiple realsenses stop sending frames.

What will happen is that the application will thrown a frame receive timeout. The IR projector of the device stays on, but there's no new frames coming in.

Restarting the application does not work, they also become unresponsive in realsense viewer.
We're using an application built in Unity with the Unity C# sdk.

The only solution is to plug the device out and back in, effectively cutting power to the device.

This has been happening very consistently, across multiple PC's with multiple realsense models.

Has anyone else encountered this?
Are there any fixes?

MartyG-RealSense, 2021-04-28:
Hi @t-kole-yipp  As a starting point in investigating your case, could you answer the questions below please:

1.  How many D455 cameras do you have attached to the same computer, and how many of those cameras are enabled at the same time?

2.  Are you using hardware sync and connecting the cameras together with physical sync cabling?

3.  Are you using USB extension cables of your own choice instead of the official 1 meter cable that is supplied with the camera?

4.  Are the cameras being used outdoors or indoors?  If they are indoors, are the cameras likely to be exposed directly to sulight in the location that they are in?

MartyG-RealSense, 2021-05-04:
Hi @t-kole-yipp  Do you require further assistance with this case, please?  Thanks!

t-kole-yipp, 2021-05-04:
Sorry for the late response.

1. About 9 camera's, though we've seen hangs with fewer.

2. No syncing

3. All sorts of cables, including very long USB A to C, official 1m + extender, 1m + extension over ethernet. Freezes don't seem to happen more frequently with any specific one cable setup.

4. Indoors, out of direct sunlight.

The biggest problem is that it happens infrequently and usually after a long time, so it's hard to replicate - yet it happens frequently enough to be a problem.

Is there a way we can debug the problem faster? Are there logs anywhere that might give a hint?

MartyG-RealSense, 2021-05-04:
If it is happening over a long period of time and you are using multiple cameras on the same computer then it is worth monitoring the computer's system resources (such as memory) to check whether they are being consumed over time.    

On Windows you can monitor the computer's resources in real-time on the **Performance** tab of the Task Manager interface.  The Task Manager can be accessed with the keyboard shortcut Ctrl-Alt-Delete.

[image]

Select the **Task Manager** option and then on the Task Manager interface, select the **More Details** option and click on the **Performance** tab.

Are all of the cameras active simultaneously?  If so, then the more cameras that are active at the same time on the same computer, the greater the drain on the computer's resources may be.  In 2018 an Intel seminar about multiple cameras recommended a computer with an **i7** processor for a 4-camera setup.  By comparison, an **Up Squared** board can handle 2 cameras.

Having said that, your case has similar symptoms to the currently active case below that has experienced problems on both Windows and Raspberry Pi.  The IR projector seems to remain active but the depth image breaks, and it occurs indoors both in daytime and night-time.  Their RGB image also goes black.  Only an unplug-replug or camera reset corrects the problem, as restarting the program does not correct it.

https://github.com/IntelRealSense/librealsense/issues/8880

A recent theory I examined in that case was that depth and RGB exposure might be dropping to '1'.

https://github.com/IntelRealSense/librealsense/issues/8880#issuecomment-829913978

t-kole-yipp, 2021-05-04:
Thanks for the response.

We have specifically selected out computers to handle such load.
The resource consumption is certainly high, but not so high that it could be a problem.
If it were to increase over time, this could be an indication of a memory leak on our side, but this is also not the case.

After the realsense stops working, the other ones continue just fine, and the application keeps running with one device less.

I'm not sure if I agree to the similarity of the linked case - in our situation the realsense freezes up completely, usually unresponsive to any program, it sounds like their depth frame just becomes unusable.

MartyG-RealSense, 2021-05-04:
It sounds as though the camera remains connected (detected) but is simply unresponsive.  Is that correct, please?

If so, then if the camera remains detected then an alternative to a cable unplug-replug may be to automatically reset the affected camera with a **hardware_reset()** instruction when the camera is no longer receiving frames.  Have you tried this approach?

t-kole-yipp, 2021-05-04:
Yes, that is correct. We will try the hardware reset soon, and we will let you know if it works.

closed 2021-05-15 by MartyG-RealSense