realsenseai/librealsense #8924: unity - ExternalException - rs2_get_option_range and option:Inter Cam Sync Mode

Hugodovs, 2021-04-28:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400 (D435) | 
| Firmware Version                   | 05.12.12.100 | 
| Operating System & Version |  Win 10) | 
| Platform                                 | PC  |
| SDK Version                           |  2.44.0 | 
| Language                            | unity 2020.1.3f1 | 
| Segment			|  others | 

### Issue Description

I was able to run the examples (with the Inspector) on the unity package. I edited something on Intel RealSense Viewer v2.44.0, and now I am having problems with the RsDeviceInspector.

If I add it to the RsDevice, I get one frame, and I receive the following error:
[image]

I have tried different firmwares, unitypackage and SDK. But I am running out of options. I believe a particular configuration was changed (related to the Inter Cam Sync Mode) and it is not allowed in Unity. 

I am trying the examples on a clean project. It works, but only without the RsDeviceInspector. Is there any way to change the current configuration outside Unity and the RealSense Viewer? Or any suggestion to make it work?

More information:
* I have done the golden reset
* In viewer: Stereo Module > Controls > Inter Cam Sync Mode is 0.000
* I am using one single camera

MartyG-RealSense, 2021-04-29:
Hi @Hugodovs  I cannot think of a circumstance in which changing a setting in the Viewer would negatively affect the Unity wrapper.  Can you confirm please that the Viewer is not open when you are using the Unity wrapper?

Since you are using Windows, it should be straightforward to install SDK 2.44.0 again with the installer file on the SDK **Releases** page.  This would completely overwrite your current 2.44.0 installation and so may clear a problem associated with it.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.44.0

[image]

Hugodovs, 2021-04-29:
Hi @MartyG-RealSense, thanks for your feedback!

I overwrite my current 2.44.0 installation and the Viewer was not open when I am using the Unity wrapper.
_____________________________________

I have also tested the Python API and got this:

[image]

Is it ok to happen?
_______________________________________

I just made the following steps in an almost (there were other nonrelated programs installed there) clean computer:

1) Installed Unity Hub and install Unity 2019.4.25f1.
2) Created a new project, Download the Intel.RealSense.unitypackage and import to the project.
3) Opened TextureDepthAndInfrared scene
4) Run. Everything seems ok.

Problem 1:
5) Add RsDeviceInspector to RsDevice (the scene continues running, but I got the previous error and the inspector does not work at all)
[image]

6) Opened TextureDepthAndColor scene. Here, I notice a drop on the fps in the beginning, the "RGB texture" is being updated normally but the "Depth texture" is frozen in the first frame. This happens without the inspector.
[image]

___________________________________________________________________________
When I open a new scene, I also get  the following error (but unrelated), probably Unity version.
[image]
___________________________________________________________________________

MartyG-RealSense, 2021-04-29:
I believe that you shouldn't set Inter Cam Sync Mode individually for both depth and color sensor but with a single definition.  A couple of examples of setting Inter Cam Sync Mode with Python code can be found in the link below.

https://github.com/IntelRealSense/librealsense/issues/2623#issuecomment-433809065

https://github.com/IntelRealSense/librealsense/issues/8313

[image]

If you are adding **RsDeviceInspector** after the program has started running then I would recommed instead adding it **before** running the program.

Hugodovs, 2021-04-29:
I am adding it before running the program (as well as after the program has started).

The goal of the python code was to show that there is a similar problem there as well. It seems that RsDeviceInspector is trying to access Inter Cam Sync Mode from a sensor (color sensor?) and it does not have it. 

I was thinking to try to avoid this with an "if" inside the loop of the RsDeviceInspector, but this is not a clean solution. Any other option?

MartyG-RealSense, 2021-04-29:
Going back to this Unity error:

[image]

I recall that Unity changed their GUI layer system a while ago.  If the **TexturesDepthAndInfrared** example dates back to before that change (its update history goes back to July 2018) then that may be why the GUI layer in that program gets auto-removed by Unity in modern versions of it such as 2019.4.

In regard to the Inter Cam Sync error: I created a new Unity project and imported the Unity wrapper into it, and then loaded the **TexturesDepthAndInfrared** scene.  I observed that RsDeviceInspector is not included in the scene by default.  When I added it to the scene in the same object as RsDevice and ran the program, I experienced the same Inter Cam Sync Mode error that you did.  So there does not seem to be a problem with your particular computer or RealSense setup.

MartyG-RealSense, 2021-05-05:
Hi @Hugodovs  Do you require further assistance with this case, please?  Thanks!

Hugodovs, 2021-05-05:
I believe there is a bug around that, but no need for further assistance
Thanks for your help!

closed 2021-05-05 by Hugodovs

MartyG-RealSense, 2021-05-05:
You are very welcome @Hugodovs - thanks for the update!

cbrownNeuro, 2022-03-01:
Hello everyone, I'm having the same stated above. When attempting to use the **RsDeviceInspector** script attached to the **RsDevice** prefab, it gives me this error:

```
ExternalException: rs2_get_option_range(options:0000020AB1133320, option:Inter Cam Sync Mode, min:0, max:0, step:0, def:0)
Rethrow as ArgumentException: object doesn't support option #42
Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged (System.IntPtr pNativeData) (at <a617af732b15461a9449f771300f29c8>:0)
Intel.RealSense.OptionInternal..ctor (System.IntPtr options, Intel.RealSense.Option option) (at <a617af732b15461a9449f771300f29c8>:0)
Intel.RealSense.OptionsList+<GetEnumerator>d__14.MoveNext () (at <a617af732b15461a9449f771300f29c8>:0)
System.Linq.Enumerable.ElementAt[TSource] (System.Collections.Generic.IEnumerable`1[T] source, System.Int32 index) (at <351e49e2a5bf4fd6beabb458ce2255f3>:0)
RsDeviceInspector.onStartStreaming (Intel.RealSense.PipelineProfile profile) (at Assets/RealSenseSDK2.0/Scripts/RsDeviceInspector.cs:70)
RsDeviceInspector+<WaitForDevice>d__6.MoveNext () (at Assets/RealSenseSDK2.0/Scripts/RsDeviceInspector.cs:31)
UnityEngine.SetupCoroutine.InvokeMoveNext (System.Collections.IEnumerator enumerator, System.IntPtr returnValueAddress) (at <d3b66f0ad4e34a55b6ef91ab84878193>:0)
```

Let me know if anyone has found a working solution.

MartyG-RealSense, 2022-03-01:
Hi @cbrownNeuro  Do you experience the **object doesn't support option #42** error with the sample programs supplied with the Unity wrapper, or does it only occur if you set up your own **RsDevice** prefab in your own project?

cbrownNeuro, 2022-03-01:
Hello @MartyG-RealSense , currently if I were to open the Unity sample programs, i.e. **TexturesDepthAndColor**, they do not have **RsDeviceInspector** script attached. When I attach it, it gives me the same error, <object doesn't support option #42>.

MartyG-RealSense, 2022-03-02:
Are you able to run **TexturesDepthAnd Color** if you load the RealSense Unity wrapper's **StartHere** scene, run it in the Unity editor with the small triangular play button at the top of the Unity window and then select **UI Image Sample (Depth and Color)** from the list of examples?

[image]

cbrownNeuro, 2022-03-02:
Yes I can. When I do so, the **RsDeviceInspector** script is not attached. When I attempt to attach it, it gives the same error listed above.

MartyG-RealSense, 2022-03-03:
**RsDeviceInspector** is a Unity component that only operates when streaming is enabled after a program is run.  It can be added to the **RsDevice** object either before or during streaming, but its options are not accessible until streaming begins.

The **TexturesDepthAndColor** example scene does not use RsDeviceInspector by default and so the component is absent when the scene is loaded as it is not a required component for that scene.  You can manually add the component to the RsDevice object using the 'Add Component' button in the Inspector if you wish to though, either before or during rumtime.

May I confirm please which RealSense camera model you are using?  I tested the TexturesDepthAndColor scene successfully in a new project created in Unity 2021.1.16f1 with the **3D** template and using the D415, D435i, D455 and L515 camera models.  Each time that I added RsDeviceInspector during runtime, it worked correctly.

I do not have a T265 to test with, but that model does not have built-in depth sensing anyway and so would not work with a depth sample program.  

[image]

I also tested adding RsDeviceInspector to an object other than the RsDevice object that contains the RsDevice script component.  RsDeviceInspector did not work during runtime if it was not placed inside the RsDevice object alongside the RsDevice component.

cbrownNeuro, 2022-03-03:
I understand that, I am using the D455. I am running Unity 2020.3.30f1, on a 3D project (although I'm mostly working in 2D).

Does the Unity version have any relevance here? I remember once a month or so ago when I initially began working with RealSense on Unity that the Inspector was working.

MartyG-RealSense, 2022-03-04:
Thank you for the confirmation of your RealSense camera model.  

The RealSense Unity wrapper has been compatible with Unity 2020 since SDK version 2.40.0 and so the 2.44 version that you are using should be fine.

When you import the RealSense Unity wrapper's package into your scene, are you using one that was downloaded from the 'Assets' file list of the 2.44.0 SDK version's **Releases** page?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.44.0

cbrownNeuro, 2022-03-04:
I don't recall which version I was on, but I updated to 2.44.0. Unfortunately, I still got the same error: `object doesn't support option #42`

MartyG-RealSense, 2022-03-04:
I am unable to replicate the error that you are experiencing.  

Can you confirm whether you are importing the **Intel.RealSense.unitypackage** package into a new empty project or into your existing project?  If it is an existing project, does the error still occur if you import the package into a new project (as if it did work in a new project then it might suggest that something in your existing project was causing a conflict).

cbrownNeuro, 2022-03-04:
When importing into an empty project it fails, as well as when importing to my existing project.

I'm looking to modify the depth frames through accessing the sensor options. Testing and tweaking through the inspector is the best way to test, and then finally set those parameters in code.

It's strange because as I mentioned before, I remember seeing the inspector work once before; I believe it was in a sample project once I added the script to RsDevice.

MartyG-RealSense, 2022-03-05:
Do you recall whether you have previously installed the **RealSense UWP driver** on your computer?  If you have then it would be advisable to remove it as it has been reported in the past to cause errors with some programs, including Unity.

https://www.intel.com/content/www/us/en/support/articles/000026698/emerging-technologies/intel-realsense-technology.html

cbrownNeuro, 2022-03-08:
It is a similar issue, it gives an error when attempting to access Inter Cam Sync Mode.

Like you said, that post does provide a solution, but unfortunately when solving the problem using Python. Is there a way to access devices on Unity without the use of the pipeline?

MartyG-RealSense, 2022-03-08:
You can get some additional control over a pointcloud in Unity by using a RealSense camera with a **Visual Effect Graph**, as described in the project in the link below.  It would likely use the pipeline since the project makes use of the RealSense Unity wrapper files, but it may offer an alternative to defining custom settings with RsDeviceInspector.

https://github.com/keijiro/Rsvfx

Alternatively, the RGB image of the camera may be able to be streamed onto a Unity texture using Unity's **WebCamTexture** instruction, which would not involve use of the RealSense Unity wrapper as the camera would be treated like an ordinary video webcam.

https://docs.unity3d.com/ScriptReference/WebCamTexture.html