realsenseai/librealsense #8929: L515 camera "box_dimension_calculate_error"

zeynepkoyun, 2021-04-29:
Hello, I was able to run the code "box_dimensioner_multicam_demo.py" for the D415 camera.   However, the L515 cannot find the objects for the camera. When I looked at the code, I noticed that the pointcloud value was empty.  What should the deep_thresold value be for this camera?

zeynepkoyun, 2021-04-30:
@MartyG-RealSense  Hi,  In the "convert_depth_frame_to_pointcloud" method, I multiplied the depth image by 0.00025 instead of 0.001. But how is the "depth_thresold" value specified in the "calculate_cumulative_pointcloud" method determined? Can you help me? (The camera I work with is L515.)

zeynepkoyun, 2021-05-04:
@ MartyG-RealSense 
While trying to run the "https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/box_dimensioner_multicam" sample project with the L515 camera, I get a dimension error. I cannot measure the values exactly.

To explain with an example:
[image]

RealSenseSupport, 2021-05-04:
Hi @zeynepkoyun 

Our main box dimensioning software is the DWS software/SDK product for L515 is here:  https://www.intelrealsense.com/dimensional-weight-software/

Regarding the python example, not all of our examples have been updated properly to support the L515 camera.  We'll need to look into this further.

zeynepkoyun, 2021-05-05:
Thanks for your answer @RealSenseSupport 

I started to investigate, but I want your opinion on a topic. Could the distortion pattern defined as None [0 0 0 0]] in the depth image for the L515 camera have an effect on the calculation?

maloel, 2021-06-03:
Hi @zeynepkoyun 
Our sample Python code should be updated in the coming weeks, likely with version 2.48 at the end of the month. The main issue there was the resolution for the L515. Does that answer your question?

zeynepkoyun, 2021-06-03:
@maloel  Thank you for your answer. When I visualized the PointCloud data, I saw a lot of scatter in the product at the corners of the object, so I couldn't find the exact box size without clearing the noise data.
I tried clearing out the dots but this time I couldn't get it to run in real time.
The scattering I mentioned is as in the image.
[image]

aseelegbaria, 2021-06-03:
Hi @zeynepkoyun, 
can you provide us with more information about the issue (how to reproduce it, how did you clear the noise data, and maybe the script you are working with)?
does it happen with L515 or D415?

zeynepkoyun, 2021-06-03:
Hi @aseelegbaria 

I didn't get any errors when I worked with the D415 camera.
While working with the L515 camera, it encountered negative errors.
Negative errors are a common occurrence in LiDAR data, which can result from scattering of photons in a returning laser pulse. This phrase is explained in its source ("https://gis.stackexchange.com/questions/272433/editing-lidar-point-cloud-to-remove-noise-outliers-present-below-and-above-groun/277260").

I also took advantage of the source I mentioned above and this source ("https://greenvalleytl.com/wp-content/lidar360_en/ToolReference/DataManagement/PointCloudTools/OutlierRemoval.html").

RealSenseSupport, 2021-06-18:
There was work done on the example, here is a link to the PR:  https://github.com/IntelRealSense/librealsense/pull/9188

MartyG-RealSense, 2021-09-11:
Thanks very much @zeynepkoyun for the update - I'm pleased to hear that you found a solution!

jerrydegroot, 2021-11-10:
> Hi @aseelegbaria
>
> I didn't get any errors when I worked with the D415 camera. While working with the L515 camera, it encountered negative errors. Negative errors are a common occurrence in LiDAR [...]

Hello @zeynepkoyun ,

Can you share what solved the problem for you?

We discovered that, after correct calibration, the "world coordinates" of the checkerboard are correct for x-y, but z is wrong. We expect that the checkerboard plane should be z=0 , because the depth_threshold of -0.01 would then keep all the points that are more than 1 cm above the checkerboard, i.e. the box, but not the board itself. However, we find that the points of our checkerboard have average z=2.4 m. There are no points <-0.01, and therefore the cumulative_pointcloud is empty. On further inspection, we find that the checkerboard plane at z = 2.4 m is roughly 2 times the distance between camera and plane, d = 1.2 m.  This seems to suggest a "+" instead of "-" solution: z = d + d = 2.4  instead of z = d - d = 0.  Alternatively, a solution could be in what you "negative errors" due to scattering of photons in LiDAR data. 

So please, can you share what solved the problem for you?

jerrydegroot, 2021-11-12:
By changing helper_functions.py line 180 into: z = depth_image.flatten()/4000 , we get "reasonable" results for single L515 camera setup (the bounding box is too large).

We expect that multicam setup will not work for L515 cameras in default asynchronous mode: both cameras will emit IR pulses at the same time, thus interfering each other's depth readings. There is a [whitepaper ](https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup) on using multiple L515 with a hardware trigger/sync. cable which should work for static object/scenery measurements. We are looking for a L515 multicam setup for dynamic object/scenery. Any suggestions welcome.