realsenseai/librealsense #8930: Camera died while updating firmware

Allius27, 2021-04-29:
Hi there.

How we might restore camera firmware?

We are using rs-fw-update to put the same firmware version to the all cameras.
rs-fw-update cought a problem while camera was updating, but not handled it.
And for the current moment the camera is unvisible for the operation system and the sdk.

SDK version is v2.39.0

Short logs:
```
Update camera firmware. Old firmware version is 05.11.01.100

updating device: 
Name: Intel RealSense D435, serial number: 043322075609, update serial number: 037323020969, firmware version: 05.11.01.100, USB type: 3.2
 22/04 00:47:39,719 ERROR [139762849879744] (types.h:313) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: No such device

firmware update started

firmware update progress: 0[%]
firmware update progress: 0[%]
firmware update progress: 0[%]
...
firmware update progress: 58[%]
firmware update progress: 58[%]
firmware update progress: 58[%]Update camera firmware. Old firmware version is unknown

nothing to do, run again with -h for help
Update camera firmware. Old firmware version is unknown

nothing to do, run again with -h for help
Update camera firmware. Old firmware version is unknown
```

> $ dmesg -w
    [  322.475890] usb 2-2: new SuperSpeed USB device number 6 using xhci_hcd
    [  322.492885] usb 2-2: New USB device found, idVendor=8086, idProduct=0adb
    [  322.492896] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
    [  322.492902] usb 2-2: Product: Intel(R) RealSense(TM) RS400 Device
    [  322.492907] usb 2-2: Manufacturer: Intel
    [  322.492912] usb 2-2: SerialNumber: 111111111111

> $ ll /dev/video*
    ls: cannot access '/dev/video*': No such file or directory

Allius27, 2021-04-29:
camera is d435

RealSenseCustomerSupport, 2021-04-29:
----------------------------------------------

Hi

Your D435 turned to RS400 and the idProduct shows "0adb". It looks the camera FW was corrupted during updating.
Please run RealSense Viewer under **Windows 10** on a PC. And see what it is recognized in the Viewer. If you see it show "D4xx recovery" (see attached screenshot), you can go ahead to recover it by updating FW again.

Attachment(s):
D400_recovery_mode.JPG - https://realsensesupport.intel.com/attachments/token/6wBJq7VJs8VNxWawaDISRsaaa/?name=D400_recovery_mode.JPG

Allius27, 2021-04-30:
[image]
I singed on the site, but I am unable to see the image.

Attach the image here, please

closed 2021-05-04 by Allius27