realsenseai/librealsense #8931: How to make kinfu work with recorded .bag file?

akhawaja2014, 2021-04-29:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  (Ubuntu 20)  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.0 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {Kinect Fusion }                          | 

### Issue Description
<Hi everyone
I have a recorded bag file captured from D435i using realsense viewer. I want to  use that rgbd data for Kinect fusion. 

I am able to make rs-kinfu work for real time point cloud fusion but when I try to replace **enable_stream()** with **enable_record_to_file("bagfile.bag")**, It shows error that I am unable to understand and I don't know how to make it work. 

Please help me. I am attaching the code and repository where I took code is 'librealsense'..>

`int main(int argc, char **argv)
{
    // Declare KinFu and params pointers
    Ptr<KinFu> kf;
    Ptr<Params> params = Params::defaultParams();

    // Create a pipeline and configure it
    rs2::pipeline p;
    rs2::config cfg;
    float depth_scale;
    cfg.enable_device_from_file("/home/tgi/Projects/librealsense/wrappers/opencv/kinfu/liver1.bag");
    //cfg.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16);
    auto profile = p.start(cfg);
    auto dev = profile.get_device();
    auto stream_depth = profile.get_stream(RS2_STREAM_DEPTH);`

The error output is as follows"

`16:47:01: Starting /home/tgi/Projects/librealsense/build/wrappers/opencv/kinfu/rs-kinfu ...
terminate called after throwing an instance of 'rs2::not_implemented_error'
  what():  This option is read-only!
16:47:01: The program has unexpectedly finished.
16:47:01: The process was ended forcefully.
16:47:02: /home/tgi/Projects/librealsense/build/wrappers/opencv/kinfu/rs-kinfu crashed.`

MartyG-RealSense, 2021-04-29:
Hi @akhawaja2014   If you are aiming to read data from a bag file instead of use live camera data, you can do so with the instruction **enable_device_from_file** instead of enable_record_to_file.  This is described at the top of the SDK's **Sample Data** page of example bag files.

https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md

To quote the page:

*****************

You can make any of the provided examples run from file, by replacing pipe.start() with:

```
rs2::config cfg;
cfg.enable_device_from_file(<filename>);
pipe.start(cfg); // Load from file
```

akhawaja2014, 2021-04-30:
Hi MartyG
Thank you for your quick response. I really appreciate your help.

I am trying to achieve Kinect fusion of a recorded bag file from the same D435i camera. The code I am using is [rs-kinfu.cpp](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/kinfu/rs-kinfu.cpp). I downloaded it and run it. It works perfectly fine with live stream. 
[image]

However when I try to use enable_device_from_file("bagfile.bag"). It does not work.

I did some debugging and apparently the problem is not in enable_device_from_file(). It is working fine. The program crashes on this line in [rs-kinfu.cpp](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/opencv/kinfu/rs-kinfu.cpp).

`dpt.set_option(RS2_OPTION_VISUAL_PRESET, RS2_RS400_VISUAL_PRESET_HIGH_DENSITY);`

I am also attaching the screenshot of program.

When I recorded the rgbd video from realsense viewer. I think the option might not be High Density. Maybe that is the problem but I am not sure. 

Please guide me on this matter.

akhawaja2014, 2021-05-03:
Thankyou so much, It works!

MartyG-RealSense, 2021-05-03:
Great news @akhawaja2014 - thanks very much for the update!   :)

closed 2021-05-09 by MartyG-RealSense