realsenseai/librealsense #8934: L515 giving an error

mpgerlek-apira, 2021-04-29:
Camera: L515, s/n f0264379
Firmware: 01.05.05.00
OS: Ubuntu 20

When turning on the depth sensor in realsense-viewer v2.44.0, my camera gives an error dialog:

```
Backend in rs2_open_multiple(sensor:0x7fd2c0026b40,
profiles:0x7fd2c0017640, count:2):
xioctl(VIDIOC_S_FMT) failed Last Error: Input/output error
```

Attempts to use the Python SDK with the camera also fail. I have an identical camera that is working just fine in the same configurations.

zeynepkoyun, 2021-05-02:
Hello, what changes have you made in the calibration for the box dimension task?

RealSenseSupport, 2021-05-10:
Hi @mpgerlek-apira 

Are you using the same cable that came with the camera?  Does anything change if and when you change the orientation of the USBC connection going into the L515 camera?  Do you get the same issue if you move to a different system if you are able to do so?  

You can also submit a ticket here:  https://support.intelrealsense.com/hc/en-us/requests/new

mpgerlek-apira, 2021-05-10:
Yes -- different cables, different systems, all give the same error.

RealSenseSupport, 2021-05-25:
Thanks.  Let's close this ticket here since you've opened with support.intelrealsense.com site.

closed 2021-05-25 by RealSenseSupport

marcinolokk, 2021-06-25:
I also encountered the same issue.
Camera: L515, s/n f0246237
Firmware 01.05.08.01
SDK: v2.47.0 (also tried 2.39.0, 2.45.0)
OS: Ubuntu 18.04
kernel: 5.4.0-77-generic (also tired older ones 4.15.xx) 

Always getting the same error:
`Backend in rs2_open_multiple(sensor:0x7f70e40161c0,
profiles:0x7f70e42aec90, count:2):
xioctl(VIDIOC_S_FMT) failed Last Error: Input/output error`

mpgerlek-apira, 2021-06-25:
(Resolution: Intel had me return my unit as defective)