realsenseai/librealsense #8937: Error when compiling matlab wrapper

jeferrandiz98, 2021-04-30:
Hello, I am trying to compile the matlab wrapper on mac OS (Big Sur 11.2.3), but I get the following error:

[ 98%] Linking CXX shared library librealsense_mex.mexmaci64
clang: error: no such file or directory: '/NODEFAULTLIB:realsense2'
make[2]: *** [wrappers/matlab/librealsense_mex.mexmaci64] Error 1
make[1]: *** [wrappers/matlab/CMakeFiles/librealsense_mex.dir/all] Error 2
make: *** [all] Error 2

Does anyone know the reason of this error? Thank you in advance for your help.

MartyG-RealSense, 2021-05-01:
Hi @jeferrandiz98   There is no guidance available for installing the RealSense MATLAB wrapper on Mac unfortunately as the wrapper is designed to be used with Windows 10.

jeferrandiz98, 2021-05-01:
I understand. I was afraid that the wrapper was only designed for Windows 10. Thank you so much for confirming that.

closed 2021-05-01 by jeferrandiz98

MartyG-RealSense, 2021-05-01:
You are very welcome.  I recalled that this subject had appeared in another case recently and tracked down the link.

https://github.com/IntelRealSense/librealsense/issues/8654