realsenseai/librealsense #8938: Align RGB and depth images(raw 16-bit)

Jonapple, 2021-05-01:
| Camera Model | { SR300} |
| Firmware Version | 05.12.12.100 |
| Operating System & Version | Ubuntu 20.04 |
| Kernel Version (Linux Only)    |  5.4.0-72-generic   | 
| Platform | Ubuntu PyCharm, CLion |
| SDK Version | 2.44.0 |
| Language | C++ Python |
| Segment | RGBD/Align |

When I extract depth images from the bag file by using the [convert tool](https://github.com/IntelRealSense/librealsense/tree/master/tools/convert), it seems the depth information will be lost if I choose to extract it into png. And the png image is scaled for visualisation.
[image]

So I used another tool to get pure depth image by using:
https://github.com/UnaNancyOwen/rs_bag2image
At this stage, there is no problem at all.
[image]
raw 16-bit depth image

However, I wish to align the RGB image and raw 16-bit depth image because they are from the different physical viewport.
From the [python sample](https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb) or [C++ sample](https://github.com/IntelRealSense/librealsense/tree/master/examples/align), I found they all directly get streams from the bag file, and I cannot use pre-exported images.

I tried to follow the example using bag files to run the experiments, they work fine and aligned depth image. I'm not aware at this stage, is this tool align frame and display the result as a scaled png (depth information still contains and I need to take them out), or extracted a png from its corresponded frame and performance the alignment (depth information is lost). Is there any way that I can customise the input with a 16-bit depth image by myself?
[image]

Any help will be appreciated :-)

MartyG-RealSense, 2021-05-01:
Hi @Jonapple  If you have the option of using a bag file then I would recommend doing so due to the complications of aligning pre-made depth and color image files.

If you would prefer to use files then the link below has a Python script provided by a RealSense user that saves color as PNG and the depth information as an array of scaled matrices saved as an .npy file and then aligns them.

https://github.com/IntelRealSense/librealsense/issues/4934#issuecomment-537705225 

If you wanted to align color and depth image files then it is likely to involve using **software_device**.   Aside from the complications of matching the images up (discussed in a link below), this will likely require you to base your application on C++ code instead of Python as although software_device is supported in Python, there are not many references to guide in doing so.

**Aligning image files**
https://github.com/IntelRealSense/librealsense/issues/1274#issuecomment-370228022

**software_device**
https://github.com/IntelRealSense/librealsense/tree/master/examples/software-device

**Readng a depth .raw file**
https://github.com/IntelRealSense/librealsense/issues/2534#issuecomment-430912589

In regard to your question about the **distance_to_object** example program, it is not converting the bag's depth information to PNG.  It is reading the camera data from the bag file as though it were accessing a live camera depth stream, as a bag contains uncompressed and unfiltered camera data.

Jonapple, 2021-05-06:
> Hi @Jonapple If you have the option of using a bag file then I would recommend doing so due to the complications of aligning pre-made depth and color image files.
> 
> If you would prefer to use files then the link below has a Python script provided by a RealSense user that saves color as PNG and the depth information as an array of scaled matrices saved as an .npy file and then aligns them.
> 
> [#4934 (comment)](https://github.com/IntelRealSense/librealsense/issues/4934#issuecomment-537705225)
> 
> If you wanted to align color and depth image files then it is likely to involve using **software_device**. Aside from the complications of matching the images up (discussed in a link below), this will likely require you to base your application on C++ code instead of Python as although software_device is supported in Python, there are not many references to guide in doing so.
> 
> **Aligning image files**
> [#1274 (comment)](https://github.com/IntelRealSense/librealsense/issues/1274#issuecomment-370228022)
> 
> **software_device**
> https://github.com/IntelRealSense/librealsense/tree/master/examples/software-device
> 
> **Reading a depth .raw file**
> [#2534 (comment)](https://github.com/IntelRealSense/librealsense/issues/2534#issuecomment-430912589)
> 
> In regard to your question about the **distance_to_object** example program, it is not converting the bag's depth information to PNG. It is reading the camera data from the bag file as though it were accessing a live camera depth stream, as a bag contains uncompressed and unfiltered camera data.

Hi @MartyG-RealSense Thank you for all the information, they really helpful. I've successfully aligned them using bag and extract them into 16bit depth images.
I got another problem, does all frames in a bag file can be aligned? Because in my bag file, if I use the SDK tool to extract png, there will be 276 depth images and 276 colour rbg images as well. I used a for loop to get frams from bag, code as below:

max_frames = 276
for idx in range(max_frames):
    frameset = pipe.wait_for_frames()
    frameset = align.process(frameset)
    color_frame = frameset.get_color_frame()
    depth_frame = frameset.get_depth_frame()

After align and extract depth images, I found that these 276 images are not the same as directly extracted by using the [SDK's convert tool](https://github.com/IntelRealSense/librealsense/tree/master/tools/convert). If I modify the max_frams to any value beyond 276, it still gets output images.

Am I using the wrong way in getting all frames in the bag file?

Thanks in advance.

MartyG-RealSense, 2021-05-06:
I don't think your bag reading code is necessarily wrong compared to **rs-convert**, as it sounds as though your code is reading the frames from the bag directly rather than converting them to files and so it is not a like-for-like comparison.

**rs_bag2image** does have a good reputation amongst RealSense users who have tried both it and rs-convert, so its results can be relied on.

In regard to achieving optimal reliability in your bag-reading script, a couple of recommendations would be:

1.  Set **set_real_time** to False if you have not done so already.

2. Set **repeat_playback** to False if you have not done so already.  This causes playback to stop at the last frame of the bag and not loop around to the start again.

https://github.com/IntelRealSense/librealsense/issues/4344#issuecomment-507387892

Jonapple, 2021-05-12:
> I don't think your bag reading code is necessarily wrong compared to **rs-convert**, as it sounds as though your code is reading the frames from the bag directly rather than converting them to files and so it is not a like-for-like comparison.
> 
> **rs_bag2image** does have a good reputation amongst RealSense users who have tried both it and rs-convert, so its results can be relied on.
> 
> In regard to achieving optimal reliability in your bag-reading script, a couple of recommendations would be:
> 
> 1. Set **set_real_time** to False if you have not done so already.
> 2. Set **repeat_playback** to False if you have not done so already.  This causes playback to stop at the last frame of the bag and not loop around to the start again.
> 
> [#4344 (comment)](https://github.com/IntelRealSense/librealsense/issues/4344#issuecomment-507387892)

Thank you! Problem solved :)

closed 2021-05-12 by Jonapple

MartyG-RealSense, 2021-05-12:
Great news @Jonapple that you were successful - thanks for the update!