realsenseai/librealsense #8939: D435i color camera 30fps when set 60fps on 848x480 resolutions

dsvua, 2021-05-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i | 
| Firmware Version                   | 05.12.12.100 | 
| Operating System & Version |   Ubuntu 20.04 | 
| Kernel Version (Linux Only)    |  5.8.0-50-generic  | 
| Platform                                 | PC |
| SDK Version                            |  current master branch  | 
| Language                            | c++ | 
| Segment			|  Robot  | 

### Issue Description
<Describe your issue / question / feature request / etc..>
In Realsense Viewer and in my C++ code I set color camera stream to 848x480 resolution and 60 fps for D435i. However, effective FPS as could be see on screen shot is 30 fps. Same for C++ code, I am getting 30fps for color stream.

Are there any changes in firmware?

[image]

MartyG-RealSense, 2021-05-01:
Hi @dsvua   Please try disabling a setting called **Auto-Exposure Priority** in the RGB section of the Viewer's options side-panel.  When Auto-Exposure is enabled and Auto-Exposure Priority is disabled then a constant FPS should be enforced.

If this method works for you, the link below provides advice about disabling Auto-Exposure Priority in C++.

https://github.com/IntelRealSense/librealsense/issues/5290

dsvua, 2021-05-03:
Thank you Marty, solution in comment works for me!

closed 2021-05-03 by dsvua