realsenseai/librealsense #8947: Minimum distance of depth of L515

JANGSOONMYUN, 2021-05-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 

In the specification of L515, the minimum depth distance is 25cm. 
However, I've found an option to change minimum distance which can be lower than 25cm in the RealSense Viewer. So I set it 0, and then checked depth images. It showed minimum depth was less than 5 cm. (But, when it was too close, the process stopped)

I don't understand why the specification said the min-depth is 25cm, even though it can be lower.
Is there any reason users should follow the specification?

zeynepkoyun, 2021-05-03:
@JANGSOONMYUN  Does the height value change in you while viewing the caliber field? Have you been able to measure the box dimensions exactly?

JANGSOONMYUN, 2021-05-04:
@zeynepkoyun Sorry I don't get it. What is the box?
The height of the camera isn't changed.

I just want to know why the specification and the real case are different.
The specification says its minimum working distance of the depth image is 25cm.
However, my target working distance is 5cm~5M.  And I was able to set the minimum working distance lower than 25cm in the Intel RealSense Viewer.

I just guess the reason of 25cm limitation is because the focus is out when the distance is smaller than 25cm. So that it has low accuracy in 0~25cm range. Ant that's why the specification determined 25cm is the minimum distance for the both depth image and color image.

But not really sure.

zeynepkoyun, 2021-05-04:
Firstly, thanks for your response.

While trying to run the "https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python/examples/box_dimensioner_multicam" sample project with the L515 camera, I get a dimension error. I cannot measure the values exactly.

To explain with an example:
[image]

JANGSOONMYUN, 2021-05-06:
@zeynepkoyun I don't think you did wrong. As you see [this](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/box_dimensioner_multicam/samplesetupandoutput.jpg), the sample result also shows inaccurate boxes.

I am not sure the reason, but I guess it is because depth image has low resolution and low accuracy. The depth camera can detect depth with about 1% error according to the distance. And especially, the depth image shows bad boundaries of objects. That's why it cannot show the correct box.

zeynepkoyun, 2021-05-06:
Cevabınız için teşekkürler @JANGSOONMYUN @RealSenseSupport 
https://github.com/IntelRealSense/librealsense/issues/7335  , aynı derinlik değerine göre elde edilen piksel değerlerinden aynı noktaları alamadığımız söyleniyor. Bu durumu test ettim ve gerçek bir değişiklik oldu ve ardından araştırma sürecine başladım.

Bir örnekle açıklamak için:
(X, y) noktalar için (derinlik değeri 1) -> (pixel_x, pixel_y)
(pixel_x, pixel_y), (1 derinlik için) -> (x1, y1).

closed 2021-06-09 by RealSenseSupport