realsenseai/librealsense #8951: How to use multiple L515 LiDAR camera properly in Inter RealSense Viewer

DragonDeux, 2021-05-03:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 LiDAR} | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win 10 | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  2.0 }                          | 
| Language                            |                            | 
| Segment			|   }                          | 

### Issue Description
Hi, 

Could you please clarify how to use two L515 LiDAR cameras with external hardware triggers? Assumed that we have already prepared the external trigger source that generates the pulses (high-low, pulse width = 40ms) for two camera separately.

In this document:  https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup, it seems like you have already done the demonstration in the Viewer to show the no interference multi-LiDAR using case:

[image]

Many appreciate for helping or even any hints about this question.

Regards

DragonDeux, 2021-05-04:
Can somebody help to resolve why these two LiDAR is free-run?

DragonDeux, 2021-05-04:
Can somebody help to resolve why these two LiDAR is free-runing?

DragonDeux, 2021-05-05:
Can someone help me to just clarify how to use the external trigger for the only one L515 camera?!!! It was working, but suddenly turn to free-run, very strange! Thank you very much. Was that a kind of bug for this production?

RealSenseSupport, 2021-05-13:
Hi @DragonDeux 

Did you open another thread on this topic?  

When you say that the trigger was working for sometime and now it isn't working, can you help explain what you were seeing and for how long it was working for before the behavior changed?

closed 2021-06-10 by RealSenseSupport