realsenseai/librealsense #8956: Cannot connect F455 to Android phone

PavKovalenko, 2021-05-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | F455 | 
| Operating System & Version |   Android                                                | 
| Platform                                 |   Android                                               |
| SDK Version                            |  2.45                           | 
| Language                            | Java                          | 
| Segment			| Smartphone                          | 

### Issue Description
Create sample from doc
https://github.com/IntelRealSense/librealsense/blob/development/wrappers/android/examples/java_example/readme.md
But my device didnt detect my Camera. A isee that phone see attached hardware but looks like doesnt recognise camera.
In logs, when attach camera:

> I/librs Enumerator: onReceive: android.hardware.usb.action.USB_DEVICE_ATTACHED
> W/librs UsbUtilities: getUsbDevice: failed to locate USB device, VID: 0x8086, PID: 0x0000
> I/librs MessagesHandler: handleMessage: realsense device attached
> I/librs Enumerator: notifyOnAttach

But `setDevicesChangedCallback-> onDeviceAttach()` have never called. And `RsContext.queryDevices().deviceCount` is 0

Tried to install app 'RS camera' from PlayStore and jfrog repo(https://egiintel.jfrog.io/ui/repos/tree/General/librealsense). Same result - apps did not see connected device.

remibettan, 2021-05-04:
Hi, 
If you are using D455: please fill the missing fields in the required info table, so that we would have all the required information to help you, as the android version.
If you are using F455: this peripheral is a completely separate device not supported with LRS SDK.
Please refer to Documentation and Code Samples link in the following link:
https://store.intelrealsense.com/buy-intel-realsense-id-solution-f455-for-facial-authentication.html

PavKovalenko, 2021-05-04:
It is correct  - I am using F455. Thank you for clarification.
Does it mean that I can try C++ samples on Android or it is only for Windows platform?

remibettan, 2021-05-05:
The best thing you can do is to ask these questions to the real owner.
Please reopen your issue in:
https://github.com/IntelRealSense/RealSenseID/issues

closed 2021-05-06 by PavKovalenko