realsenseai/librealsense #8958: Hardware sync recording using RealSense-viewer

LuckyFace, 2021-05-04:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> 05.12.12.100) | 
| Operating System & Version |   {Win 10 } |                              | 
| Platform                                 | PC |

### Issue Description
Hi, I am trying to use multiple D435 cameras to acquire synchronized RGB videos.
I tried hardware sync and connected cameras as posted in the whitepaper.
I launched realsense-viewer and set as one camera to master (inter cam sync mode - 1) and others as slave (inter cam sync mode - 2). Then I turned on stereo and rgb together and pushed the record button of the master camera. But I get only recorded videos of the master camera. Has anyone have experience on multi camera recording with realsense-viewer?
Is multi-camera hw-sync only possible with C++ or python based SDK?

MartyG-RealSense, 2021-05-04:
Hi @LuckyFace  The record button for each camera on a multi-camera setup in the Viewer only records a bag file for the streams of that specific camera.

Also, if you are using '2' as the slave setting then only depth will be synchronized.  To enable color camera sync too on D435, you should use '3' on the slave cameras to set them to **Full Slave** mode.

https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras#section-appendix-a

There are a couple of things to bear in mind.   

-  The '3' setting is part of a 'genlock' multicam feature that is not validated by Intel and is not considered 'mature', whilst the original multicam sync system (which covers 1 and 2 only) is considered mature.

-  As mentioned in the Full Slave notes in the above link, the FPS on the slave cameras should be set higher than the FPS of the master camera.

If you wish to explore writing your own multicam recording application, the information listed in the link below is a good place to start.

https://github.com/IntelRealSense/librealsense/issues/6303#issuecomment-619600661

sam598, 2021-05-05:
@MartyG-RealSense My understanding is that mode '3' is still a placeholder and does not actually work on any RealSense camera.
https://github.com/IntelRealSense/librealsense/issues/8419#issuecomment-784461663

Has this changed?

MartyG-RealSense, 2021-05-05:
@sam598  I have not currently got any further information about genlock.

sam598, 2021-05-05:
@MartyG-RealSense I am talking about **Full Slave** mode '3' not **GenLock** modes '4+'.

@LuckyFace is asking to synchronize RGB video between multiple cameras. The documentation says that requires **Full Slave** mode. But the documentation does not reflect the reality of the current hardware and software.

In this thread about the D455 https://github.com/IntelRealSense/librealsense/issues/8419#issuecomment-784461663 @RealSenseSupport says "Mode 3 is NOT mature yet. RGB sync will be implemented later.".

Regarding the D435 in your previous comment https://github.com/IntelRealSense/librealsense/issues/5861#issuecomment-586844971 you qouted @dorodnic saying "D435 allows Depth to Color sync, and Depth to another Depth sync, but not Color to another Color, so at least one (or both) of the color sensors will not be synced"

It has been said previously that the D435 and D415 have a physical hardware limitation that prevents the color camera from ever being hardware synced, and that there was no plan to fix this.

So unless something has changed, the answer to @LuckyFace should be that no, it is not possible to sync RGB video of the D435. He will have to buy new cameras and wait for the firmware update.

MartyG-RealSense, 2021-05-05:
@sam598   I do not have any additional information to provide beyond what is stated in the External Synchronization paper.  Given that the genlock system described in that paper is unvalidated by Intel though, it may be best to avoid use of the genlock system for the time being or making project plans based upon it until the feature becomes mature.

An alternate approach to syncing recordings from multiple cameras that has been discussed in the past is to use camera metadata.

https://github.com/IntelRealSense/librealsense/issues/2186

sam598, 2021-05-05:
@MartyG-RealSense I am not talking about the **Genlock mode**. **Full Slave mode** was part of the **Genlock update** that includes **Genlock mode**. I understand neither modes are "mature", and the desire to lump them together.

But these are **VERY** different camera modes with different features and limitations. Please stop calling **Full Slave** **"Genlock"** as it further confuses the situation.

sam598, 2021-05-05:
You had suggested that @LuckyFace use the **Full Slave** mode, but did not mention if it will work with their camera, or if it even works at all. If you do not have additional information it may be prudent to get some from your coworkers, before potentially sending a user on a "wild goose chase".

Using the frame timestamps as a reference is a good suggestion and could be useful, but it does not give synchronized exposures.

MartyG-RealSense, 2021-05-05:
@sam598  I genuinely understand your frustrations about genlock, but please refrain from leaving comments about it on multiple cases that others have created because it diverts attention from that person's own issues.  You are welcome to create your own issue to follow up about your concerns, but the reply is not going to change from previous answers.

sam598, 2021-05-05:
@MartyG-RealSense I really appreciate all of the work you do, but please read my comments in this specific thread, and please don't assume that I am always commenting about genlock.

This is not about genlock. It is not relevant to this issue. You are the one who keeps bringing up genlock in this thread, not me, and it is very confusing.

I am trying to get @LuckyFace as much information as possible about hardware syncing RealSense RGB cameras. According to past issues this does not seem to be possible with any device, mode or configuration. Am I wrong?

MartyG-RealSense, 2021-05-05:
@sam598   I acknowledge the point that technically, Genlock refers to mode 4+, whilst 3 is Full Slave.  I tend to refer to the overall system in the paper as "External Synchronization (Genlock)" because it sums up in a single phrase the main functions of the system.  Section 4 Point 3 of the paper also places references about syncing with RGB under the 'Genlock' heading.  

https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras#section-4-genlock-usages

So I should be more careful about that distinction between 3 and 4+ in the future.  Thanks!

sam598, 2021-05-05:
Thank you @MartyG-RealSense, I appreciate that.

So if you were saying that there is no new information about Mode 3 or the hardware limitation on the D435, then there is currently no way for @LuckyFace to hardware sync multiple RGB cameras? 

Their best solution is to take the metadata timestamps and to try and interpolate the data that they need?

MartyG-RealSense, 2021-05-05:
Multi-cam recording methods in ROS and Python, such as using **rosbag record -all** in ROS to record all active topics or starting recordings of all cameras at the same time in Python, are discussed in the link below.  They are not hardware-synced though, so metadata may be the best route if RGB syncing-up is a necessity.  I do not have another option to suggest for achieving it.

https://github.com/IntelRealSense/librealsense/issues/8062

sam598, 2021-05-05:
Alright. I saw in the latest version of the SDK there were some undocumented updates to INTER_CAM_SYNC.

I wrote this script to test the offset of RGB and Depth frames, and I did several tests with the latest SDK (2.45.0) and firmware (5.12.13.50)

```
import pyrealsense2 as rs

ctx = rs.context()
device_idx = 0
device_name = "{insert serial number here}"
for i in range(len(ctx.devices)):
    sn = ctx.devices[i].get_info(rs.camera_info.serial_number)
    print(sn)
    if sn == device_name:
        device_idx = i

pipeline = rs.pipeline()
config = rs.config()

config.enable_device(device_name)

config.enable_stream(rs.stream.depth, 848, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 848, 480, rs.format.rgb8, 30)

exposure = 5000

sensor = ctx.devices[device_idx].query_sensors()[0]
sensor.set_option(rs.option.inter_cam_sync_mode, {insert sync mode here})
sensor.set_option(rs.option.exposure, exposure)

sensor = ctx.devices[device_idx].query_sensors()[1]
sensor.set_option(rs.option.exposure, exposure / 100)

cfg = pipeline.start(config)

depth_time = 0
color_time = 0

try:
    while True:
        frames = pipeline.wait_for_frames()
        depth = frames.get_depth_frame()
        color = frames.get_color_frame()

        if depth:
            depth_time = depth.get_frame_metadata(rs.frame_metadata_value.sensor_timestamp)

        if color:
            color_time = color.get_frame_metadata(rs.frame_metadata_value.sensor_timestamp)

        print(abs(depth_time - color_time))

finally:
    pipeline.stop()
```

The results are.... unexpected.

**Two D435 cameras**

```
Test #1

Camera A - Mode "0" - Offset consistent around 2500us
Camera B - Mode "0" - Offset consistent around 2500us
```

```
Test #2

Camera A - Mode "1" - Offset drifts between 0 - 17000us
Camera B - Mode "0" - Offset consistent around 2500us
```

```
Test #3

Camera A - Mode "1" - Offset drifts between 0 - 17000us
Camera B - Mode "2" - Offset drifts between 0 - 17000us
```

```
Test #4

Camera A - Mode "1" - Offset drifts between 0 - 17000us
Camera B - Mode "3" - Offset drifts between 0 - 17000us
```

**Two D455 cameras**

```
Test #1

Camera A - Mode "0" - Offset drifts between 0 - 17000us (?)
Camera B - Mode "0" - Offset drifts between 0 - 17000us
```

```
Test #2

Camera A - Mode "1" - Offset consistent around 3850us (!?)
Camera B - Mode "0" - Offset drifts between 0 - 17000us
```

```
Test #3

Camera A - Mode "1" - Offset consistent around 3850us
Camera B - Mode "2" - Offset consistent around 3850us (!!!)
```

```
Test #4

Camera A - Mode "1"  -Offset consistent around 3850us
Camera B - Mode "3" - Offset drifts between 0 - 17000us (???)
```

So, in practice this seems to be inconsistent with both the documentation and the advice from Intel.

I am thrilled that there is now a RealSense camera capable of RGB sync, but now the rabbit hole goes deeper...

LuckyFace, 2021-05-06:
Thank you very much for your comments sam598 and MartyG-RealSense.
Since the synced RGB videos are my primary objective, I will try to use metadata for RGB sync with simultaneous recording command like "rosbag record -all" in ROS. 
Really hope for advances in RGB sync in a simpler way (e.g. within the viewer) in the future.

MartyG-RealSense, 2021-05-06:
Thanks very much @LuckyFace for the feedback - good luck!

closed 2021-05-18 by MartyG-RealSense