realsenseai/librealsense #8959: Decimation filter creating patches

mathislm, 2021-05-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | 05.12.13.50 | 
| Operating System & Version |   Linux (Ubuntu 20) | 
| Kernel Version (Linux Only)    |  5.8.0-50-generic  | 
| Platform                                 | PC |
| SDK Version                            |  2.45.0 | 
| Language                            |  python |
| Segment			|  others | 

### Issue Description
I've apply applying a decimation filter to my frames this way : 
```python
decimate_filter = pyrealsense2.decimation_filter(2)
frames = pipeline.wait_for_frames()
frames = decimate_filter.process(frames).as_frameset()
frames = align.process(frames)

depth_frame = frames.get_depth_frame()
color_frame = frames.get_color_frame()
```
But it doesn't reduces the amount of points. Instead, it is creating patches of 3*3 points with the same depth. I suppose it's to conserve the full color information while reducing the depth complexity, but I would rather have an actual decimation filter that reduces the number of points. Am I supposed to do that myself or is there an option I am missing ?

mathislm, 2021-05-04:
[image]

an exemple of those patches, with a decimation magnitude parameter of 4.

MartyG-RealSense, 2021-05-04:
Hi @mathislm   An example of Python scripting for decimation and align in the link below may be a useful reference.

https://github.com/IntelRealSense/librealsense/issues/2356#issuecomment-465766323

mathislm, 2021-05-04:
Thank you.
Though, this is what I am doing in the code I shared I think. Am I missing something ?

MartyG-RealSense, 2021-05-04:
They are structurally similar, though I note that your script has not got **align_to** set up before calling **align.process**

[image]

mathislm, 2021-05-05:
My bad, I forgot to add that in the issue description. This is how I declare my variable `align` :
```python
align = pyrealsense2.align(pyrealsense2.stream.color)
```

MartyG-RealSense, 2021-05-05:
No problem.  How does it behave if you use a decimation value of '3' instead of '2'?

MartyG-RealSense, 2021-05-16:
Hi @mathislm  Do you require further assistance with this case, please?  Thanks!

mathislm, 2021-05-17:
Hello,
I understand with this previous link that the api doesn't really provide a way to apply filters "frame-wide". This means I'll have to apply my filters on myself, with scipy maybe. It is a bit troublesome but I'll manage. Thank you.

MartyG-RealSense, 2021-05-17:
You are very welcome @mathislm - good luck!

closed 2021-05-17 by mathislm