realsenseai/librealsense #8965: The problem when D435 turns off auto exposure

b2992872, 2021-05-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Linux (Ubuntu 18) /   | 
| Kernel Version (Linux Only)    |  ( 4.14)                                         | 
| Platform                                 |  NVIDIA Jetson   |
| SDK Version                            |  {  2.0 }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {example }                          | 

### Issue Description
As in the title, I want to turn off the automatic exposure and manually set the exposure value
[image]
I used the following program，The result I get is still automatically exposed，I don't know what went wrong
"sensor .set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE,1);"
"sensor.set_option(RS2_OPTION_EXPOSURE,100);"
[image]

MartyG-RealSense, 2021-05-05:
Hi @b2992872   To turn Auto-Exposure off, it should be set to '0' instead of '1', as 0 is disabled and 1 is enabled.

**sensor.set.option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 0);**

There are a couple of other factors to bear in mind:

1.  There are separate Auto-Exposure settings for depth and color.  In your code, you are accessing the depth auto-exposure.

2.  Exposure values for depth and color use different scales of size, with the depth exposure typically having larger values.  For example, the default depth exposure value is '8500' and the default color exposure is '156'.   Your code sets an exposure value of '100' but is setting it for the depth exposure and not the color exposure.  Is this intentional, please?

b2992872, 2021-05-05:
@MartyG-RealSense 
I set it to 0 at the beginning, but I tried to change it back to 1 when I found it was invalid. I’m sorry for themisunderstanding。

 I want to manually adjust the exposure because I hope to increase the FPS. The current FPS is too low. I don’t know what effect the depth exposure and color exposure have on FPS. ?

MartyG-RealSense, 2021-05-05:
When auto-exposure is enabled and the color stream is being used, you can enforce a constant FPS instead of allowing the FPS to vary if you disable an RGB option called **Auto-Exposure Priority**.  You can disable Auto-Exposure Priority in C++ using code in the link below.

https://github.com/IntelRealSense/librealsense/issues/5290

If you need auto-exposure to be disabled but a constant FPS to be enforced when manual exposure is used then the link below describes the maths for choosing an exposure value.

https://github.com/IntelRealSense/librealsense/issues/1957#issuecomment-400715579

b2992872, 2021-05-06:
After disabling auto exposure, do I have to set a fixed exposure value like this
[image]

MartyG-RealSense, 2021-05-06:
By using an index number of '1' with query_sensors() you are accessing the RGB sensor and so the RGB exposure (as the index number for the depth sensor is '0').  As '8500' is the default value for the separate **depth** exposure and the RGB exposure uses a smaller scale of values, setting RGB exposure to this high value may be causing your display problem.

In a test in the RealSense Viewer, setting RGB exposure to '8500' caused the image to become bright white and the FPS speed to slow to a crawl.

[image]

b2992872, 2021-05-06:
Thank you!, I will try again

closed 2021-05-10 by b2992872