realsenseai/librealsense #8966: network device align not work

tsengyushiang, 2021-05-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D415 | 
| Firmware Version                   | 05.12.12.100 | 
| Operating System & Version |   {Win ( 10) / Linux (Ubuntu 16.04)   | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC  |
| SDK Version                            |  v2.40 viewer, v2.44 build from soruce code                           | 
| Language                            |  C/C++                          | 
| Segment			|                          | 

### Issue Description
- I use a ubuntu16.04 mini PC with `rs-server` to send realsense data to my win10 PC
- and use following code to start remote realsense device
```
    dev = new rs2::net_device(url);
    dev->add_to(ctx);
    pipe = new rs2::pipeline(ctx);

    cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
    cfg.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_BGR8, 30);

    rs2::pipeline_profile profile = pipe->start(cfg);
```
- it successfully get frames back form mini PC, but when i use ` rs2::align align_to_color{ RS2_STREAM_COLOR };` for align depth to color I only get the depth image has the same size as color image.
- I also use official viewer to check the issue and I get a similar [result](https://imgur.com/a/uHqZhqt)

MartyG-RealSense, 2021-05-05:
Hi @tsengyushiang  What you are describing is normal when using depth to color alignment.  The depth field of view size is aligned to the size of the color field of view.  

More information about the effects of alignment can be found in the opening paragraph of the documentation for the **rs-align** SDK example program.

https://github.com/IntelRealSense/librealsense/tree/master/examples/align

I would also recommend using librealsense **2.45.0** if you plan to test a Network Camera connection in the Viewer, as this SDK version contains a fix for connecting a Network Camera to the Viewer.

https://github.com/IntelRealSense/librealsense/pull/8751

tsengyushiang, 2021-05-05:
- I just test the v2.45.0 viewer on my network camera, it's still have some misalignment between depth and color.
- snapshot shows point cloud color on my finger is not correct, some skin color is on and background wall.
[image]

- and I try to use v2.44.0 SDK with opencv overlay depth and image found out that depth and color are not align well, on left side I draw image with opencv, and right side I use intrinsic parameters to project them in 3d space and render with opengl.

[image]

```
glm::vec3 RealsenseDevice::colorPixel2point(glm::vec2 pixel) {
    int i = pixel.y;
    int j = pixel.x;
    int index = i * width + j;

    float depthValue = (float)p_depth_frame[index] * intri.depth_scale;

    glm::vec3 point(
        (float(j) - intri.ppx) / intri.fx * depthValue,
        (float(i) - intri.ppy) / intri.fy * depthValue,
        depthValue
    );

    return point;
}

bool RealsenseDevice::fetchframes() {

    rs2::frameset frameset; // Wait for next set of frames from the camera

    if (pipe->poll_for_frames(&frameset)) {
        rs2::frameset data = align_to_color.process(frameset);
      
        rs2::frame depth = data.get_depth_frame();
        rs2::frame color = data.get_color_frame();

        memcpy((void*)p_depth_frame, depth.get_data(), width * height * sizeof(uint16_t));
        memcpy((void*)p_color_frame, color.get_data(), 3 * width * height * sizeof(uchar));

        // copy to memory
        for (int i = 0; i < height; i++) {
            for (int j = 0; j < width; j++) {
                int index = i * width + j;

                glm::vec3 point = colorPixel2point(glm::vec2(j,i));
                vertexData[index * 6 + 0] = point.x;
                vertexData[index * 6 + 1] = point.y;
                vertexData[index * 6 + 2] = point.z;

                vertexData[index * 6 + 3] = (float)p_color_frame[index * 3 + 2] / 255;
                vertexData[index * 6 + 4] = (float)p_color_frame[index * 3 + 1] / 255;
                vertexData[index * 6 + 5] = (float)p_color_frame[index * 3 + 0] / 255;                
            }
        }

        if (opencvImshow) {
            rs2::frame depthColorize = depth.apply_filter(color_map);

            const int w = depthColorize.as<rs2::video_frame>().get_width();
            const int h = depthColorize.as<rs2::video_frame>().get_height();

            const int cw = color.as<rs2::video_frame>().get_width();
            const int ch = color.as<rs2::video_frame>().get_height();

            cv::Mat image(cv::Size(w, h), CV_8UC3, (void*)depthColorize.get_data(), cv::Mat::AUTO_STEP);
            cv::Mat image2(cv::Size(cw, ch), CV_8UC3, (void*)color.get_data(), cv::Mat::AUTO_STEP);

            cv::Mat dst;
            addWeighted(image, 0.5, image2, 0.5, 0.0, dst);
            // Update the window with new data
            imshow(serial.c_str(), dst);
        }
        else {
            cvDestroyWindow(serial.c_str());
        }
        return true;
    }
    return false;
}
```

MartyG-RealSense, 2021-05-05:
In my experience the effect that you observe with the hand (small color hand and over-sized depth 'occlusion' silhouette) is normal for observing the hand at close range in the Viewer.  As the hand is moved further away from the camera the black silhouette should shrink to more tightly fit the color image of the arm.

The last time that I tested this phenomenon, I found that using a **Depth Unit Scale** of 0.0001 instead of the default 0.001 value for this setting helped to reduce the size of the outline when the hand is close to the camera.

https://github.com/IntelRealSense/librealsense/issues/8228

tsengyushiang, 2021-05-05:
I think this effect only show when using network device, when connect with usb3.0 and use same config it work perfect.
- network :
[image]
- usb3.0
[image]

I just try setting Depth Unit Scale to 0.0001, but there is small different from default 0.001 value.
[image]

MartyG-RealSense, 2021-05-05:
Are you able to tell whether the color image is updating slower than the depth image?  **rs-server** is currently only supporting a maximum FPS of 6 when using 1280x720 color resolution.  640x480 color at 30 FPS is supported.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#section-3-6-software-limitations

tsengyushiang, 2021-05-05:
I use 640x480 for both color/depth camera, and didn't move my hand. here is another snapshot for a static scene. and the artiface is locate at the boundary of the water bottle.
[image]

I am not sure does this effect cause by fps. Is there any suggestion to make sure I'm using the correct depth and image at the same time?

MartyG-RealSense, 2021-05-05:
Ideally an occlusion silhouette should be evenly distributed around an object so that the width of the shadow is roughly equal on both sides of the object.  The position of the occlusion on the arm and the bottle both seem to be biased towards one particular side, even on the USB 3 images.

It may be worth performing a **Dynamic Calibration** of the camera, which provides a robust calibration of both the depth and RGB sensors in order to eliminate mis-calibration between depth and RGB as a cause.  The Dynamic Calibration software tool can be downloaded for Windows from the link below.

https://downloadcenter.intel.com/download/29618/Intel-RealSense-D400-Series-Dynamic-Calibration-Tool

If you prefer to use Linux, page 14 onwards of the user guide for the tool has Linux installation instructions.

https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

In regard to checking the resolution and FPS settings during runtime to see whether they have changed after enablling the streams: you could do so in the Viewer by starting the depth and RGB streams in **2D** mode and then left-clicking on the 'i' icon on top of each stream panel to turn it blue (On) and generate live-updating stream information that shows resolution and FPS.

Note that the depth stream readout typically shows a resolution that is half the size of the one that is actually set (eg 424x240 instead of 848x480).  So this is nothing to be concerned about.

[image]

tsengyushiang, 2021-05-05:
Thanks you @MartyG-RealSense for the reply.

I just try Dynamic Calibration tool with [guide](https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html) and get following results frame after calibrate.
- from the network (rs-server) still has mis-calibration between depth and RGB
[image]
- from USB3.0
[image]

Maybe something go wrong when sending configuration from rs-server v2.44.0?

yzkzero, 2021-05-06:
Hi,

I met the same issue when using rs-server. The issue also existed in earlier version.

Our solution is to implement our own alignment algorithm rather than using librealsense SDK's. But it would be nice if SDK can solve the issue.

tsengyushiang, 2021-05-06:
Hi, @yzkzer

Thank you for pointing out the issue , have you found any solution or workaround ?

MartyG-RealSense, 2021-05-06:
@yzkzero  Yes, thanks so much for sharing your experience with @tsengyushiang  :)

tsengyushiang, 2021-05-06:
Hi, @MartyG-RealSense 
Do you know when will librealsense SDK  start fixing the issue? 
Should I implement our own alignment algorithm just like @yzkzero ,if so and I will close this issue.

MartyG-RealSense, 2021-05-06:
Hi @tsengyushiang   I do not have information about planned fixes to LRS-Net, though updates do get released (such as the fix in SDK 2.45.0 for using the camera as a network camera in the RealSense Viewer, which had problems in 2.43 and 2,44).

The next generation of networking support being worked on that supports a greater number of modes is **LRS-Net 2.0**.

https://github.com/IntelRealSense/librealsense/pull/8343

There is not a time estimate available for when it will be released as part of librealsense though, so in the meantime it may be best to implement your own solution.

tsengyushiang, 2021-05-06:
thanks you @MartyG-RealSense . I will try SDK v2.45.0 and  try to implement my own solution

closed 2021-05-06 by tsengyushiang

MartyG-RealSense, 2022-01-07:
Hi everyone,

A RealSense user at https://github.com/IntelRealSense/librealsense/issues/10132 also experienced depth-color alignment problems on an rs-server network connection and found that a workaround to obtain correct alignment was to hard-code the intrinsic and extrinsic values, as described in https://github.com/IntelRealSense/librealsense/issues/10132#issuecomment-1007245155

I have highlighted the new information to Intel.

MartyG-RealSense, 2022-01-07:
Intel provided feedback to me.   An ethernet side branch of the SDK will be released soon, so they recommend waiting to see whether that fixes the alignment issue when using networking.  A time schedule for release of the side branch is not currently available.