realsenseai/librealsense #8968: Couldn't resolve request problem

DuyguSerbes, 2021-05-05:
I am using D435i realsense. When I run the code I have below error. Can you explain the reason of the problem?

1 devices have been found
Traceback (most recent call last):
  File "/home/duygu/PycharmProjects/realsense/main.py", line 283, in <module>
    device_manager.enable_all_devices()
  File "/home/duygu/PycharmProjects/realsense/main.py", line 149, in enable_all_devices
    self.enable_device(serial, enable_ir_emitter)
  File "/home/duygu/PycharmProjects/realsense/main.py", line 135, in enable_device
    pipeline_profile = pipeline.start(self._config)
RuntimeError: Couldn't resolve requests
   
My OS: Ubuntu 18.04

 * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2021-05-05:
Hi @DuyguSerbes  Advice provided in the link below, where a RealSense user encountered the same errors and also experienced a **Couldn't resolve request** error, should be a good place to start in investigating your problem.

https://github.com/IntelRealSense/librealsense/issues/2818

DuyguSerbes, 2021-05-05:
Hello again,

When I reviewed #2818. The possible reason of this issue would be the use of USB2 instead of USB3. I plugged my camera to super speed usb 3. But it behaves like plugging to USB 2. Also, in realsense viewer, I cannot open Infrared 2 option. Only one infrared listed in the menu. But as far as I know, 2 infrared should be there. @MartyG-RealSense said previously it happens due to again USB 2.0. But I have used SS.

MartyG-RealSense, 2021-05-05:
The absence of Infrared 2 would indicate that the connection is being detected as USB 2, as Infrared 2 is only available on a USB 3 connection.  Sometimes a camera may be mis-identified as USB 2 even if plugged into a USB 3 port.  

You could try adding a **hardware_reset()** instruction to your Python script to see if it helps the camera to be detected correctly as being on USB 3 when the program launches.

https://github.com/IntelRealSense/librealsense/issues/3329#issuecomment-475293475

In the RealSense Viewer, you can test whether a reset makes a difference by going to the **More** option near the top of the options side-panel and selecting the **Hardware Reset** option from its menu.

[image]

DuyguSerbes, 2021-05-05:
I try it on realsense viewer with different USB sockets. But it didn't work unfortunately.

Also currently I start to work with python. What is the python library of harware_reset() funtion? 

Thanks

MartyG-RealSense, 2021-05-05:
Python code is mentioned in this link:

https://github.com/IntelRealSense/librealsense/issues/3329#issuecomment-475293475

```
ctx = rs.context()
devices = ctx.query_devices()
for dev in devices:
dev.hardware_reset()
```

DuyguSerbes, 2021-05-06:
Actually it did not solve my problem. Is there any other way to ensure my connection on USB 3.X? I have tried on both realsense-viewer and python script.

MartyG-RealSense, 2021-05-06:
You could possible avoid the **Couldn't resolve requests** error if you use a resolution and FPS that is supported on both USB 2 and 3, so that if the camera is mis-detected as USB 2 then the error should not be triggered.  Such a resolution that is supported on both USB 2 and 3 is **640x480 at 30 FPS**.

[image]

Past discussions on the subject have suspected that USB cable quality may be a factor in USB 2 mis-detections.  Are you using the official 1 meter cable supplied with the camera or your own choice of cable, please?

MartyG-RealSense, 2021-05-18:
Case closed due to no further comments received.

closed 2021-05-18 by MartyG-RealSense

AlphaRalph, 2021-05-25:
Hello,
I encountered a very similar problem and need help to figure it out.
Using the Realsense Viewer it shows me all three connected D435 cameras correctly within their USB 3.2 ports. The cameras are connected directly to three individual USB3 ports of the windows PC, without a hub. FW Version is 5.12.5, librealsense version is 2.35.2
Also when using the rs-fw-update.exe -l it show me all three devices with the correct USB Version.
Enabling the cameras with arbitrary settings within the viewer works perfectly fine.

When i try to enable the cameras via my python program it constantly fails with the "RuntimeError: Couldn't resolve requests" error at the first camera it tries to enable. The python program is based on the "box_dimensioner_multicam" example.
In addition I'd like to mention that this problem currently only occurs at one specific device, whereas the 10 other devices with the exact same software work more or less flawlessly.
I also noticed occasional timeout errors, where all three running cameras suddenly report a timeout error and stop streaming.

Thanks in advance and regards
Ralph

MartyG-RealSense, 2021-05-25:
Hi @AlphaRalph   A quick way to test whether one of the cameras is mistakenly being detected as being on a USB 2 connection would be to change the FPS from '15' to '6' in the script.

6 FPS is the only FPS speed supported for D415 and D435 / D435i at 1280x720 resolution on a USB 2 connection (the D455 has a minimum FPS of 5).  Therefore, if the **Couldn't resolve requests** error no longer occurs at 6 FPS then that would suggest that USB 2 mis-identification is the cause of the problem, as the error would occur at 15 FPS on USB 2 due to that FPS speed being unsupported in USB 2 mode.

AlphaRalph, 2021-05-25:
Thanks for the fast reply, but the camera settings are already:
848x480px @ 6fps

And those settings work when i try to activate the camera streams from the realsense viewer, but the exact same settings don't work in my python program, even though they work on 10 other devices of the same type.

MartyG-RealSense, 2021-05-25:
Have you tried swapping the USB cable of the camera that has the problem with a cable from one of the other cameras to eliminate the possibility of a faulty USB cable, please?

AlphaRalph, 2021-05-25:
Unfortunately this is not possible, since the device is in anther country that I am. I'm only connected via TeamViewer.
But I assume, that the cable might not be the cause of the issue, since all cameras work perfectly fine within the Realsense Viewer.

MartyG-RealSense, 2021-05-25:
And it is the same camera with a specific serial number each time, and not the 'first camera detected' in a set of 3 (which may be a different camera depending on which camera in the set gets detected first)?

AlphaRalph, 2021-05-25:
When i disable the failing camera within the device manager, so that only the other two are recognised by the program, the next time i try it, another camera fails to be enabled.
I've also tried hardware reset and reinstalling the firmware for each camera, nothing helped.

MartyG-RealSense, 2021-05-25:
I note your mention that your program is **based** on box_dimensioner_multicam.  When the **realsense_device_manager** part of this example is adapted by RealSense users for their own program, it tends not to go smoothly.  Could you describe the extent of any modifications you have made to the original box_dimensioner_multicam program please?

AlphaRalph, 2021-05-25:
In the DeviceManager i use a dictionary of <serialNumber, rs.config()> for each individual device, since i had troubles using only a single config for all devices.
In the further process I use a separate thread to retrieve all frames from all devices, but the program doesn't even get that far, since it fails during the "self._configs[device_serial].enable(device_serial)" step.

MartyG-RealSense, 2021-05-25:
The original box_dimensioner_multicam program should detect the attached devices at start-up of the program and automatically perform a calibration to the empty chessboard for those cameras before requesting that an object is placed on the chessboard once that calibration is completed.  So it is usable with different quantities of camera, even with just a single camera attached.

Was it the original unaltered version of the box_dimensioner_multicam program that caused you troubles?

AlphaRalph, 2021-05-25:
Of course I modified the calibration and processing part a lot, to fit my needs, but the DeviceManager class is nearly unchanged, except the described modifications.
I simply can't understand why there are 10 devices working normally, while one doesn't work at all.

MartyG-RealSense, 2021-05-25:
May I confirm that the problem is not with one specific camera every time, but with one camera in the **currently attached set**.  So that if 1 camera out of 3 failed in a test and then you attached 3 different cameras, 1 camera from **that particular set** would always fail.  So all the cameras are okay, and it is how the cameras are being **detected** that is the problem?

AlphaRalph, 2021-05-25:
Yes, allmost. It's always the first enumerated failing, since the ".enable(serialnumber)" throws an exception the program doesn't try any longer to enable the other cameras.
I don't see the difference in enabling the camera from code which permanently fails, to enabling it from the realsense viewer, which works perfectly fine.

MartyG-RealSense, 2021-05-25:
In the link below in another case that had the **Couldn't resolve requests** problem with box_dimensioner_multicam, a RealSense team member suggested an edit to **realsense_device_manager.py**.  The RealSense user reported that their problem was resolved after making the change.

https://github.com/IntelRealSense/librealsense/issues/2818#issuecomment-446899019

AlphaRalph, 2021-05-25:
If already seen #2818, but I'm using Python 3.6
Could an upgrade to a higher FW/ Librealsense version fix the issue?

MartyG-RealSense, 2021-05-25:
I would normally recommend using recent SDK versions and firmwares if possible.  If your cameras are currently in a mission-critical situation in another country right now then it may not be the best time to upgrade in case the project becomes broken.

SDK 2.35.2 was a notable point in SDK release history, as it was when new code for handling multicam was introduced that was designed to tackle problems related to **rs2::pipeline**.  

https://github.com/IntelRealSense/librealsense/pull/6467

It could not though help with other types of multicam problem such as incompatibility with a specific model of USB hub (you mentioned that you are not using a hub with this particular project though).

So if you wanted to check whether this transition point to new multicam code is affecting your project, you could try installing SDK **2.35.0**, the version before the change was introduced.  You should not need to change your firmware for this SDK version, as the **5.12.5.0** version you are using now is also the recommended firmware for 2.35.0.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.35.0

AlphaRalph, 2021-05-26:
I tried the following consecutively:
* Downgrad to pyrealsense 2.35.0
* Upgrade to pyrealsense 2.45.0 + FW 5.12.12.100
* Changed 1-config-per-device back to 1-config-for-all-devices (like in the box_dimensioner_multicam example)

Nothing changed a thing about the observed behaviour.
In addition I tried the following:
I have three cameras, lets call them A, B and C.
A is the first in the enumeration, so the first that is tried to be enabled. It fails every time.
If I disable A and B in the device manager, C can be enabled (only 1 device in the enumeration).
If I disable A and C, B can be enabled
But if I disable B and C, A can't be enabled, even though it is the only device in the enumeration.
Nevertheless, using the realsense viewer it's still possible to enable all devices. I'm desperate...

MartyG-RealSense, 2021-05-26:
And you do not have any streams enabled in the RealSense Viewer whilst you are running your Python program, please?

AlphaRalph, 2021-05-26:
No, when I try the program, the viewer is closed.

AlphaRalph, 2021-05-26:
To clarify, in the following code

`pipeline = rs.pipeline()`
`self._config.enable_device(device_serial)`
`pipeline_profile = pipeline.start(self._config)`

the statement `pipeline.start(self._config)` throws the exception.

MartyG-RealSense, 2021-05-26:
Let's go back to the error that you were experiencing, **RuntimeError: Couldn't resolve requests**.  There are two main reasons why you might encounter this error - if the streaming mode being requested is invalid (perhaps due to the camera being wrongly detected as being in USB 2 mode) or if there is an error in the stream definition code that means that the RealSense SDK cannot understand the request.

Looking back to an earlier comment, you said that you had set the stream to **848x480px @ 6fps**.  If the camera was being mis-detected as being in USB 2 mode though, 848x480 would cause a **Couldn't resolve requests** error.  This is because USB 2 mode does not support 848x480 - after 640x480 the next supported resolution is 1280x720.

[image]

AlphaRalph, 2021-05-26:
I tried every single resolution setting and it's definitely not the cause of the problem. Especially because if there is only one camera enabled in the device manager, it is possible to enable this single camera.

MartyG-RealSense, 2021-05-26:
A RealSense user recently shared a multicam Python script.  Their config and pipeline start code is similar to yours.  It may therefore be useful to look at an earlier point in their script to see how they handle serial numbers compared to your application.

https://github.com/ivomarvan/samples_and_experiments/blob/master/Multiple_realsense_cameras/multiple_realsense_cameras.py#L39

A RealSense application that uses a single camera does not require a serial number, as the number is for the purposes of distinguishing between different cameras.  So if the serial number wasn't being set correctly in a multicam setup, conceivably there could be a problem when trying to access more than one camera.

AlphaRalph, 2021-05-26:
I've just noticed, that the realsense viewer offers just for one camera the both infrared streams. Could this indicate that this specific camera is using USB3, whereas the other two devices are just using USB2, even though the viewer and the rs-fw-update.exe display an USB 3.2 connection?
Might it be possible, that the USB driver/ port is the cause of the problem?

MartyG-RealSense, 2021-05-26:
Only the left infrared camera (Infrared) is available on a USB 2 connection.  To access Infrared 2, a USB 3 connection is required.  Usually the camera would have the number **2.1** beside its name in the options side-panel if it were being detected as USB 2 though.

AlphaRalph, 2021-05-26:
The viewer says USB 3.2 for all devices, but on two of the three cameras only the depth stream is available. This causes my python program to fail, because i request the depth and the first infrared stream!
So only one question is left. How can this happen and more importantly, how to fix it?

MartyG-RealSense, 2021-05-26:
That could explain the **Couldn't resolve requests** error if the infrared stream can not be accessed.  

Do you mean that only the Depth selection box is shown in the Viewer and the Infrared section is completely missing?

[image]
  
If so, it would be rare: I can recall one past case of this occurring.

AlphaRalph, 2021-05-26:
Exactly

MartyG-RealSense, 2021-05-26:
In the situation of stream types not being accessible in the Viewer or other applications, guidance on the Intel page suggests a range of options.  Since one of the camera works but the other two do not, this would suggest that the SDK is okay and that the best suggestion of the ones listed on that page may be to try a firmware update of the affected cameras.

https://www.intel.com/content/www/us/en/support/articles/000030263/emerging-technologies/intel-realsense-technology.html

Bearing in mind that you have been using SDK 2.35.0 and 2.35.2, the maximum firmware version I would suggest going up to is **5.12.7.100**.  This is because from firmware 5.12.8.200 onwards, there were internal changes in the firmware that required a minimum of SDK **2.39.0**.

AlphaRalph, 2021-05-26:
Ok, I'll try the recommended FW, but I highly suspect the actual hardware of the PC either beeing faulty or having a problem with the USB driver.
If I was able to fix it, I'm going to note down the solution here.