realsenseai/librealsense #8970: RealSense d435 MATLAB settings

Fredzibob, 2021-05-05:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.12.100 | 
| Operating System & Version |   Win10  | 

| Platform                                 | PC  |
| SDK Version                           |   2.44.0   | 
| Language                               |  MATLAB      | 
| Segment			         |  Robot                          | 

Hi!

I using a RealSense d435 to get a pointcloud in MATLAB.
Everything is working but when I use Computer Vision Library's ROI function, it takes almost 2 seconds per iteration, which is not useful in my application.

I therefore wanna use the built in parameters on the RealSense like in the RealSense Viewer. It has a "depth table" that worked really well.
Therefore my question:

How to I use the options in Matlab? I have looked through the examples, but cant seem to find what i want. Only that option (34) max_distance might be what im looking for. How to I incorporate this in my matlab code?

My simple matlab script:
`% Make Pipeline object to manage streaming
pipe = realsense.pipeline();
% define point cloud object
pcl_obj = realsense.pointcloud();
% Start streaming on an arbitrary camera with default settings
pipe.start();

for j = 1:1
frames = pipe.wait_for_frames();
% Select depth frame
depth = frames.get_depth_frame();

% get point cloud points without color
pnts = pcl_obj.calculate(depth);
vertices = pnts.get_vertices();

% optional: populate MATLAB point cloud object
pCloud = pointCloud(vertices);
roi = [-inf,inf,-inf,inf,-inf,0.5]; %Region of interest
roi_idx = findPointsInROI(pCloud,roi); %Indices of the points in ROI
pcl2 = select(pCloud,roi_idx);
%display point cloud
pcshow(pcl2);
%pause(0.01);
disp(num2str(j))
end

pipe.stop();`

Fredzibob, 2021-05-11:
Hmm, it does not seem like its loading the config file... Its not doing any clamping/threshold at all.
Are we sure its actually uploading a file? It seem more like its just creating a file and uploading it.

MartyG-RealSense, 2021-05-11:
If you created the custom json in the RealSense Viewer by exporting a json file, did you define a depth clamp in the **Advanced Controls > Depth Table** section of the Viewer's options side-panel before exporting the json, please?

[image]

Fredzibob, 2021-05-12:
My json does "unfortunately" not have the extra comma at the end.

Would this MATLAB scripts load the settings and show a point cloud?
right now, it only shows a pointcloud without depth table applied

`% Make Pipeline object to manage streaming
pipe = realsense.pipeline();
% define point cloud object
pcl_obj = realsense.pointcloud();
% Start streaming on an arbitrary camera with default settings
pipe.start();
advanced_mode_example_1;

 for j = 1:10
frames = pipe.wait_for_frames();
% Select depth frame
depth = frames.get_depth_frame();

% get point cloud points without color
pnts = pcl_obj.calculate(depth);
vertices = pnts.get_vertices();

% optional: populate MATLAB point cloud object
pCloud = pointCloud(vertices);
%roi = [-inf,inf,-inf,inf,-inf,0.5]; %Region of interest
%roi_idx = findPointsInROI(pCloud,roi); %Indices of the points in ROI
%pcl2 = select(pCloud,roi_idx);
%display point cloud
pcshow(pCloud);
pause(0.01);
disp(num2str(j))
 end
pipe.stop();
function advanced_mode_example_1()
    % Make a context object
    ctx = realsense.context();
    % Get a device from context
    devs = ctx.query_devices();
    dev = devs{1};

    % Make sure the device supports advanced mode
    if ~dev.is('advanced_mode')
        error('Device doesn''t support advanced mode!');
    end
    
    % Make sure the device is in advanced mode
    adv = dev.as('advanced_mode');
    if ~adv.is_enabled()
        adv.toggle_advanced_mode(true);
        
        % Pause execution to give the device time to reconnect
        oldState = pause('on');
        pause(2)
        pause(oldState);

        % Grab the device
        devs = ctx.query_devices();
        dev = devs{1};
        adv = dev.as('advanced_mode');

        % Make sure the device supports advanced mode
        if ~adv.is_enabled()
            error('Device didn''t enter advanced mode');
        end
    end
    
    json_string = adv.serialize_json();
    json = jsondecode(json_string);

    % Modify json, jumping through a few hoops because of 
    % the interaction between how the json is defined and matlab
    %json.controls_color_gain = string(str2double(json.controls_color_gain) - 4);
    %json_string = jsonencode(json);
    % matlab replaces - with _ to create valid structure field names, we now need to switch them back
    %json_string = strrep(json_string, '_', '-');

    % Upload new json
    adv.load_json(json_string);
    
end`

MartyG-RealSense, 2021-05-12:
I have examined the above script at length.  Part of the difficulty in debugging it is the difference between the standalone **advanced_mode_example.m** script and integrating that into a larger overall application.  My MATLAB wrapper programming knowledge is also limited.

As an example of the complexity, I would normally expect ctx instructions such as **devs = ctx.query_devices();** to be placed **before** the pipe.start instruction in a script.  However, if the pipe is started and **then** the block of json loading code is being called then the ctx instructions in that json-loading code block may be ineffective.

Fredzibob, 2021-05-13:
Do you know who wrote the matlab wrapper and .m scripts? Just so we can for sure get a confirmation of how it should work ideally? And it what order things are to be called.

MartyG-RealSense, 2021-05-13:
The GitHub history of the MATLAB wrapper shows that @lramati was the original developer.  The full history of the wrapper can be viewed at the link below,

https://github.com/IntelRealSense/librealsense/commits/master/wrappers/matlab

closed 2021-05-25 by MartyG-RealSense