realsenseai/librealsense #8971: Failing resolution/FPS configuration of RealSense L515 camera

abotoi, 2021-05-05:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | Intel RealSense L515 | 
| Firmware Version                   | 01.05.03.00 | 
| Operating System & Version | Ubuntu 20.04.2 LTS  | 
| Platform                                 | PC  |
| SDK Version                            |  2.43.0-0                      | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|   Robot                          | 

### Issue Description
I observe the below behavior:
I try to choose a non-default resolution for the depth frame. The problem was that no data was published on the topics with either of the combination (1024x768 or 320x240).
When I specifically specify the default resolution, camera node successfully started and data could be visible.

RealSense L515 Camera stream profiles supported for Depth Sensor are the one below: RealSense L515 Camera stream profiles supported for Depth Sensor are the one below: 
**Case 1)** Depth  **1024x768      @ 30Hz     Z16**   -> error when try to apply - camera node seems to start, only /camera/image_raw topic is visible, but not data is published on this topic.  
user@glaic3n144:/data_samples$ ros2 topic hz /camera/image_raw

user@glaic3n144:/data_samples$ fg
Note that after sending the command “ros2 launch realsense2_camera rs_launch.py depth_width:=1024 depth_height:=768 depth_fps:=30. &”, the message **“RealSense Node Is Up!” is not displayed.**

**Case 2)** Depth  640x480       @ 30Hz     Z16   -> Default resolution (only this one was applying) -> message “RealSense Node Is Up” is displayed.
Data is published:
ros2 topic hz /camera/depth/image_rect_raw
average rate: 29.856
        min: 0.032s max: 0.037s std dev: 0.00096s window: 31

**Case 3)** Depth  **320x240       @ 30Hz     Z16**   -> error when try to apply - camera node seems to start, topics are available, but not data is published on this topics.  
ros2 topic hz /camera/color/image_raw
WARNING: topic [/camera/color/image_raw] does not appear to be published yet
Note that after sending the command “ros2 launch realsense2_camera rs_launch.py depth_width:=320 depth_height:=240 depth_fps:=30. &”, the message **“RealSense Node Is Up!” is not displayed.**

**Output log:
Case 1)** ros2 launch realsense2_camera rs_launch.py depth_width:=1024 depth_height:=768 depth_fps:=30. &
[realsense2_camera_node-1] [INFO] [1619441674.796572929] [RealSenseCameraNode]: depth stream is enabled - width: 1024, height: 768, fps: 30, Format: Z16
....
[realsense2_camera_node-1] [ERROR] [1619441675.295955722] [RealSenseCameraNode]: An exception has been thrown: /opt/ros2_ws/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:1899:Device already streaming!
[realsense2_camera_node-1] Device already streaming!
[realsense2_camera_node-1] stop_streaming() failed. UVC device is not streaming!
[realsense2_camera_node-1] terminate called after throwing an instance of 'rs2::wrong_api_call_sequence_error'
[realsense2_camera_node-1]   what():  close() failed. UVC device was not opened!
[ERROR] [realsense2_camera_node-1]: process has died [pid 535, exit code -6, cmd '/opt/ros2_ws/install/realsense2_camera/lib/realsense2_camera/realsense2_camera_node --ros-args -r __node:=camera -r __ns:=/camera --params-file /tmp/launch_params__2w09r3q'].

**Case 3)** ros2 launch realsense2_camera rs_launch.py depth_width:=320 depth_height:=240 depth_fps:=30. &
[realsense2_camera_node-1] [INFO] [1619442308.544069038] [RealSenseCameraNode]: depth stream is enabled - width: 320, height: 240, fps: 30, Format: Z16
...
[realsense2_camera_node-1] [ERROR] [1619442309.053691379] [RealSenseCameraNode]: An exception has been thrown: /opt/ros2_ws/realsense-ros/realsense2_camera/src/base_realsense_node.cpp:1899:Device already streaming!
[realsense2_camera_node-1] Device already streaming!
[realsense2_camera_node-1] stop_streaming() failed. UVC device is not streaming!
[realsense2_camera_node-1] terminate called after throwing an instance of 'rs2::wrong_api_call_sequence_error'
[realsense2_camera_node-1]   what():  close() failed. UVC device was not opened!
[ERROR] [realsense2_camera_node-1]: process has died [pid 770, exit code -6, cmd '/opt/ros2_ws/install/realsense2_camera/lib/realsense2_camera/realsense2_camera_node --ros-args -r __node:=camera -r __ns:=/camera --params-file /tmp/launch_params_uysigjl7'].

I am testing in docker on Ubuntu 20.04.2 LTS. Ros version foxy, RealSense SW version 2.43.0-0.
Platform NUC with - Intel(R) Core(TM) i7-9750H CPU @ 2.60GHz

doronhi, 2021-05-09:
infra and depth are 2 separate streams that belong to the same sensor. Although the wrapper does not enforce it, it is not possible to set different resolutions to these streams.
The following command should work for you:

ros2 launch realsense2_camera rs_launch.py depth_width:=1024 depth_height:=768 depth_fps:=30. infra_width:=1024 infra_height:=768 infra_fps:=30.

In a future version, we plan to set the configuration per sensor and not per stream and thus to avoid such issues.

MartyG-RealSense, 2021-07-02:
Hi @abotoi   Do you require further assistance with this case, please?  The case will be closed after 7 days from the time of writing this if we do not hear from you.  Thanks!

MartyG-RealSense, 2023-07-18:
Case closed due to no further updates for the L515 camera model.

closed 2023-07-18 by MartyG-RealSense