realsenseai/librealsense #8975: Realsense L515 LiDAR external trigger not working

DragonDeux, 2021-05-05:
| | |
|---|---|
|**librealsense**|2.45.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense L515|
|**Serial Number**|f0270251|
|**Firmware Version**|01.05.05.00|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|0003b507b62d|
|**Firmware Update Id**|0003b507b62d|

Hi, 

Could you please help us to figure out why the external hardware trigger not working for the L515 camera?

DragonDeux, 2021-05-06:
Was this a super complicated question???

RealSenseSupport, 2021-05-07:
Hi @DragonDeux 

There were some changes to the whitepaper recently regarding using the sync capability for L515.  Here is the link to the latest:  https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup

Can you let us know exactly what you are seeing and what is meant by the trigger not working for you?

DragonDeux, 2021-05-07:
Hi @RealSenseSupport 

Yes, I have already read this article. My question is quite simple. I'm using Rappery Pi4 as an eternal trigger generator, which with a 60ms bandwidth. I would like first to use only one L515 camera controlled by the external trigger(HW). However, L515 seems completely ignore the external trigger setting and free-run, some snapshot for better understanding of my question attached below:

Trigger:
[image]

Wires:
[image]

Viewer:
[image]

Regards,

RealSenseSupport, 2021-05-07:
Hi @DragonDeux 

Thanks for the information.  Have you experimented only dropping the sync signal low and keeping it low to determine if the L515 is getting the sync signal and blanking the laser and not streaming depth images to the host system?  This way we debug the actual sync signal itself and camera getting sync signal.  I've also seen users in the past have the sync signal on the wrong pin.  Meaning having the ground signal on the sync signal and vice versa.  This would cause the polarity to be reversed meaning the camera would turn off laser and stop streaming to host when signal is high.  I've also seen where users have had the sync signal connected to the no connect pin which is pin 3.  So debugging and making sure the wires are connected to the proper pins is where I'd first start the debug efforts.

Let me know if you were able to do these things and your findings.

DragonDeux, 2021-05-07:
Hi @RealSenseSupport 

I have double verified the cabling that is correct. The problem is the external synchronization was working initially by 2 days ago,but suddenly broken(seems like). Moreover, I have 3 L515 camera that have encountered the same issues! That's why I have opened two issues about this problem. 

Regards

RealSenseSupport, 2021-05-07:
Hi @DragonDeux 

This is interesting.  You've checked all the hardware pins and connection and the setup was working.  Can you elaborate as working?  You were getting proper signal and blanking on the camera when the signal was low on the sync ping?

Also you stated you opened two issues, is the other one related to this exact topic we are discussing here?  If you can close out one or the other that way the discussion is in a singular place.

DragonDeux, 2021-05-09:
Hi @RealSenseSupport 

Yes, I'm pretty sure there is no cabling problem since it was working at begin of the experiment as I said in the previous message. Besides, I have also used multimeters to verify the connectivity between each side of the wire, which is correct without any short or open circuit, this verification has been done (doing even) for my three L515 cameras. 

**My question;**

Is there a strict order for properly trigger the L515 camera by using the external HW signals? 

**Answer to your question:**

"You were getting a proper signal and blanking on the camera when the signal was low on the sync ping?" 

The L515 camera has completely ignored the external trigger commands, il free-run.... (For three of my L515 cameras). Generally, did L515 do not take any acquisition if "inter cam sync mode" had been activated, right? In my case, L515 does not take into account whether the parameters are activated or not, it just free-run to take depth maps at 30FPS. 

Regards

RealSenseSupport, 2021-05-26:
Hi @DragonDeux 

Question1 answer:  There is no strict order of triggering the HW signal.  

Question1 answer:  It is a combination of the inter_cam_sync mode and HW signal that either put the camera in position to stream to host or not.  As outlined in the whitepaper, the inter_cam_sync should be set ahead of any manipulation of the HW sync signal.  When HW sync signal is high, then frames will be sent to host...when HW sync signal is low then no frames will be sent to host.

closed 2021-06-18 by RealSenseSupport